MarlinFirmware/Marlin #23429: [BUG] (bug summary)Marlin 2.0.9.3 HOTBED PIT TUNING

SpRiNgToIfEl, 2022-01-02:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

In the new version 2.0.9.3 I have problems with HOT BED PID TUNING. With the command "M303 E-1 C3 S80" only the first and second pid test are run around 80 °C, the third pid test is runs over 90 °C and then Marlin aborts the test due to overheating.

A test at 60 ° C and 70 ° C does not fail and all tests are completed successfully.

Ender 5 Plus Aluminum print bed with silicone heating mat (KIS3D 30).

### Bug Timeline

N.A.

### Expected behavior

Test only max. 70 °C

### Actual behavior

hotbed thermal runaway

### Steps to Reproduce

1. Start the Printer and connect to Octoprint or Prontaface.
2. Enter Command "M303 E-1 C3 S80"
3. PID TUNE Test 1 and 2 are successfully
4. PID TUNE Test 3 runs in thermal runaway >90°C
5. 

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 5 Plus

### Electronics

BTT SKR PRO, TMC2208, KIS3D SILICONE HEATING MAT (30)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[image]

SpRiNgToIfEl, 2022-01-02:
one last test, only tested with two cycles at 80 ° C, no values output. what could be the problem?

[image]

thinkyhead, 2022-01-05:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

NezSez, 2022-01-12:
If using git this can help speed up the tests:
https://git-scm.com/docs/git-bisect

muistiloppu, 2022-01-12:
I have experienced exactly same problem with the bed PID and I made bug repot of it too #22982.

thisiskeithb, 2022-01-12:
Duplicate of #22982

closed 2022-01-12 by thisiskeithb