MarlinFirmware/Marlin #23431: Visual studios reports no valid marlin file

mpete53, 2022-01-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2x bug fix from github
Visual studios, open folder, marlin folder.
files show and can be edited.
there is an report report opening folder
Please update your includepath, Squiggles are disabled for this translation unit C:\Marlin setup Ender3\marlin-2.0X\Marlin\src
MarlinCore.ccp
when auto marlin clicked it states 
current  open folder doesn't contain a (valid) Marlin 2x project.

### Version of Marlin Firmware

2v bugfix downloaded 1/1/22

### Printer model

ender 3

### Electronics

bigtree skr mini e3 v2

### Your Slicer

Cura

mpete53, 2022-01-02:
Yes downloaded it yesterday

sjasonsmith, 2022-01-02:
Does it work if you move it to a folder with no spaces in the name?

Also when opening the folder please be sure you are opening the folder that has the platformio.ini file inside it.

mpete53, 2022-01-02:
Changed the name and I was in the wrong folder runs now but failed.
error message
[C:\users\mark\,platformio\packages\framwork-arduinoststm32\variants\Marlin_F103Rx\PeripheralPins.c]
C:\MarlinSetupEnder3\Marlin-2.0X\.pio\build\STM32F103CR_btt\sconsign39\dblite: No such file or directory
Thanks much

sjasonsmith, 2022-01-02:
> dblite: No such file or directory

I'm not sure why this happens, but when it does I just build again and it always works.

mpete53, 2022-01-02:
Thanks again
seems to be working

closed 2022-01-02 by sjasonsmith

thinkyhead, 2022-01-03:
> I'm not sure why this happens

It seems to be a mystery in general! The only thing I can think is that maybe a short `sleep` could be added somewhere in the SCONS / PIO scripts to give the filesystem a chance to spawn the files it's looking for.