MarlinFirmware/Marlin #23434: [BUG] (Soft Endstop (Z_MIN_POS) for Z axis interfers with (NOZZLE_TO_PROBE_OFFSET) Z)

sk8board, 2022-01-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Changing Soft Endstop Z_MIN_POS in Configuration.h affects NOZZLE_TO_PROBE_OFFSET for the Z axis.

### Bug Timeline

2.0.9.3 December 30, 2021

### Expected behavior

I expect NOZZLE_TO_PROBE_OFFSET will not change when changing Soft Endstop Z_MIN_POS.

### Actual behavior

When Soft Endstop Z_MIN_POS is changed, then NOZZLE_TO_PROBE_OFFSET needs to be incrementally changed by the same amount.

### Steps to Reproduce

**Trial 1**

Configuration.h
#define Z_MIN_POS 0

G code entered in Octoprint Terminal
M502
M500
M851 Z0
M211 S0; turn off soft endstop
G28
M500
M501
M851; Terminal shows NOZZLE_TO_PROBE_OFFSET { 25.0, 27.1, 0 }
M211; Terminal shows Soft Endstop (Z_MIN_POS) Min:  X0.00 Y0.00 **Z0.00**   Max:  X300.00 Y295.00 Z300.00

Result: the measured offset from bltouch probe to nozzle is -2.0mm (which is expected)

**Trial 2**

**Configuration.h**
#define Z_MIN_POS 2

G code entered in Octoprint Terminal
M502
M500
M851 Z0
M211 S0; turn off soft endstop
G28
M500
M501
M851; Shows NOZZLE_TO_PROBE_OFFSET { 25.0, 27.1, 0 }
M211; Shows Soft Endstop (Z_MIN_POS) Min:  X0.00 Y0.00 **Z2.00**   Max:  X300.00 Y295.00 Z300.00

Result: the measured offset from bltouch probe to nozzle is 0.0mm (expected measured to be -2.0mm)

**Trail 3**

**Configuration.h**
#define Z_MIN_POS -5

G code entered in Octoprint Terminal
M502
M500
M851 Z0
M211 S0; turn off soft endstop
G28
M500
M501
M851; Shows NOZZLE_TO_PROBE_OFFSET { 25.0, 27.1, 0 }
M211; Shows Soft Endstop (Z_MIN_POS) Min:  X0.00 Y0.00 **Z-5.00**   Max:  X300.00 Y295.00 Z300.00

Result: the measured offset from bltouch probe to nozzle is -7.0mm (expected measured to be -2.0mm)

**Summary**
Z_MIN_POS = 0 and NOZZLE_TO_PROBE_OFFSET = 0, the measured offset from bltouch probe to nozzle is -2.0mm (which is expected)
Z_MIN_POS = +2 and NOZZLE_TO_PROBE_OFFSET = 0, the measured offset from bltouch probe to nozzle is 0.0mm (expected measured to be -2.0mm)
Z_MIN_POS = -5 and NOZZLE_TO_PROBE_OFFSET = 0, the measured offset from bltouch probe to nozzle is -7.0mm (expected measured to be -2.0mm)

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/7799630/Config.zip)

### Version of Marlin Firmware

2.0.9.3

### Printer model

Two Trees Sapphire Plus

### Electronics

Stock MKS Robin Nano V1.2  (mks_robin_nano35)

### Bed Leveling

No Bed Leveling

### Host Software

OctoPrint

thinkyhead, 2022-01-05:
Try defining `Z_PROBE_LOW_POINT` as -10 and repeat your testing.

sk8board, 2022-01-07:
I have changed `Z_PROBE_LOW_POINT` to -10 and performed the same test again. The result was the same. Changing `Z_PROBE_LOW_POINT` to -10 did not change the result.

sk8board, 2022-01-28:
I believe my testing shows there is a bug. When Soft Endstop `Z_MIN_POS` is not zero, then `NOZZLE_TO_PROBE_OFFSET` needs to be incrementally changed by the same amount.

I assume `Z_MIN_POS` should not affect `NOZZLE_TO_PROBE_OFFSET`. If my assumption is incorrect, then I have confirmed a bug.

sk8board, 2022-02-20:
@thinkyhead  is my work good enough to confirm a bug?

descipher, 2022-02-20:
What is the purpose for setting z_min_pos other than zero? 
I would think the offset works correctly here but setting z_min_pos to something other than zero seems very strange. In each case you have a z-probe offset of 0 If you set the physical location to -5 then the probe will need to move to -5 to touch where the physical layer is. The hotend would be @ 0mm above the physical plane in this case. The actual distance between the probe and the hotend are equal if the offset is zero. If the probe is not actually at 0 then the difference on contact with the plane is the offset.

sk8board, 2022-03-05:
> What is the purpose for setting z_min_pos other than zero?

Here is what the Marlin configuration definition states:

> **Usually** the [XYZ]_MIN_POS values are set to 0, because endstops are positioned at the bed limits. 

The word "Usually" indicates there are use cases to have a `Z_MIN_POS` other than zero. The fact that developers went thru the work to create`Z_MIN_POS` also indicates there is a use case they were considering.

sk8board, 2022-03-10:
@thinkyhead here is another person experiencing this same bug. Should the bug tag be added to this issue?

issue #23796

sk8board, 2022-04-17:
@thisiskeithb since another person is experiencing the same bug, should this bug be considered verified?

thinkyhead, 2022-04-21:
> The fact that developers went thru the work to create `Z_MIN_POS` also indicates there is a use case they were considering.

Developers like me? Not necessarily. Go back to Marlin 1.0.0 and there isn't much to indicate any intent or use case. I struggle to come up with one off the top of my head….

It sounds like we should simply use 0 in some places where `Z_MIN_POS` is now being used. Hopefully someone digs into the code to fix it up so that `Z_MIN_POS` can be whatever is preferred. But, we may end up needing to assert that Z0 (native machine space) is where the bed is found.

> Result: the measured offset from bltouch probe to nozzle is

Where is the "measured offset" coming from? Are you referring to the Z Probe Offset Wizard?

InsanityAutomation, 2022-04-21:
Z_MIN_POS can be useful to allow a nozzle below Z in cases where a wiper pad is installed off the bed to the side, parking or probe/tool stowing and other considerations. Its likely that in the case where the Z_MIN is not 0, you also want a manual home position defined, especially if homing with a probe that is reading the bed.

descipher, 2022-04-21:
Yes, Z_MIN_POS could be used as mentioned. But obviously the code never was written that way. In this case the OP implies the offset is no longer correct when Z_MIN_POS is set to a negative value. That is not correct. The Z probe offset is clearly the distance between the probe and surface contact and not the value of the Z_MIN_POS define. The code would need changes to work correctly with a negative value. The implied intent is not a bug it’s an FR.

sk8board, 2022-04-24:
> But obviously the code never was written that way. In this case the OP implies the offset is no longer correct when Z_MIN_POS is set to a negative value.

If what you are stating is the intent, then please update the definition of Z_MIN_POS to show the value should never be negative. If I would have known that up front, I would have saved many hours of chasing my tail to discover the value does not work when negative.

https://marlinfw.org/docs/configuration/configuration.html

Thank you for your hard work with making Marlin what it is.

sk8board, 2022-05-26:
@thinkyhead @descipher 

> But obviously the code never was written that way. In this case the OP implies the offset is no longer correct when Z_MIN_POS is set to a negative value.

If what you are stating is the intent, then please update the definition of Z_MIN_POS to show the value should never be negative. If I would have known that up front, I would have saved many hours of chasing my tail to discover the value does not work when negative.

https://marlinfw.org/docs/configuration/configuration.html

Thank you for your hard work with making Marlin what it is.

InsanityAutomation, 2022-05-26:
On idex systems, a value below 0 is actually required in conjunction with a manual home position in order to allow tool offsets to function.

descipher, 2022-05-26:
Required on Z? Why?

descipher, 2022-05-26:
Are there any slicers that generate gcode for a negative Z plane?

InsanityAutomation, 2022-05-26:
> Are there any slicers that generate gcode for a negative Z plane?

No, it came about a couple years ago when it was more strictly enforced, and prevented the tool offset shift when bringing the second tool below 0 prior to a coordinate shift. The workaround was to set z min to -3 and manual home to 0 giving a 3mm tool tolerance.

descipher, 2022-05-26:
No probe involved, so yes you could that. I can think of other ways to address a tool alignment problem but I suspect it's more a design issue.

Triangulix, 2022-07-30:
Maybe I have the same issue, but I'm not sure.
I'm currently trying to find out why the z-offset wizard does not lower the nozzle to the point where the probe touches the surface.

sk8board, 2022-09-30:
@thinkyhead @descipher 

I can understand that making a change to support a negative value for `Z_MIN_POS` is not desirable. If that is the case, then please update the definition of `Z_MIN_POS` to show `Z_MIN_POS` should not be negative.

https://marlinfw.org/docs/configuration/configuration.html#movement-bounds

Simply adding a sentence (as shown in bold below) can save a user an enormous amount of troubleshooting time.

> These values specify the physical limits of the machine. Usually the [XYZ]_MIN_POS values are set to 0, because endstops are positioned at the bed limits. **Z_MIN_POS should never have a negative value.** [XYZ]_MAX_POS should be set to the farthest reachable point. By default, these are used as your homing positions as well. However, the MANUAL_[XYZ]_HOME_POS options can be used to override these, if needed.

I appreciate your support.

closed 2022-12-23 by github-actions[bot]