MarlinFirmware/Marlin #23439: [BUG] (MAX31865 no longer works after recent changes)

kurtis-potier-geofabrica, 2022-01-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin now reports 989 as the temperature for both E0 and E1 after the recent changes to the MAX31865.cpp and associated files. With the older versions of Marlin, we were able to read temperatures successfully with the same hardware. Currently running on a Bigtreetech GTR V1.0 board using SPI pins as follows:

#define TEMP_0_CS_PIN                       PD4 
#define TEMP_0_SCK_PIN                      PB3 
#define TEMP_0_MISO_PIN                     PB6 
#define TEMP_0_MOSI_PIN                     PG15 /

#define TEMP_1_CS_PIN                       PC12 
#define TEMP_1_SCK_PIN           TEMP_0_SCK_PIN
#define TEMP_1_MISO_PIN         TEMP_0_MISO_PIN
#define TEMP_1_MOSI_PIN         TEMP_0_MOSI_PIN

These pins are different from the defaults in the firmware, but did work on previous versions of Marlin. Also have confirmed that SPI bus can be read by a Raspberry pi directly before entering the board. I can revert to the older firmware and get readings.

### Bug Timeline

new

### Expected behavior

Read temperatures correctly. 

### Actual behavior

Temperatures read 989 on boot. Err is then displayed, 

### Version of Marlin Firmware

2.0.9.3

### Electronics

BTT GTR V1.0 STM32F4

### Add-ons

M5 extender

### Additional information & file uploads

[MAX31865_issue.zip](https://github.com/MarlinFirmware/Marlin/files/7803810/MAX31865_issue.zip)

kurtis-potier-geofabrica, 2022-01-04:
After some more diagnosis, can confirm that the firmware is trying to access the software SPI bus (turned on debug and added a short delay before the print statements so I could connect over serial before the board errors out).

zeleps, 2022-01-04:
Hi @kurtis-potier-geofabrica,

first of all, uncomment line 28 in temperature.cpp (#define IGNORE_THERMOCOUPLE_ERRORS), which will allow you debug the issue properly. Please keep in mind that this disables basic thermal runaway protection, so don't try to heat up your printer with this setting. Also enable MAX31865 debugging option (line 44 @ MAX31865.h).

Which is the last marlin release that worked for you? Is the SPI you're trying to use a hardware SPI? If yes, you may also try #define TEMP_SENSOR_FORCE_HW_SPI to see if it works for you.

I haven't had much luck using it with software SPI myself, but I didn't really look into it further, since I'm using HW SPI on my setup. Please provide some feedback on the questions above and we'll see what we can try next.

kurtis-potier-geofabrica, 2022-01-04:
I will do that and be careful of that. The Marlin release 2.0.9.1 is working (downloaded in late September of 21). I have enabled MAX31865 debugging. I'm using Software SPI. I used to the debugging for MAX31865 to confirm this as well. I can also consider what hardware SPI would look like for my board. I'm using a screen that I believe also uses that bus though. My only reason to upgrade is that I get the occasional amount of interference where the temp will spike around 30C. I've seen this go away on my pellet extruder printers by using bang bang heating instead of PID. The new options implemented in 2.0.9.3 look very intriguing so that I could use PID again with my smaller systems without any random spikes. I will try out all of these options tomorrow. Thank you!

thinkyhead, 2022-01-05:
Changes were most recently made in #23216 by @zeleps — so please check to see if reverting those changes helps. I am taking changes in this area on faith, so your testing is very important to helping get this stuff stabilized.

zeleps, 2022-01-05:
There are also some important changes regarding software SPI in #22682, which are not included in 2.0.9.1. Try a commit before #23216 first and before #22682 after that, to see which causes the problem. I'll try to help you pinpoint the issue and fix it.

Regarding hw SPI, you can share an SPI with other devices (like the screen you're mentioning, that's what I'm doing), just use a different CS pin.

kurtis-potier-geofabrica, 2022-01-05:
Good info. This gives me a bunch of things to try. I will have some results to share later today.

zeleps, 2022-01-05:
If you're thinking of trying hw SPI btw, take a look at this hack: https://github.com/GadgetAngel/Adafruit-MAX31865-V1.1.0-Mod-M/issues/1#issuecomment-827122286

kurtis-potier-geofabrica, 2022-01-05:
I flashed with everything prior to #23215 (Fix and improve MAX31865 by Zeleps). I get somewhat normal temps (room temp + 5C) for a few seconds and then Temp Measurement Errors. If I flash back to 2.0.9.1 (before #22682), then I get normal room temps and there is no errors. It looks like there was some renaming also done between #23215 and #22682. My working version is after #22660, so the problem is somewhere between then and now (likely #22682 or the renaming). This is still on the SW spi bus. I will next look at ordering parts for the hack Zeleps mentioned (unless I have other reasonable access on the board) there so that can be an option for me to try for now.

zeleps, 2022-01-05:
That's good testing @kurtis-potier-geofabrica! Now that you mention it, the few-seconds-normal pattern has occurred in my tests as well, but I didn't test it further with sw SPI. I'll try some tests myself and see what I can find.

kurtis-potier-geofabrica, 2022-01-05:
That would be great! Thank you

zeleps, 2022-01-06:
I switched to sw SPI, and just like @kurtis-potier-geofabrica, I am receiving normal values at first (for a few reads):

`Recv: MAX31865 RTD MSB:64 LSB:118`

and then, after a few initializations (SD, capabilities dump etc), SPI only reads 0xFF:

`Recv: MAX31865 RTD MSB:255 LSB:255`
`Recv: MAX31865 read fault: 65535`

Most obvious explanation is that some initialization process affects SPI. I'll try to pin it down and return with more info.

zeleps, 2022-01-06:
Just confirmed that disabling SD support eliminates the issue. @kurtis-potier-geofabrica please verify this. I'll delve into this later.

kurtis-potier-geofabrica, 2022-01-06:
@zeleps I tried this, but it did not fix the problem for me. (I tried on both the non-working versions I've been experimenting with). My output from the SPI debug is a little different. I get the attached ouptut with ignore thermocouple errors on and max31865 debug on. Will look into the code more as well and see if I can find anything and try things out. I'm not the strongest with C++ but not totally illiterate either.  
[image]

zeleps, 2022-01-06:
The last bit of the LSB is an error flag, and in all the values you've posted the bit is set. From the messages I understand that this is the version prior to my changes, so could you please share the output you're getting with #23215?

I took a look at your board's pinout and I see that you're using the stepper driver SPI pins. Are you using the SPI for controlling the stepper drivers as well?

kurtis-potier-geofabrica, 2022-01-06:
I did test both, but didn't log output. Here is the output after #23215. For my board, I am not using the spi bus for steppers. My drivers are external [drivers](https://www.amazon.com/Digital-Microstep-Stepper-Controller-2-phase/dp/B07YWZRXGR/ref=sr_1_4?crid=1M7PQQ8DEZUKB&keywords=microstep+driver&qid=1641489371&sprefix=microstep+driver%2Caps%2C100&sr=8-4).
I have them configured as A4988, which is not using the spi. I do not use any of the other pins on the traditional E0,E1,E2 for steppers. My extruder is on the traditional Z axis slot and my Z axis steppers are on the M5 extender. My full pins file is attached here for reference. 

[image]
[pins_BTT_GTR_V1_0.zip](https://github.com/MarlinFirmware/Marlin/files/7823610/pins_BTT_GTR_V1_0.zip)

zeleps, 2022-01-06:
The line `MAX31865 Fault: (1) >>` shows the fault register value of the sensor. This value - 1 - is not a valid fault register value. It's strange, because the RTD values you're getting (e.g. 16599) are valid room temperature values, but the least significant bit is set, indicating fault.

Obviously your case is different from my case. It would have some value if you could connect to the serial as soon as the screen is cleared on reboot, in order to catch the early initialization messages. This would show what the first few reads look like, and when faulty reads occur.

It would also be a good indication for a possible issue with SoftwareSPI.cpp if you could hook up the sensor on the hw SPI (MISO=PB14, SCK=PB13, MOSI=PB15, any free pin will do for CS) and see if it works.

There is also this text by @GadgetAngel, which is specific to your board, and it might give you some insight: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/179#issue-741317998

kurtis-potier-geofabrica, 2022-01-06:
I temporarily added a delay of 3 seconds to the softSPIBegin function and was able to connect during this delay period.
[image]

I will test out the hardware spi bus as well and see how that does. I'll assume that I can keep my current chip select pins the same and just switch the communication lines over as you have outlined there.

kurtis-potier-geofabrica, 2022-01-06:
The behavior is the same on the hardware spi bus for me as the software. I dont have screen hooked up, so can't see quite as much. I slipped in an M105 before error to get a temp report. T0 read -242 while T1 read about room temperature. 
[image]

zeleps, 2022-01-06:
Ok, now T0 reads only 255 from the SPI (obvious read error) while T1 is working correctly. This is quite different than the screenshots you sent before, where both sensors read the room temperature with some fault indication. What was the exact setup you used this time?

kurtis-potier-geofabrica, 2022-01-06:
I did the SPI pins as you suggested above and left my chip selects as PD4 and PC12.

zeleps, 2022-01-06:
You must omit the definitions for sck/miso/mosi though, in order to enable hw SPI (or uncomment `#define TEMP_SENSOR_FORCE_HW_SPI` in Configuration_adv.h). You only need to declare the CS pins.

kurtis-potier-geofabrica, 2022-01-07:
Yeah, I had #define TEMP_SENSOR_FORCE_HW_SPI uncommented.

zeleps, 2022-01-07:
I'll get a second sensor on Monday and do some tests with two sensors. Since one of them works, this could have something to do with the SPI initialization being done more than once. I'll get back to you on this. Meanwhile, you could unplug one of the sensors and see if the other works.

zeleps, 2022-01-08:
@kurtis-potier-geofabrica have you tried 1-shot mode? (commenting out `#define MAX31865_USE_AUTO_MODE`)

kurtis-potier-geofabrica, 2022-01-08:
Yeah, I've tried both modes to see if that mattered. Didn't change anything for me unfortunately.

zeleps, 2022-01-09:
I got a second sensor and run some tests. It seems that the second sensor does not initialize correctly in auto mode (it reads the correct temperature, but returns the fault bit set, and the fault register returns 64 - low threshold), but it does work after a while in 1-shot mode. I tried a hack that seems to work on my system (run a fault detection cycle during initialization), and now both sensors work ok in any mode under hw / sw SPI (sw SPI still conflicts with SD support). @kurtis-potier-geofabrica could you try this: https://github.com/zeleps/Marlin/commit/5d745e4b16828d2ea13f69e834a3c83d7523882f and see if it works for your setup?

GadgetAngel, 2022-01-09:
I remember seeing this problem before.  

If I am reading this correctly, the current problem is that with SPI and the MAX31865 you get an error bit set when doing TEMP readings. 

I believe the problem is a startup issue between the two modules talking to each other (the MAX31865 module and the Printer's mother board).  The TEMP readings are correct but for some reason the MAX31865 module sets the error bit.  I believe this is a buffering problem on the side of the MAX31865 module.  

Since we can not fix the firmware on the MAX31865 board, I messed around with the idea of doing a short temperature reading dump (5 TEMP readings and dump the values) after the first TEMP read and throw them away if they are determined to be obvious errors generated by the MAX31865 module. 

 I tried to watch the TEMP readings and set a acceptable range that I would expect the TEMP values to fall in.  So If a TEMP readings come in and the next TEMP value is 10 degree higher then this is a obvious temperature error because temperature does not change values instantaneously when you are reading temperature values in milliseconds.

I determined later that Marlin does do a check of temperature readings and I could not get it working.  But I honestly believe the MAX31865 module is setting the error bit and it should not be doing this.  
 
The error bit could also being set due to noise on the SPI lines.  When I did my testing I had bought two Logic Analyer devices so I could look at the actual data bits coming from the MAX31865 modules [Logic Pro 8 (Red) - Saleae 8-Channel Logic Analyzer ](https://www.amazon.com/gp/product/B074TWM5WX) and [DreamSourceLab DSLogic Plus USB-Based Logic Analyzer with 400MHz Sampling Rate ](https://www.amazon.com/gp/product/B08C2LCBGL).

Try dumping or ignoring the first 10 reading coming from the MAX31865 board (ignore the error bits) and then look at the error bit from temperature sample 11 and on.  This might fix the issue.

zeleps, 2022-01-09:
@GadgetAngel, this really clears things up, thanks! I found out all sorts of random things happening on power on, so I'm now running an automatic fault detection cycle and manually set threshold registers (I found out that sometimes they initialize to wrong values, e.g. I had low threshold = 0xffff after power on). This seems to tackle all issues so far and the cost is minimal (less than 1ms), so I'm not making it optional.

I've also implemented the hack you mentioned - just in case - so now you can `#define MAX31865_IGNORE_INITIAL_FAULTY_READS XX` to ignore and clear the fault bit for a limited (XX) number of times. This doesn't affect the temperature reading, so if the reading is not a valid temperature, a temp error will still occur.

@kurtis-potier-geofabrica please try this:

https://github.com/zeleps/Marlin/tree/bugfix-2.0.x-FixMAX31865

kurtis-potier-geofabrica, 2022-01-09:
I will give this a shot. These solutions look like they have a good chance of fixing the problems. My cables are overall pretty solid (running shielded cables the whole way, properly grounded). Even with the current working version, I still see some random spikes and get the occasional error. I don't see this when reading through a Raspberry Pi, so it definitely has something to do with how the MAX31865 and the board talk to each other.

zeleps, 2022-01-09:
Judging from your screenshots, I think that resetting the threshold registers will do the trick for you. Please remember to uncomment MAX31865.h line 44 (`#define DEBUG_MAX31865`).

kurtis-potier-geofabrica, 2022-01-10:
I have (mostly) good news. If I use hardware SPI and:
#define TEMP_SENSOR_FORCE_HW_SPI
#define MAX31865_SENSOR_WIRES_0 2
#define MAX31865_SENSOR_WIRES_1 2

//#define MAX31865_50HZ_FILTER
#define MAX31865_USE_READ_ERROR_DETECTION

//#define MAX31865_USE_AUTO_MODE
#define MAX31865_MIN_SAMPLING_TIME_MSEC 100
#define MAX31865_IGNORE_INITIAL_FAULTY_READS 10

I am able to read both temperatures successfully. The readings seems stable without any random spikes (I'm reading over pronterface) that I would often see before. I am not using the screen at the moment because I have jumpers to those pins (I have to source the device  to splice in that gadget angel was using). I was not able to replicate this success on software spi. I would initially get sort of decent readings (mid 30s at room temp) and then T0 would spike to mid 300s after a few seconds. I tried changing a number of settings to see if different options could get this to change, but the behavior remained the same. I'm going to get my wiring to a little more stable setup and hopefully get back my screen and then try some printing to see how it performs and if I get errors like I occasionally do with 2.0.9.1.

zeleps, 2022-01-10:
Ok, it's obvious that some process is fiddling with the sw SPI. In my case it was SD. The last time you tried disabling SD was in [this comment](https://github.com/MarlinFirmware/Marlin/issues/23439#issuecomment-1006688517), and the result you posted was correct reads (meaning the SPI worked ok) with the fault bit set (which is now corrected by the latest changes I made). Please try this again (sw SPI, disable SDSUPPORT), and if it works we can be certain that SD SPI handling breaks the sw SPI used by the sensor.

I tried scratching the surface of the sw SPI issue on my system (SDSUPPORT + sw SPI for MAX31865), but it's something I haven't had any experience with, and I'm not sure I can help any more with that. So, unless someone else takes this up, I suggest you try the hw SPI with the injection hack I suggested above. I tried it with two sensors yesterday on my system and it works fine.

In order to do it properly, you'll need a couple of 2x5 female IDC connectors and a couple of 2x5 Dupont connectors (2x3 will also do) with male crimps (2x3 crimps). Crimp the IDC connectors at a convenient position on your EXP2 ribbon cable (same orientation as the end connectors):

[image]
[image]

Then, make 2 Dupont connectors as shown (green = MISO, grey = SCK, blue = MOSI):

[image]
[image]

CS pins may be hooked up to any pair of free digital pins.

I suspect you've already figured out the above, but I guess it might be helpful to post them here for people who end up here looking for the same thing.

kurtis-potier-geofabrica, 2022-01-10:
I have those parts on order, should be coming in on Wednesday. I will double check the software spi, but I believe those were the settings I had when I tested (no sdsupport). I understand the hardware spi may be more robust, so I think its best I move to that long term.

zeleps, 2022-01-10:
In that test (disabled SD), your SPI seemed to work ok, but the issue with threshold registers having invalid values remained, so you got normal reads with the fault bit set. It should work flawlessly now.

kurtis-potier-geofabrica, 2022-01-10:
I tried again with the software spi bus. I got the same result as before. This could potentially be board specific. I'm personally fine with using the hardware spi since it seems like the most optimal way to go anyway and we now have it working. If you have other ideas, I'm happy to test them though. Attached is some output from my testing with software spi. 
[image]

zeleps, 2022-01-10:
Can you please try and capture serial output right after boot? It starts emitting as soon as the display goes blank (before boot screen appears) and there's some relevant data logged at that time that would be useful...

kurtis-potier-geofabrica, 2022-01-10:
Yep, I will do this. I have a print running for a couple hours right now with the output going to pronterface and auto report of temps on (and MAX debugging) so I can look and see how the performance is over a print. After, I will convert back and try to log the output again. Will probably have to throw the delay in again on startup since the connection doesn't happen too fast.

zeleps, 2022-01-10:
What I want to see is a line like `MAX31865 Regs: CFG XX|RTD YY|...`, it's a dump of all register values at initialization. Preferably, after a power on, not just a reset. It will probably show why you're receiving faulty values.

I'd also suggest you use AUTO_MODE. There's no sense in 1-shot mode in marlin, just waste of time.

kurtis-potier-geofabrica, 2022-01-11:
got the output needed. Here it is, settings are same as before with the exception of using auto mode. Also, once I converted back to hardware SPI, I used auto mode and that also works fine. 
[image]

zeleps, 2022-01-11:
Thank you so much for all the trouble you went through! Unfortunately, the numbers don't clear up things, in fact it seems that the temperatures read are off the charts at first (1st sensor), then they settle towards room temperature. I added some more debug info, in case of fault. If you could do one last test (I promise I won't ask for more after this) with sw SPI and AUTO_MODE with my latest commit, that would be great (unfortunately, I cannot reproduce the issue you're having with sw SPI on my board).

zeleps, 2022-01-11:
Also, a `M43 I` to dump your pin usage and states would be useful.

kurtis-potier-geofabrica, 2022-01-11:
Added your commits on today and yesterday to the MAX31865.cpp file. My results unfortunately are not different. I still get a high reading on T0 and a slightly high but not terrible reading on T1. I tried M43 I, for some reason, it is not recognized and I get back echo:Unknown command: "M43 I", also tried regular M43. I have no issue trying stuff out. I would like to help get this to a stable state since I plan to work with these sensors a lot more in the future. 
[image]

zeleps, 2022-01-11:
This does not make sense, at least not with what can be made out from the sensor's datasheet. It appears that the sensor is in an invalid state. My only guess is that the sw SPI does not handle signaling as it should in your case, but that's not something I can help with, at least not without a way to reproduce the error, which probably means I need to get the same board as you have. So, since hw SPI is performing well now, I'll leave the sw SPI problem to someone else to tackle.

It would still be worthwhile to look at your pin states, you can enable M43 by uncommenting `#define PINS_DEBUGGING` at the end of Configuration_adv.h. I'm not sure I'll find something interesting, but still it costs little to do this.

Thanks again for the tests, at least we managed to find and correct some of the issues.

kurtis-potier-geofabrica, 2022-01-11:
Got that turned on and went back to the software spi quickly. Here is the output, it is very lengthy:
PIN: PA2  (A2)   M42 P37          HEATER_BED_PIN                         Output = 0
PIN: PA3  (A3)   M42 P38          <unused/unknown>                       Input  = 1
PIN: PA4  (A4)   M42 P39          <unused/unknown>                       Output = 1
PIN: PA5  (A5)   M42 P40          <unused/unknown>                       Alt Function: 5  - SPI1/SPI2
PIN: PA6  (A6)   M42 P41          <unused/unknown>                       Alt Function: 5  - SPI1/SPI2
PIN: PA7  (A7)   M42 P42          <unused/unknown>                       Alt Function: 5  - SPI1/SPI2
PIN: PA8         M42 P16          <unused/unknown>                       Input  = 1
PIN: PA9         M42 P17          <unused/unknown>                       Input  = 1
PIN: PA10        M42 P18          <unused/unknown>                       Input  = 1
PIN: PA11        M42 P19          <unused/unknown>                       Alt Function: 10 - OTG
PIN: PA12        M42 P20          <unused/unknown>                       Alt Function: 10 - OTG
PIN: PA13        M42 P21          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
PIN: PA14        M42 P22          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
PIN: PA15        M42 P23          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
PIN: PB0  (A8)   M42 P43          HEATER_2_PIN                           Output = 0
PIN: PB1  (A9)   M42 P44          HEATER_0_PIN                           Output = 0
PIN: PB2         M42 P5           <unused/unknown>                       Input  = 0
PIN: PB3         M42 P28          TEMP_0_SCK_PIN                         Output = 0
.                                 TEMP_1_SCK_PIN                         Output = 0
PIN: PB4         M42 P29          <unused/unknown>                       Alt Function: 0  - system (misc. I/O)
PIN: PB5         M42 P30          <unused/unknown>                       Input  = 1
PIN: PB6         M42 P31          TEMP_0_MISO_PIN                        Input  = 1
.                                 TEMP_1_MISO_PIN                        Input  = 1
PIN: PB7         M42 P32          E0_DIR_PIN                             Output = 1
PIN: PB8         M42 P33          E0_STEP_PIN                            Output = 0
PIN: PB9         M42 P34          E0_ENABLE_PIN                          Output = 1
PIN: PB10        M42 P6           <unused/unknown>                       Input  = 1
PIN: PB11        M42 P7           <unused/unknown>                       Input  = 1
PIN: PB12        M42 P8           <unused/unknown>                       Input  = 1
PIN: PB13        M42 P9           <unused/unknown>                       Input  = 1
PIN: PB14        M42 P10          <unused/unknown>                       Input  = 1
PIN: PB15        M42 P11          <unused/unknown>                       Input  = 1
PIN: PC0  (A10)  M42 P45          <unused/unknown>   Analog in =     0   Input  = 0
PIN: PC1  (A11)  M42 P46          <unused/unknown>                       Input  = 1
PIN: PC2  (A12)  M42 P47          <unused/unknown>                       Input  = 1
PIN: PC3  (A13)  M42 P48          <unused/unknown>                       Input  = 1
PIN: PC4  (A14)  M42 P49          <unused/unknown>                       Input  = 0
PIN: PC5  (A15)  M42 P50          <unused/unknown>                       Input  = 1
PIN: PC6         M42 P12          <unused/unknown>                       Input  = 1
PIN: PC7         M42 P13          <unused/unknown>                       Input  = 1
PIN: PC8         M42 P14          FAN2_PIN                               Alt Function: 3  - TIM8..11 (probably PWM)
PIN: PC9         M42 P15          <unused/unknown>                       Input  = 0
PIN: PC10        M42 P24          <unused/unknown>                       Input  = 1
PIN: PC11        M42 P25          <unused/unknown>                       Input  = 1
PIN: PC12        M42 P26          E2_CS_PIN                              Output = 1
.                                 TEMP_1_CS_PIN                          Output = 1
PIN: PC13        M42 P0           Y_MIN_PIN                              Input  = 0
.                                 Y_DIAG_PIN                             Input  = 0
.                                 Y_STOP_PIN                             Input  = 0
PIN: PC14        M42 P1           <unused/unknown>                       Input  = 1
PIN: PC15        M42 P2           X_STEP_PIN                             Output = 0
PIN: PD0         M42 P81          E2_DIR_PIN                             Input  = 1
PIN: PD1         M42 P82          E2_STEP_PIN                            Input  = 1
PIN: PD2         M42 P27          E2_ENABLE_PIN                          Input  = 0
PIN: PD3         M42 P83          E2_DIAG_PIN                            Input  = 1
PIN: PD4         M42 P84          TEMP_0_CS_PIN                          Output = 1
PIN: PD5         M42 P85          E1_DIR_PIN                             Output = 0
PIN: PD6         M42 P86          E1_STEP_PIN                            Output = 0
PIN: PD7         M42 P87          E1_ENABLE_PIN                          Output = 1
PIN: PD8         M42 P73          <unused/unknown>                       Alt Function: 7  - USART1..3
PIN: PD9         M42 P74          <unused/unknown>                       Alt Function: 7  - USART1..3
[... cut ...]
PIN: PI7         M42 P139         Z_MAX_PIN                              Input  = 0
.                                 Z4_MIN_PIN                             Input  = 0
PIN: PI8         M42 P114         <unused/unknown>                       Input  = 1
PIN: PI9         M42 P115         <unused/unknown>                       Input  = 1
PIN: PI10        M42 P116         <unused/unknown>                       Input  = 1
PIN: PI11        M42 P117         <unused/unknown>                       Input  = 1

zeleps, 2022-01-11:
The only thing that sticks out is that you're using the same pin for E2_CS and TEMP1_CS:

> PIN: PC12 M42 P26 E2_CS_PIN Output = 1
> . TEMP_1_CS_PIN Output = 1

This shouldn't be a problem, since you're not actually using E2 or TMC driver control, but I guess you could use any other unused pin instead and see if it helps.

Did you ever try using only one of the sensors (and completely unhook the other)?

Anyway, since you're moving to hw SPI and I've run out of ideas, let's leave it at that. Unless someone else has something to add / try.

kurtis-potier-geofabrica, 2022-01-12:
Did not get around to do only the one sensor. Also, agreed, no driver is hooked up to those pins, so not shared.

zeleps, 2022-01-12:
Hey, I just noticed that in this line (62@MAX31865.cpp)

`TERN(TARGET_LPC1768, SPI_QUARTER_SPEED, TERN(ARDUINO_ARCH_STM32, **SPI_CLOCK_DIV4**, 500000)),`

SPI_CLOCK_DIV4 might be too fast for the sensor, can you try SPI_CLOCK_DIV8 and see if it works?

GadgetAngel, 2022-01-13:
> This does not make sense, at least not with what can be made out of the sensor's datasheet. It appears that the sensor is in an invalid state. My only guess is that the sw SPI does not handle signaling as it should in your case, but that's not something I can help with, at least not without a way to reproduce the error, which probably means I need to get the same board as you have. So, since hw SPI is performing well now, I'll leave the sw SPI problem to someone else to tackle.
> 
> It would still be worthwhile to look at your pin states, you can enable M43 by uncommenting `#define PINS_DEBUGGING` at the end of Configuration_adv.h. I'm not sure I'll find something interesting, but still it costs little to do this.
> 
> Thanks again for the tests, at least we managed to find and correct some of the issues.

I have a BTT GTR board I can send to you (if you live in the USA) so you can debug the issue.  Where do you live?

zeleps, 2022-01-13:
Mine's zeleps#8169, I'll dm you later when I'm somewhere more comfortable.

kurtis-potier-geofabrica, 2022-01-13:
I can definitely try this out. My wiring is a little more permanently switched to HW SPI now, but converting back for a test shouldn't be too bad.

zeleps, 2022-01-13:
Technically, you don't have to change your wiring, just declare the hw SPI sck / miso / mosi pins in your config as TEMP_0_SCK_PIN etc. This will start the SPI in sw mode (obviously, you need to comment out FORCE_HW_SPI. If this doesn't work with your current version but works with the latest commit, this is good news.

zeleps, 2022-01-13:
Of course, if it works either way, it would be helpful to try the old wiring with the new code.

kurtis-potier-geofabrica, 2022-01-14:
Had some other things to work on with the printer, but I did get around to testing this today with SW SPI. It did not work. I did convert back to the old physical setup for this as well. Since this works on your board, it could be something unique to my setup. HW SPI is still working well though.

zeleps, 2022-01-15:
Ok, I've ordered a BTT GTR 1.0 to see what's going on, this has intrigued me a lot. It's shipping from China (couldn't find one anywhere else), so it'll take a couple of weeks. I'll get back with results when I get it.

zeleps, 2022-02-03:
Hey all, just got the GTR, tested with @kurtis-potier-geofabrica pin setup (sw SPI @ PD4, PB3, PB6, PG15), works fine with one sensor (TEMP_0) on the first try. This is a bare minimum setup, nothing else on the board but the sensor and a touch TFT (not even stepper drivers). Will try with two sensors later, but I'm not expecting anything less.

zeleps, 2022-02-03:
@kurtis-potier-geofabrica I did a test with your config / pins files unchanged (as you posted them here), but with one sensor connected. The connected sensor worked fine (the other obviously did not). You don't have any jumpers on the stepper UART/SPI selection headers, do you?

This is how I wired the board, do you think there's something I missed or should also try?

[image]

zeleps, 2022-02-03:
Update: 2 sensors also work fine.

kurtis-potier-geofabrica, 2022-02-03:
@zeleps I have the SPI jumpers on and then connect where you would normally connect a stepper driver. I don't believe this is any different to your setup. With this being said, I think its fair to push the changes to main. Since switching to hardware spi on your branch, I really can't say I've had any errors or issues.

GadgetAngel, 2022-02-04:
I am so glad that this is finally working.

PR #23496 (2022-02-05): Fix dual MAX31865 initialization issues
In response to #23439, I added some extra initialization steps to MAX31865.cpp, which ensure no fault exists on init and reset the threshold registers to their default values. This seems to solve a hardware issue, where two MAX31865 sensors are hooked up _on the same SPI_ and they initialize with fault.

I also added another initialization fault bypassing mechanism, as suggested by @GadgetAngel. This can be configured by the corresponding setting (MAX31865_IGNORE_INITIAL_FAULTY_READS) in Configuration_adv.h.

This PR does not close #23439, since there se [... cut]
closed 2022-04-21 by thinkyhead