MarlinFirmware/Marlin #23440: [BUG] (STOP button does not stop/reset the print status)

Dentrax, 2022-01-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Pressing the Stop button stops the printing, but it's impossible to print the same model since the progress/status does not reset. (Which it was working as expected in v1.0.1)

I just upgraded my Marlin from v1.0.1 to latest.

### Expected behavior

It should reset the print status.

### Actual behavior

It does not reset the print status. It seems like I clicked the _pause_ button and wanted to resume the print.

The only workaround to print the same model from the beginning is that rebooting the printer by shutting it down.

### Steps to Reproduce

1. Start a print
2. After some progress complete, press the "Stop" button
3. Try to select the same model again to print and see that progress is not changed

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3 V2

### Electronics

Stock

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

https://user-images.githubusercontent.com/16493751/147976519-2757ef75-6b18-40a3-80e5-0ea57181093c.mp4

thinkyhead, 2022-01-05:
Does it work properly with 2.0.9.2 … or 2.0.9.1 ?

Dentrax, 2022-01-05:
> Does it work properly with 2.0.9.2 … or 2.0.9.1 ?

I directly jumped from `v1.0.1` so I'm not sure. 🤷‍♂️

thinkyhead, 2022-01-05:
Well if you get a chance to try either one out, please report your results. I am quite behind in dealing with issues and submissions at this time, so I will not able to devote any time to in-depth testing this week.

descipher, 2022-01-05:
Is that screen still closed source? If so we can't do anything with it ..

Dentrax, 2022-01-05:
> Is that screen still closed source? If so we can't do anything with it ..

By saying "closed source", you mean `T5UIC1_V20_4页面_191022.BIN` file that included in [DWIN_SET](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2/DWIN_SET)?

The-EG, 2022-01-05:
Which UI is this? It would be one of:
```c++
#define DWIN_CREALITY_LCD             // Creality UI
//#define DWIN_CREALITY_LCD_ENHANCED  // Enhanced UI
//#define DWIN_CREALITY_LCD_JYERSUI
```

with the default in the example being 'Creality UI'.

The-EG, 2022-01-05:
> Is that screen still closed source? If so we can't do anything with it ..

Not exactly, this DWIN is much simpler than others. Assets (images, icons) are stored on the LCD itself and it can be commanded to display them, via serial interface. There are no built in menus, dialogs, etc.

There are currently 4 different UIs for this screen, however, and the UI pictured could be the stock 'CreaityUI' or either of the two derivatives, 'Enhanced' or 'Jyers'. Assuming this is a bug, the fix will need to be performed for the UI in question.

descipher, 2022-01-05:
> Not exactly, this DWIN is much simpler than others. Assets (images, icons) are stored on the LCD itself and it can be commanded to display them, via serial interface. There are no built in menus, [...]

Very good, there are some encrypted ones .. you can't work with them.

mriscoc, 2022-01-06:
> Which UI is this? It would be one of:
> 
> ```c++
> #define DWIN_CREALITY_LCD             // Creality UI
> //#define DWIN_CREALITY_LCD_ENHANCED  // Enhanced UI
> //#define DWIN_CREALITY_LCD_JYERSUI
> ```
> 
> with the default in the example being 'Creality UI'.

Seems to be the `DWIN_CREALITY_LCD` flavor.

thisiskeithb, 2022-01-09:
Fixed in #23481

closed 2022-01-09 by thisiskeithb