MarlinFirmware/Marlin #23441: [BUG] (UI: "Heating" label does not update during the cleaning state)

Dentrax, 2022-01-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I am not sure which condition triggered this UI bug but, the label indicates "E1 Heating" during the cooling down state. Pressing the "Cooldown" button does not affect either the UI or status.

I just upgraded my Marlin from v1.0.1 to latest.

P.S: I used the corresponding configs: https://github.com/MarlinFirmware/Configurations/tree/2.0.9.3/config/examples/Creality/Ender-3%20V2 

### Expected behavior

"Heating" label should be hidden during the cooling down state.

### Actual behavior

"Heating" label is shown during the cooling down state.

### Steps to Reproduce

1. Stop the printing?

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3 V2

### Electronics

Stock

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

https://user-images.githubusercontent.com/16493751/147978019-45436a06-0c5b-4f90-a59f-1f3b9f39c9fb.mp4

thinkyhead, 2022-01-05:
It may simply be leaving the old message up. The only thing the "Cooldown" option in that UI does is call `thermalManager.cooldown()` and nothing else.

Dentrax, 2022-01-05:
So you mean `thermalManager.cooldown()` does not interact with UI?

thinkyhead, 2022-01-05:
```cpp
static void cooldown() {
  zero_fan_speeds();
  disable_all_heaters();
}
```

Dentrax, 2022-01-05:
Hmm, should we add a function something like `refresh_status_ui()`? Would you like any help with this issue? Not making any promises but I'd be interested in looking into it. 🙏

thinkyhead, 2022-01-05:
The method `ui.reset_status()` resets the status message to the default for the current machine state. It should be used by UI implementations when appropriate. It is the direct method to do this, so it doesn't check whether there is an "important" message on the screen which should not be cleared.

thinkyhead, 2022-01-05:
e.g., For `dwin.cpp` line 2762…
```cpp
      #if HAS_HOTEND || HAS_HEATED_BED
        case PREPARE_CASE_COOL:
          thermalManager.cooldown();
          ui.reset_status();
          break;
      #endif
```

Sonicboom247, 2022-01-10:
> e.g., For `dwin.cpp` line 2762…
> 
> ```c++
>       #if HAS_HOTEND || HAS_HEATED_BED
>         case PREPARE_CASE_COOL:
>           thermalManager.cooldown();
>           ui.reset_status();
>           break;
>       #endif
> ```

Question, will this be merged soon? Now seeing similar issue on an Ender 3 V2 and Artillery Hornet. Except both read Bed Cooling the entire time they are printing.

thisiskeithb, 2022-01-10:
There's also the bed heating/cooling status reset bug that may be affecting this: https://github.com/MarlinFirmware/Marlin/pull/23135

Sonicboom247, 2022-01-13:
> > Question, will this be merged soon? Now seeing similar issue on an Ender 3 V2 and Artillery Hornet. Except both read Bed Cooling the entire time they are printing.
> 
> There's also the bed heating/cooling status reset bug that may be affecting this: #23135

seems like the ender 3 is now behaving as it should and is only displaying the file name instead of Bed Cooling., and that's 10 min into a print.  I still need to apply fix to Hornet and test.

Sonicboom247, 2022-01-13:
Don't even recall which printer i was working on last night but one or two of them received an update and now.
[image]
And after bed dropped two degrees it went back to time left.
[image]

Even though time left and remaining are quite off, it syncs then goes off a bit.

thisiskeithb, 2022-04-09:
#23135 has been merged and there's a new option `STATUS_MESSAGE_TIMEOUT_SEC` to timeout low-priority status messages.

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2022-04-21 by thinkyhead