MarlinFirmware/Marlin #23447: [BUG]Ender 3V2 EnhancedUI - LPC1769 compiling error

tome9111991, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello.
with latest Bugfix i got this error in VC on compiling with  EnhancedU on SKR E3 Turbo (lpc1769)

```
Marlin\src\lcd\e3v2\enhanced\dwin.cpp: In function 'void ICON_Button(bool, int, const frame_rect_t&, const text_info_t&, FSTR_P)':
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:298:50: error: cannot convert 'FSTR_P' {aka 'const __FlashStringHelper*'} to 'const char*'
  298 |     const uint16_t x = ico.x + (ico.w - strlen_P(caption)*DWINUI::fontWidth()) / 2,
      |                                                  ^~~~~~~
      |                                                  |
      |                                                  FSTR_P {aka const __FlashStringHelper*}
In file included from C:\Users\Tommi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Print.h:25,
                 from C:\Users\Tommi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Stream.h:27,
                 from C:\Users\Tommi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/HardwareSerial.h:29,
                 from C:\Users\Tommi\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:169,
                 from c:\users\tommi\3d druck\ender 3v2\marlin\marlin\src\hal\shared\marduino.h:36,
                 from c:\users\tommi\3d druck\ender 3v2\marlin\marlin\src\hal\lpc1768\hal.h:39,
                 from c:\users\tommi\3d druck\ender 3v2\marlin\marlin\src\hal\hal.h:30,
                 from Marlin\src\lcd\e3v2\enhanced\../common/../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\e3v2\enhanced\../common/dwin_api.h:24,
                 from Marlin\src\lcd\e3v2\enhanced\dwin_lcd.h:31,
                 from Marlin\src\lcd\e3v2\enhanced\dwinui.h:31,
                 from Marlin\src\lcd\e3v2\enhanced\dwin.h:32,
                 from Marlin\src\lcd\e3v2\enhanced\dwin.cpp:34:
c:\users\tommi\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\string.h:41:17: note:   initializing argument 1 of 'size_t strlen(const char*)'
   41 | size_t  strlen (const char *);
      |                 ^~~~~~~~~~~~
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:300:28: error: 'y' was not declared in this scope
  300 |     DWINUI::Draw_String(x, y, caption);
      |                            ^
```

maybe @mriscoc had an idea what the problem is?

config:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/7807312/config.zip)

### Version of Marlin Firmware

latest bugfix

ellensp, 2022-01-04:
This makes it compile, I cant test it currently.

```DIFF
diff --git a/Marlin/src/lcd/e3v2/enhanced/dwin.cpp b/Marlin/src/lcd/e3v2/enhanced/dwin.cpp
index 3612fb6bd1..6f2c249185 100644
--- a/Marlin/src/lcd/e3v2/enhanced/dwin.cpp
+++ b/Marlin/src/lcd/e3v2/enhanced/dwin.cpp
@@ -295,7 +295,7 @@ void ICON_Button(const bool selected, const int iconid, const frame_rect_t &ico,
     DWIN_Frame_AreaCopy(1, txt.x, txt.y[selected], txt.x + txt.w - 1, txt.y[selected] + txt.h - 1, ico.x + (ico.w - txt.w) / 2, (ico.y + ico.h - 28) - txt.h/2);
   }
   else {
-    const uint16_t x = ico.x + (ico.w - strlen_P(caption)*DWINUI::fontWidth()) / 2,
+    const uint16_t x = ico.x + (ico.w - strlen_P((PGM_P)caption)*DWINUI::fontWidth()) / 2,
                    y = (ico.y + ico.h - 28) - DWINUI::fontHeight() / 2;
     DWINUI::Draw_String(x, y, caption);
   }

```

ellensp, 2022-01-04:
issue was added in https://github.com/MarlinFirmware/Marlin/pull/23240

mriscoc, 2022-01-04:
> maybe @mriscoc had an idea what the problem is?

I only see that problem while compile for STM32 creality_maple environment, with the STM32 creality it is ok.
Thanks @ellensp for fix it.

Maybe the strlen_P definition must be changed:
From:
```cpp
#define strlen_P(a) strlen((a))
```
To:
```cpp
#define strlen_P(s) strlen((const char *)(s))
```

thinkyhead, 2022-01-05:
`__FlashStringHelper*` parameters are converted to `PSTR` via the `FTOP` macro.

closed 2022-01-05 by thinkyhead

tome9111991, 2022-01-05:
I'm trying to learn a little c / c ++. 

the problem was that he couldn't convert the string? 

with FTOP (I think this is a function) is it possible to convert.

 How do I know that I need FTOP for this? is there any documentation?

thinkyhead, 2022-04-21:
> How do I know that I need FTOP for this? is there any documentation?

The code contains many usage examples of `FTOP` and the other string helpers, so just study the code.

commit cc919ae8 (2022-01-05): 🐛 Fix strlen_P parameter error

Fixes #23447