MarlinFirmware/Marlin #23448: [BUG] Unable to merge upstream/bugfix-2.0.x into local branch of bugfix-2.0.x

Taomyn, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I try to merge bugfix-2.0.x into my local branch with Github Desktop it throws an error about an untracked file, .vscode/extensions.json and aborts.

I've been merging bugfix-2.0.x almost daily until this point and now I am stuck.

### Bug Timeline

2nd Jan

### Expected behavior

Clean merge

### Actual behavior

Github Desktop aborts the merge

### Steps to Reproduce

Choose my local branch of bugfix-2.0.x
Attempt to merge upstream/bugfix-2.0.x

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

n/a

### Electronics

n/a

### Add-ons

n/a

### Additional information & file uploads

[image]

[image]

thisiskeithb, 2022-01-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-01-04 by thisiskeithb

Taomyn, 2022-01-04:
@thisiskeithb I would have used Discord as I did in the past but you now require Discord to know my phone number which is something I will not provide - of the other numerous Discords I frequent that all enabled that option, they all removed it after the backlash and loss of members. Nor do I have a Facebook account.

sjasonsmith, 2022-01-04:
If you delete the file locally then it will allow you to pull. I believe this file was previously listed in the .gitignore and is now part of the repo. Git is warning you that you might be overwriting something with important contents in it.

Taomyn, 2022-01-04:
> If you delete the file locally then it will allow you to pull. I believe this file was previously listed in the .gitignore and is now part of the repo. Git is warning you that you might be overwriting something with important contents in it.

Thank for the the useful explanation, I did as you stated and I can compile normally again.