MarlinFirmware/Marlin #23449: [BUG] after 0d91b077 Robin Nano V3 FAN_PIN is always on when hardware PVM is used

compihu, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Starting with 0d91b07797c0d248eab25a64351db959a866bdc7 (seems to be originally introduced as da830e6ced7f7c7e509e748104245064d1c1b265 on bugfix-2.0.x) blowers (connected to PC14/FAN_PIN are alwasy on on my Flsun SR (equipped with MKS Robin Nano v3)
previous commit (b033da1782579d27ed05d9acbf0b2ccb433bdbb8) does not show the same behaviour - blowers start spinning on boot for 1-2 seconds then they go off.
FAN1_PIN (PB1) works as expected
Using FAN_SOFT_PWM also makes PC14 work as expected.

### Bug Timeline

Introduced with 0d91b07797c0d248eab25a64351db959a866bdc7 (da830e6ced7f7c7e509e748104245064d1c1b265)

### Expected behavior

Working hardware PWM also on PC14

### Actual behavior

FAN_PIN (PC14) is always on

### Steps to Reproduce

See the bug description

### Version of Marlin Firmware

2.0.9.3

### Printer model

Flsun SR

### Electronics

MKS Robin Nano V3

### Additional information & file uploads

[Congiguration.zip](https://github.com/MarlinFirmware/Marlin/files/7807627/Congiguration.zip)

descipher, 2022-01-04:
This is not a bug.
PC14 is not a PWM pin therefore FAN_SOFT_PWM is the only way to achieve variable speed with that pin.
Ref: https://www.mouser.com/datasheet/2/389/stm32f405rg-956214.pdf
Page 47

sjasonsmith, 2022-01-04:
It is possible the FAN_SOFT_PWM was previously force-enabled in the pins file. I know I’ve recently seen mention of a build error from a pins file if not enabled, maybe this board needs that added.

descipher, 2022-01-04:
Yes, that's possible and would explain a change in behavior, however I did check for it and there was no overrides.

thinkyhead, 2022-01-05:
Pins files that definitely need `FAN_SOFT_PWM` should be defining `FAN_SOFT_PWM_REQUIRED`, which will prevent compilation with this sanity check:
```cpp
#if HAS_FAN0
  #if CONTROLLER_FAN_PIN == FAN_PIN
    #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."
  #elif ENABLED(FAN_SOFT_PWM_REQUIRED) && DISABLED(FAN_SOFT_PWM)
    #error "FAN_SOFT_PWM is required. Enable it to continue."
  #endif
#endif
```

closed 2022-03-18 by github-actions[bot]