MarlinFirmware/Marlin #23450: [BUG] SKR 1.4 Turbo TMC2209 Axis Moves Rough but fine when steps per mm is doubled

Russ79, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi Everyone,
I've been scratching my head with this a few days, appreciate your help

I have a custom built corexy printer 
BTT SKR 1.4 Turbo with BTT TMC2209 drivers in uart mode

When i have the steps per mm set correctly so the machine moves 10mm on a 10mm command, the x/y axis move very rough and noisy and can barley move 20mm/s.  I tried tons of different speeds, microstep settings, square stepping, linear advance, normal marlin/bugfix marlin, stealth chop on and off, and always the same outcome,

But when i set steps per mm to double the correct value so the machine moves 20mm on a 10mm command, the axis move perfectly smooth and can reach speeds of 150mm/s with no issues.  

This is repeatable with any microstep setting its always rough with the right scaling and smooth when doubled.

and when i say its rough, the machine wants to shake itself apart.

Please let me know if anymore information is needed to help 

### Bug Timeline

new

### Version of Marlin Firmware

2

ellensp, 2022-01-04:
Video?
Config files?
Steps/mm in question?
M122 output ?
Stepper motors details?

thinkyhead, 2022-01-05:
Does M502, M500, reboot make any difference?

borland1, 2022-01-06:
Agree, you probably just need to initialize your EEPROM settings with firmware defaults.

Very hard to judge why with no configuration files to review.

Description is not clear. Did you upload revised firmware or use LCD menu to double the step/mm?

`#define HYBRID_THRESHOLD`, if used, you may need to set x/y stealth-chop transition values higher in `Configuration_adv.h`.

closed 2022-03-24 by github-actions[bot]