MarlinFirmware/Marlin #23452: [BUG] M569 is not working if EXTRUDERS is set to 0

tbnobody, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In my configuration I set extruders to 0
`#define EXTRUDERS 0`

If I execute `M569 S0 X Y` I get the following error message:
```
M569 S0 X Y
echo:M569 Invalid E stepper 0
```

### Version of Marlin Firmware

2.0.9.3

GMagician, 2022-01-04:
Not given config fiels but what about E0_DRIVER_TYPE value?

GMagician, 2022-01-04:
... or any E_DUAL_STEPPER_DRIVERS, MIXING_EXTRUDER, MMU?

tbnobody, 2022-01-04:
Oh Sorry...
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7810491/Configuration.zip)

GMagician, 2022-01-04:
I used your cnf with some little mod for my board and it works. What do you get with just 'M569'?

tbnobody, 2022-01-04:
Currently I get 
```
X driver mode:  spreadCycle
X2 driver mode: spreadCycle
Y driver mode:  spreadCycle
Y2 driver mode: spreadCycle
Z driver mode:  spreadCycle
```

Btw. the correct spreadCycle mode is getting set, but I get the error message as shown above

GMagician, 2022-01-04:
Ok, got it. You are not using updated sources. It has been fixed 5 days ago

tbnobody, 2022-01-04:
Ohhh Sry. I forgot to fetch before I rebased my config onto the bugfix release. Sorry for this inconvenience.

closed 2022-01-04 by tbnobody