MarlinFirmware/Marlin #23454: [BUG] 2.0.9.3 causes loss of stepper motor control/power

ansonl, 2022-01-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling a previously working configuration with release 2.0.9.3 and the bugfix-2.0.x on Ultimaker Original Plus (with Ultimaker 2+ extrusion and LCD upgrade), stepper motors have jerky, weak movement and are unusable. 

This affects extruder gantry and feeder motors. When homing the stepper motors will barely have enough power/steps to move the extruder head slowly and will sometimes stall in one place resulting in a homing failure (reset). the feeder motor will grind against itself and not move if under tension resistancewith the filament.

I have tested the below configuration files (just need to change the version number) with release 2.0.9.2 and 2.0.9.1 and the printer works fine with the previous release. Only starting with 2.0.9.3 does the stepper motor issue occur. I am using stock [Ulticontroller v2](https://github.com/Ultimaker/Ultimaker2/tree/master/1249_Ulticontroller_Board_(x1)) and UMO+ motors on the printer. For example, the feeder motor model is [SY42STH38-1684A](https://www.pololu.com/file/0J714/SY42STH38-1684A.pdf) in the attached spec sheet and the exact same motor performs normally when reverted to Marlin 2.0.9.2.

I recorded a video clip of the printer autohoming on both 2.0.9.2 and 2.0.9.3 so you can see the visible and audible differences.
Marlin 2.0.9.2 - https://youtu.be/vopTKNTfs9E
Marlin 2.0.9.3 - https://youtu.be/rIAoPvyVy4g

### Bug Timeline

New bug in 2.0.9.3

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ultimaker Original+

### Electronics

Ulticontroller v2

### Add-ons

Ultimaker 2+ extrusion and LCD upgrade

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7811066/Configuration.zip)
[SY42STH38-1684A.pdf](https://github.com/MarlinFirmware/Marlin/files/7811039/SY42STH38-1684A.pdf)

thinkyhead, 2022-01-05:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

ellensp, 2022-01-05:
I wonder if this is related to that controllers method of setting stepper driver current
ie
```
#define MOTOR_CURRENT_PWM_XY_PIN              44
#define MOTOR_CURRENT_PWM_Z_PIN               45
#define MOTOR_CURRENT_PWM_E_PIN               46
// Motor current PWM conversion, PWM value = MotorCurrentSetting * 255 / range
#ifndef MOTOR_CURRENT_PWM_RANGE
  #define MOTOR_CURRENT_PWM_RANGE            2000
#endif
#define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}

```
Perhaps this no longer functions after recent pwn changes?

thinkyhead, 2022-01-05:
So… you are suggesting this is a case where pins are sharing a PWM timing, but due to changes related to `FAST_PWM_FAN` the PWM timing is now being set differently. To wit …
```cpp
#define FAST_PWM_FAN      // Increase the fan PWM frequency. Removes the PWM noise but increases heating in the FET/Arduino
#if ENABLED(FAST_PWM_FAN)
  //#define FAST_PWM_FAN_FREQUENCY 31400  // Define here to override the defaults below
  //#define USE_OCR2A_AS_TOP
  #ifndef FAST_PWM_FAN_FREQUENCY
    #ifdef __AVR__
      #define FAST_PWM_FAN_FREQUENCY ((F_CPU) / (2 * 255 * 1))
    #else
      #define FAST_PWM_FAN_FREQUENCY 1000U
    #endif
  #endif
#endif
```

Looking at #23331 the only change was to relocate `FAST_PWM_FAN_FREQUENCY` to the configuration, but the logic is the same. Before that was #23326 which changed non-AVR to use 1000U by default. But, this issue is about an AVR board, so that should not be different from 2.0.9.1.

I see also one small change to the `ULTIMAIN_2` pins from #23194 on Nov. 27 by @ansonl…
```diff
- #define E0_AUTO_FAN_PIN                     77
+ #define E0_AUTO_FAN_PIN                     69
```

…but I don't see any overlaps with that pin. However, for those upgrading from older Marlin and using this default pin there is bound to be some effect.

descipher, 2022-01-05:
What connectors are the part and extruder cooling fan on?
#23194 refers to UM2 v2.1.4 however you have v2.1.1 and PIN 69 is not attached to anything on v2.1.1.

descipher, 2022-01-05:
> I wonder if this is related to that controllers method of setting stepper driver current ie
> 
> ```c++
> #define MOTOR_CURRENT_PWM_XY_PIN              44
> #define MOTOR_CURRENT_PWM_Z_PIN               45
> #define MOTOR_CURRENT_PWM_E_PIN               46
> // Motor current PWM conversion, PWM value = MotorCurrentSetting * 255 / range
> #ifndef MOTOR_CURRENT_PWM_RANGE
>   #define MOTOR_CURRENT_PWM_RANGE            2000
> #endif
> #define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}
> ```
> 
> Perhaps this no longer functions after recent pwm changes?

I think you may be onto a possibility. The pins_ULTIMAKER_2 PWM sets are not matching the schematics connections.

PL5 40 -> X Vref 
PL4 39 -> Z Vref
PL3 38 -> E Vref
Y = none

The schematic would require the following mapping:

#define MOTOR_CURRENT_PWM_XY_PIN              40
#define MOTOR_CURRENT_PWM_Z_PIN               39
#define MOTOR_CURRENT_PWM_E_PIN               38

ansonl, 2022-01-05:
@thinkyhead, all, I tested past bugfix commits and narrowed the breaking change down to #23048 

The previous commit #23050 is working and starting with #23048, the stepper motor issue occurs.

ansonl, 2022-01-05:
> What connectors are the part and extruder cooling fan on? #23194 refers to UM2 v2.1.4 however you have v2.1.1 and PIN 69 is not attached to anything on v2.1.1.

The extruder cooling fan is attached to PJ6 in the Ultimaker 2+. I actually have Ultimainboard v2.1.4 that came with the Ultimaker Original+ and is shared by UMO+, UM2, and UM2+ models. From what I have read, the v2.1.1 controller board differs in that the part cooling fan is always on versus controllable in the v2.1.4. [UM2+ extrusion upgrade on UMO+ overview](https://www.youmagine.com/designs/um2-upgrade-on-umo-assembly-manual)

ansonl, 2022-01-05:
@descipher 

> #define MOTOR_CURRENT_PWM_XY_PIN 40 #define MOTOR_CURRENT_PWM_Z_PIN 39 #define MOTOR_CURRENT_PWM_E_PIN 38

I tried setting the ultimain2 board pins as you listed for Marlin project state at #23048 and checked the schematic as well. I get the same stepper motor issues with the pin changes.

descipher, 2022-01-05:
Be sure to load defaults in the EEPPROM firmware if not already done.
What about SOFT_FAN_PWM?

descipher, 2022-01-05:
I believe the issue is related to frequency. The call to set duty for the motor current pin was previously analogwrite() which always defaults to 1000HZ. There is a comment in the area that claims to set the prescaler to 1 but I'm not certain that's working. To isolate this issue please edit the following line, compile and test.

[src/module/stepper.cpp
](https://github.com/MarlinFirmware/Marlin/blob/da830e6ced7f7c7e509e748104245064d1c1b265/Marlin/src/module/stepper.cpp#L3256)
```
-        #define _WRITE_CURRENT_PWM(P) set_pwm_duty(pin_t(MOTOR_CURRENT_PWM_## P ##_PIN), 255L * current / (MOTOR_CURRENT_PWM_RANGE))-        
+        #define _WRITE_CURRENT_PWM(P) analogWrite(pin_t(MOTOR_CURRENT_PWM_## P ##_PIN), 255L * current / (MOTOR_CURRENT_PWM_RANGE))
```

ansonl, 2022-01-06:
@descipher @thinkyhead 
Replacing set_pwm_duty and analogWrite in stepper.cpp as suggested solved the problem.

I took another look at the pin mapping for the Ultimain2 board and the original pin values are actually correct. 

```
#define MOTOR_CURRENT_PWM_XY_PIN              44
#define MOTOR_CURRENT_PWM_Z_PIN               45
#define MOTOR_CURRENT_PWM_E_PIN               46
```

[image]

The Arduino pin mapping assignment is different from the Atmega2560 pin number for the pins in question. The PJ6 pin mapping that I added in a past pull request does not have an Arduino pin equivalent and I have directly activated PJ6 inside previous contributed code to temp.cpp so I can fix this in my next contribution (ready to push) that improves upon the Ultimaker 2 LCD controller functionality.

descipher, 2022-01-06:
Very good! Right I should have realized the schematic is not using the Arduino pin numbering abstraction. I will put the 2560 on the scope and see what our actual frequency output is and then come up with a proper solution.

descipher, 2022-01-06:
Ok, there is actually three issues here.

1) The AVR HAL set_pwm_duty call does not set a frequency if it was not previously initialized. So the duty is set but no timer count will run.
2) There is no Marlin HAL PWM_FREQUENCY default defined to use in 1)
3) The MOTOR_CURRENT_PWM has no default to use but implies 31400HZ via 
SET_CS5(PRESCALER_1) in the stepper code however this may or may not set correctly depending on the clock frequency.

PR to resolve these issues is now started.

descipher, 2022-01-06:
@ansonl R you good for testing this?

ansonl, 2022-01-06:
@descipher Just tested the commit you posted.

The stepper motors move normally now, however there is a loud distinct whine whenever each of the stepper motor axes is powered on even when stopped. Each X Y Z stepper motor makes a buzzing whine when any of them are powered on individually and the sound goes away when "disable steppers" is selected in the menu. I attached video of homing and the sound coming from the controller board. In the video you get to hear 1-2-3 whining sounds overlayed on each other. 

https://youtu.be/Nqu8vplyqZs

ETE-Design, 2022-01-06:
@ansonl Have had exactly that problem on my MKS Robin Lite Board, had to experiment with:
`// Motor current PWM pins
#define MOTOR_CURRENT_PWM_XY_PIN            PB0
#define MOTOR_CURRENT_PWM_Z_PIN             PA7
#define MOTOR_CURRENT_PWM_E_PIN             PA6
#define MOTOR_CURRENT_PWM_RANGE (65535/10/3.3) // (255 * (1000mA / 65535)) * 257 = 1000 is equal 1.6v Vref in turn equal 1Amp
#define DEFAULT_PWM_MOTOR_CURRENT { 1000, 1000, 1000 } // 1.05Amp per driver, here is XY, Z and E. This values determined empirically.`

To get it working... Asked on the Discord also, but seems like not many know how to calculate the values needed... So had to guess, no one could explain how to find "CURRENT_PWM_RANGE" (65535/10/3.3)
@descipher Mabye you can explain the process?

descipher, 2022-01-06:
Done. Please retry.

ansonl, 2022-01-06:
@descipher Tried the new frequency. Same symptoms and sound.

descipher, 2022-01-06:
> @descipher Mabye you can explain the process? Both DEFAULT_PWM_MOTOR_CURRENT and MOTOR_CURRENT_PWM_RANGE would be great...

Each board will have its own RC charge pump design so those calcs will not apply to the ULTIMAIN2 board.
Basically the circuit has a fixed voltage range and we can adjust the voltage level using the PWM duty cycle. It usually directly maps.
```
// Motor current PWM conversion, PWM value = MotorCurrentSetting * 255 / range
#ifndef MOTOR_CURRENT_PWM_RANGE
  #define MOTOR_CURRENT_PWM_RANGE           2000
#endif
#define DEFAULT_PWM_MOTOR_CURRENT  {1300, 1300, 1250}
```
The circuit formula is shown, 1300 is in mA 1300 * 255 / 2000 = the PWM duty required.

descipher, 2022-01-06:
I think something else is happening, the scope reports 62.5% @ 31kHz using Ramps/ATMega2560 1.4 board.

ETE-Design, 2022-01-06:
@descipher Anyway to contact you directly, cause my question is not related to the Ultimaker 2, but another printer... And would like to do the calculations correct instead of guessing... Have board Schematic + Stepper Specs... So mabye I could ask your expertice?

ETE-Design, 2022-01-06:
> Each board will have its own RC charge pump design so those calcs will not apply to the ULTIMAIN2 board. Basically the circuit has a fixed voltage range and we can adjust the voltage level using the [...]

But how to find / calculate MOTOR_CURRENT_PWM_RANGE?

descipher, 2022-01-06:
Ok, found the issue, it's a conditional. NEEDS_HARDWARE_PWM does not get set. Should have it does later tonight.

descipher, 2022-01-06:
> But how to find / calculate MOTOR_CURRENT_PWM_RANGE?

The range is determined by the designer of the charge pump circuit.

ETE-Design, 2022-01-06:
> 

Can I find that part in the Schematic for the Board, and where to look at it then? BTW have send you an mail....

ETE-Design, 2022-01-06:
> > But how to find / calculate MOTOR_CURRENT_PWM_RANGE?
> 
> The range is determined by the designer of the charge pump circuit.

So if it looks like this it is not set right?
`#define MOTOR_CURRENT_PWM_RANGE (65535/10/3.3)`

Can I find that part in the Schematic for the Board, and where to look at it then? BTW have send you an mail....

descipher, 2022-01-07:
Please retest.

descipher, 2022-01-07:
@ETE-Design 

This may help you see how it works.

This is the Ultimate v2.1.4
[image]

This is a circuit sim of it.
[image]

You can play with the values to see how it is calculated

https://www.falstad.com/circuit/

[pwm-current-cp.txt](https://github.com/MarlinFirmware/Marlin/files/7831688/pwm-current-cp.txt)

ansonl, 2022-01-07:
@descipher Tested the latest change. The stepper motors appear to move normally but I haven't tested a print on it just yet. 
However while the noise when steppers are powered on is less noticeable, it is still noticeably louder than before (tested on 2.0.9.2). I used my phone to measure dB a few inches from the controller board with steppers powered on and your patch emits about 50dB while the previous 2.0.9.2 code emits a 46dB whine.

descipher, 2022-01-07:
This is the PWM output, this will not be audible. Perhaps there are other reasons, like the PWM duty previously was less accurate and did not drive the motors with the calculated power of  X 1300mA,Y 1300mA, Z 1250mA.

Notwithstanding that,  without the actual device there is no way for me to resolve it further other than what it have done here by validating the required output for the design.

[image]

ansonl, 2022-01-07:
I previously tested with 1200mA, 1200mA, 1250mA and tried again with  X 1300mA,Y 1300mA, Z 1250mA. The noise decreased 1dB on both 2.0.9.2 and your patch. The latest current values for the Ultimaker should be 1200mA, 1200mA, 1250mA based on this [commit from Ultimaker](https://github.com/Ultimaker/UltimakerMarlin/commit/96f534aaf3339c8f2ac797f000d5ae4b5ded2a1b#diff-c7a3fa925098d81e51840820898b66e3d3e48f121a83e4d0b6c65e5ab4f2fba0). 

Does this mean that the frequency code needs to be re-calculated for every future stepper driver power change with the new code versus the previous Arduino `analogWrite` code? If so it's not very accessible for users who want to experiment with different currents? 
Maybe we could have a conditional statement that utilizes `analogWrite` only for AVR/RAMPS until a further solution is found.

descipher, 2022-01-08:
I don't think that applies to the ULTIMAIN_2.
Olliver Schinagl committed on Jul 16, 2015 
[image]

ansonl, 2022-01-08:
The altium file may have been opened and saved at a later date, but the ULTIMAIN_2 v2.1.4 was last revised in 2014 on the last schematic page listing revisions.
[image]

ETE-Design, 2022-01-08:
@descipher So this is the part I'm looking for?
[image]

Values is this:
R43/44/45/46/47/48 = 1K
C34/35/36/37/38/39 = 1uF

How to find:

`#define MOTOR_CURRENT_PWM_RANGE`

descipher, 2022-01-08:
I have searched all of 1.1.x in the upstream, I do not see any change to the default value in pins_ULTIMAIN_2. So maybe it was never updated by Ultimaker, you could certainly use the lower current setting they used in 1.x.x. It will be a little slower.
As far as PWM duty change, I can check what it was actually outputting in 2.0.9.2. It maybe the same output.

descipher, 2022-01-08:
> `#define MOTOR_CURRENT_PWM_RANGE`

What driver is it?

ETE-Design, 2022-01-08:
> > `#define MOTOR_CURRENT_PWM_RANGE`
> 
> What driver is it?

HR4988

descipher, 2022-01-08:
> HR4988
Is there a circuit diagram of the driver, because that circuit would have a maximum duty of 30 to reach 1300mA using a vRef of 1.385v which is unusual.

ETE-Design, 2022-01-08:
@descipher Like this
[image]
?

descipher, 2022-01-08:
3.3 v makes more sense now.

descipher, 2022-01-08:
@ETE-Design 

I can't read Chinese, however the typical Polulu 4988 clone has a max peek current of 2A if you have a good head sink you could run 1.3A max but to be safe I would use 2400 as the range. that will be a vRef of about 1.75v which is typical and should result in about 1.1A max 
If it runs cool you could decrement the range -100 and check the temps after a 15min print.

descipher, 2022-01-08:
2.0.9.2 has the identical PWM duty, thus it not the reason for any audible change.

[image]

ansonl, 2022-01-10:
I was unable to consistently reproduce the increased whine volume so I would consider this fixed for now after incorporating #23463 . Thanks.

Grenite, 2022-01-15:
Shouldn't the logic levels of PL3,PL4,PL5 be 5V at high? Some math shows that VREF max (100% duty cycle) with a 5V input is 1.16V-suggesting that the theoretical maximum current of the stepper driver can go as high as 2.9A which is absurdly high. A 3.3V input gives a VREF maximum 0.765V which gives a much more reasonable max current of 1.9125A. But at least according to the schematic, the MEGA2560 on the UltimainBoard 2.1.4 is running at 5V-unless I'm missing some magic where it can output 3.3V logic while operating at 5V.

descipher, 2022-01-15:
Correct for the 2560, but the user who posted the question on calc’s is not using the 2560 his board is 3.3v.

Grenite, 2022-01-16:
That makes sense now. But it now appears there is a mismatch on the max current range defined in Marlin vs the hardware definition (by the resistors on the Vref and Sense pins) of 2A vs 2.9A. This would cause the drivers to output more current than what the controller thinks (Which would explain why I noticed the steppers are running hotter after I started building Marlin from source over using the Ultimaker's default builds).

descipher, 2022-01-16:
Are you aware that ultimaker reduced the current to 1200mA in their stock builds. They originally used 1300mA. Also the 3.3v guy has a completely different circuit design, his question was off topic. Ultimaker has not updated the Marlin upstream, or at least I have not found one. There is nothing wrong with running faster, just need slightly better cooling than what is on the stock board.

Grenite, 2022-01-16:
I was also going off topic and I didn't intend to refer to the 3.3v guy in anyway whatsoever in my previous comment. Yes I was aware that 1200mA was what was originally used, but anecdotally even when I set the 1200mA current in the V2 firmware builds I noticed the steppers were running hotter than before-quite toasty to the touch. I even experimented as going low as ~800mA on the XY steppers to get the motors to run as cool as before.

descipher, 2022-01-16:
Interesting, do you have the most current bug fix-2.0.x? I can certainly recheck those PWM waveforms. What was the version of stock firmware that you observed cooler running A4988’s?

Grenite, 2022-01-16:
I'm currently using a 2.0.9.2 (non-bugfix branch) build of Marlin for the ultimain2 board in question. The stock firmware I was referring to is the one for UMO+ [https://github.com/Ultimaker/Marlin/tree/Marlin_UM_Original_Plus](url). I'm quite sure that the temperature differences I'm talking about is a symptom of a different issue altogether, it's just that your earlier discussions/posts here confirms my suspicions of the temperature differences.

descipher, 2022-02-04:
@thisiskeithb This one can be closed. :)

ellensp, 2022-02-04:
closing @descipher request

closed 2022-02-04 by ellensp