MarlinFirmware/Marlin #23458: [BUG] LVL_BED_CORNERS missing from Marling menu

Sonicboom247, 2022-01-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

latest bugfix 2.0.x from yesterday seems to have disabled/broken Level Bed Corners.  Now Level Corners is replaced with Bed Tramming even though LVL_BED_CORNERS is defined and bed tramming is undefined

### Bug Timeline

started Yesterday with Bugfix 2.0

### Expected behavior

Level Bed Corners to be displayed on the Menu

### Actual behavior

Level Bed corners is no longer available but Bed tramming is

### Steps to Reproduce

In Marlin Mode, select Motion bed leveling 

### Version of Marlin Firmware

2.0.9.3 

### Printer model

Artillery Hornet

### Electronics

Stock Artillery Ruby board and display

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

#define LCD_BED_LEVELING    //SonicBoom

#if ENABLED(LCD_BED_LEVELING)
  #define MESH_EDIT_Z_STEP  0.002 // (mm) Step size while manually probing Z axis.    //SonicBoom 1/4
  #define LCD_PROBE_Z_RANGE 4     // (mm) Z Range centered on Z_MIN_POS for LCD Z adjustment
  
// Add a menu item to move between bed corners for manual bed adjustment
#define LEVEL_BED_CORNERS    //SonicBoom

#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET_LFRB { 20, 20, 20, 20 } // (mm) Left, Front, Right, Back insets    //SonicBoom 1/4
  #define LEVEL_CORNERS_HEIGHT      0.0   // (mm) Z height of nozzle at leveling points
  #define LEVEL_CORNERS_Z_HOP       4.0   // (mm) Z height of nozzle between leveling points
  //#define LEVEL_CENTER_TOO              // Move to the center after the last 
   
  #define LEVEL_CORNERS_LEVELING_ORDER {RF,LF,LB,RB}    //SonicBoom 1/4

//#define ASSISTED_TRAMMING
#if ENABLED(ASSISTED_TRAMMING)

magichappens89, 2022-01-28:
I have the exact same problem. Any update on this?

ellensp, 2022-01-28:
No printer model named, so cannot lookup this display
"Stock Artillery Ruby board and display" does not result in any images or details of the display.
No configuration files provides so cannot look up the display that way either.

ellensp, 2022-01-28:
FYSETC_MINI_12864_X_X  according to Marlin example configs for the Artillery Hornet. 
But seriously, we should have to waste time searching for this information.. 

Attach your configs, as requested!

magichappens89, 2022-01-28:
I have the same issue but another printer. The board is a GT2560 A+. Here my config file: https://ctxt.io/2/AABgGQ80Fg

ellensp, 2022-01-28:
Can one of you try changing this one line and verify it is the issue you are talking about?

ellensp, 2022-01-28:
there is a second on in Marlin/src/lcd/menu/menu_bed_leveling.cpp

```CPP
  #if ENABLED(LEVEL_BED_CORNERS)
    SUBMENU(MSG_BED_TRAMMING, _lcd_level_bed_corners);
  #endif
```

Should be

```CPP
  #if ENABLED(LEVEL_BED_CORNERS)
    SUBMENU(MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
  #endif
```

magichappens89, 2022-01-28:
Thanks for the reply. Unfortunately I can´t compile after this change.
```
In file included from Marlin\src\lcd\menu\menu_motion.cpp:33:0:
Marlin\src\lcd\menu\menu_motion.cpp: In function 'void menu_motion()':
Marlin\src\lcd\menu\menu_motion.cpp:465:13: error: 'MSG_LEVEL_CORNERS' is not a member of 'Language_en'
     SUBMENU(MSG_LEVEL_CORNERS, _lcd_level_bed_corners);
```

ellensp, 2022-01-28:
Found where this was done, and it was done deliberately by thinkyhead...
https://github.com/MarlinFirmware/Marlin/commit/b1bc2e80036131cde58b4fbc02b76ac08b5a7924

So basically Level Corners => Bed Tramming

closed 2022-01-28 by ellensp

magichappens89, 2022-01-28:
So the feature was actually renamed?

Sonicboom247, 2022-01-28:
@ellensp my apologies for not posting the actual config, i figured an excerpt of the actual issue would be enough.  also, yes you can select Bed tramming and it performs the same as Level corners, just odd that with Level corners enabled and bed tramming disabled, the menu still performs corner leveling aka Bed tramming