MarlinFirmware/Marlin #23459: [BUG] Cooling fan full speed after 2.0.9.3 update

marlinfwgeek, 2022-01-05:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

After updating to 2.0.9.3, the cooling fan on both my Anet A8 and Anet A8 Plus run at full speed after power on, even though '0%' is displayed for fan speed. If I move the cooling fan speed setting to 100% and then back to 0%, the fan then turns off but will go back to 100% if the printer is power cycled.

### Bug Timeline

First started with updating to 2.0.9.3 - was not an issue going back to 2.0.7 or later

### Expected behavior

Cooling fan should be off at power on and/or when setting is set to 0%

### Actual behavior

Print cooling fan runs at 100% even though the setting is set to 0%

### Steps to Reproduce

Power up the printer and the print cooling fan runs at 100%

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8 and Anet A8 Plus

### Electronics

Stock electronics

### Add-ons

Anet A8 full graphics display

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

descipher, 2022-01-05:
Please attach configuration files.

marlinfwgeek, 2022-01-05:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7816835/Marlin.zip)

thisiskeithb, 2022-01-05:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23418

closed 2022-01-05 by thisiskeithb