MarlinFirmware/Marlin #23460: [BUG] Homing X failed with 2.0.9.3 but switch seems to be ok

fiechr, 2022-01-05:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

My Anycubic i3 Mega S gives me a homing failed error, when starting to home X or auto home. I've  manually reset/re-init the EEPROM, but that didn't help.

Printer moves hotend towards the switch, until the switch clicks - but then the printer is halted instead of successfully homing.

Y and Z homing works.

M43 pin debugging also shows that all switches work fine (including X min switch).

Same configuration worked fine with 2.0.9.2 (and below).

### Bug Timeline

Worked fine up to including 2.0.9.2

### Expected behavior

Printer homes X axis without issue.

### Actual behavior

Printer halts, need to reset.

### Steps to Reproduce

Try to start homing X or start auto home.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anycubic i3 Mega S

### Electronics

Stock board

### Add-ons

Fysetc display (FYSETC_MINI_12864_1_2), BLTouch (probe and Z endstop)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

fiechr, 2022-01-05:
These are my configuration files.

[MarlinConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/7817370/MarlinConfiguration.zip)

descipher, 2022-01-05:
Does comment out  #define DETECT_BROKEN_ENDSTOP change anything?

descipher, 2022-01-05:
Scott put up a patch related to X-axis, maybe pull the latest and retry.

fiechr, 2022-01-06:
Commenting out DETECT_BROKEN_ENDSTOP does change something, but in a weird way:

Now X and Y homing seems to work, but when homing Z, the Z axis only travels a bit upwards and then the BLTouch is activated (since it also acts as Z endstop), but the bed is out of reach, so I get a BLTouch error now and the printer it halted.

I've just re-flashed 2.9.0.2 again to confirm that everything really worked with that release and yes, all axis home and move as expected.

fiechr, 2022-01-06:
Between 2.0.9.2 and 2.0.9.3 there were some changes in pins_TRIGORILLA_14.h. Before I had added these lines to the file to make it work (I removed the stock display and daughter PCB and therefore had to re-wire some pins).

```
[...]

#define Z2_MIN_PIN                            Z_MAX_PIN // Blue/black wire
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                       X_MAX_PIN // Green wire
#endif
#define CONTROLLER_FAN_PIN                    TG_FAN1_PIN

#include "pins_RAMPS.h"

[...]
```

Starting with 2.0.9.3, I got some build errors because of missing definition of TG_FAN1_PIN, if I remember correctly. So I decided to set two of these values in the main configuration files to the actual pin numbers (see config above). Now only

```
[...]
#define Z2_MIN_PIN                            Z_MAX_PIN  // Blue/black wire

#include "pins_RAMPS.h"
[...]
```
remains in the pins_TROGORILLA_14.h.

But Z_MIN and Z2_MIN aren't used anyway anymore and the controller fan seems to work fine.

So at the moment, I'm unsure if this is my own fault because of wrong configuration, that for some reason, worked fine until now (will recheck this!) or if this is really a bug.

My Ender 3 Pro and Sidewinder X1 work fine with 2.0.9.3.

[image]

descipher, 2022-01-06:
Seem wrong to me, that board map to this info:
https://user-images.githubusercontent.com/3864712/76686181-ecb93e00-6619-11ea-8255-a60dd31f4e53.png
So it's a ramps pin equiv. for basic endstops.
The stock Mega S has multi z endstops. Did you remove one? Because it's commented out.   //#define Z_MULTI_ENDSTOPS
That may be the issue.

```
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  // Enable if Z motor direction signals are the opposite of Z1
  //#define INVERT_Z2_VS_Z_DIR
  //#define INVERT_Z3_VS_Z_DIR
  //#define INVERT_Z4_VS_Z_DIR

  //#define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _ZMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif
```

descipher, 2022-01-06:
You can reference the config repo:
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/AnyCubic/i3%20Mega

descipher, 2022-01-06:
```
// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING

/**
 * Z_MIN_PROBE_PIN
 *
 * Define this pin if the probe is not connected to Z_MIN_PIN.
 * If not defined the default pin for the selected MOTHERBOARD
 * will be used. Most of the time the default is what you want.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 */
#define Z_MIN_PROBE_PIN 43 // Pin 32 is the RAMPS default
```
Pin 43 is on AUX 2 but the probe is shown on X_MAX_PIN

fiechr, 2022-01-06:
Yes, I removed both Z endstops and use the BLTouch as (only) Z endstop.
I'm also using Z_STEPPER_ALIGN_XY.

descipher, 2022-01-06:
I would think this should be set in Configuration.h 
#define Z_MIN_PROBE_PIN X_MAX_PIN

descipher, 2022-01-06:
You are welcome, don't forget to close the report.

closed 2022-01-06 by fiechr

descipher, 2022-01-06:
Not sure you need to touch Z2_MIN_PIN.

fiechr, 2022-01-06:
> Not sure you need to touch Z2_MIN_PIN.

Since I don't use it, I guess not. But if I decide to use the Z switch(es) again, it may be better to keep this line to remember that it was changed on hardware level.

descipher, 2022-01-06:
What does not make sense to me is the use of ANYCUBIC_LCD_I3MEGA option to define the config for endstops.

pins_TRIORILLA_14.h
```
#elif EITHER(ANYCUBIC_LCD_CHIRON, ANYCUBIC_LCD_I3MEGA)
  #define Y_STOP_PIN                          42
  #define Z2_MIN_PIN                          43
  #ifndef Z_MIN_PROBE_PIN
    #define Z_MIN_PROBE_PIN                    2
  #endif
```

descipher, 2022-01-06:
Now it makes sense, they were using free pins for the LCD.