MarlinFirmware/Marlin #23469: [BUG] Axis name missing in Motion/Home menu

steblerjun, 2022-01-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Axis name (letter) missing in "Motion" submenu on stock display (Formbot T-Rex 3) with INDIVIDUAL_AXIS_HOMING_MENU enabled: menu item is only "Home" without an axis name (for each axis X Y Z ) 

### Expected behavior

Please add the axis name (letter X / Y / Z...) to the individual "Home" menu item

### Actual behavior

Homing of the "selected" (I don't see the axis name, first "Home"-menuitem is X-homing command, second "Home"-menuitem for the Y axis and so on...) axis is working, just the axis name (letter) is missing next to the "Home"-command on display

### Steps to Reproduce

Go into "Motion" submenu on stock display of 3d printer and have a look to the next menuentries after "Auto Home"

### Version of Marlin Firmware

Marlin 2.0.9.3 and Marlin bugfix-2.0.x

### Printer model

Formbot T-Rex 3

### Electronics

stock

ellensp, 2022-01-08:
No Configs provided, and no screen type listed, can only presume marlin example config for this machine is what this user is using. REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

steblerjun, 2022-01-12:
I'm so sorry: I didn't see the reading error after flashing firmware on my T-Rex; the firmware size was too big so the axis name where missing in the motion menu (and maybe there where more other things missing?!) - now I have disabled some features in marlin, rebuild and flashed my T-Rex again: now the axis names appeared again in the motion menu! My mistake, sorry again.

closed 2022-01-12 by steblerjun