MarlinFirmware/Marlin #23472: [BUG] Wrong temperature shown during filament change, leading to thermal runaway

rglory, 2022-01-07:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

After filament sensor trigger tool goes to parking and starts to unload filament. During print temperature was 220C. After parking in a second later it shows 190C and starts heating. If heating is not fast enough thermal runaway error happens. If heating is fast then after filament change and head is ready to go back into position it suddenly shows extruder temperature 240C and starts cooling.

### Bug Timeline

Seen it in 2.9.0.5

### Expected behavior

Expected to show proper temperature not to overheat and not provide thermal runaway error.

### Actual behavior

Shows lower temperature, overheat and may crash firmware due to thermal runaway.

### Steps to Reproduce

1. Let filament sensor trigger.
2. Watch temperature drops 30C
3. Heating back to "220C"
4. After filament change temperature shows "240C"

### Version of Marlin Firmware

2.0.9.3

### Printer model

Prism

### Electronics

BTT Skr v1.3

### Host Software

OctoPrint

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7831000/config.zip)

rglory, 2022-01-07:
I did some additional testing. So when during print I issue "change filament" command, it works fine, shows the same temperature as it was printing and starts to unload immediately. But if filament sensor triggers temperature drops immediatly (in this case from 210C to 188C), it starts heating and I am getting thermal runaway.

I added 2 screenshot showing the difference. I have video if necessary, but they are quite big.
[image]
[image]

ellensp, 2022-01-08:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

rglory, 2022-01-08:
I did the test with bugfix version and can confirm that problem persists.

ellensp, 2022-01-08:
Could this be a fault on your machine, ie a wiring issue on the Thermistors?
Can you try this test.  Move the hotend end somewhere centralized and safe. 
Heat the hotend to 210, wait for it to stabilize. (your display says 210) 
Move the axies to the park position, not using park or filament change code.  
check the temperature. was it constant?

rglory, 2022-01-08:
> Could this be a fault on your machine, ie a wiring issue on the Thermistors?

No this printer prints for hours I would notice bad wiring for sure. When I tested for bugfix-2.0.x I noticed that temperature dropped immediately after sensor triggered, but printer went to parking after couple seconds (was printing from buffer), so does not fit that agenda.

Again if I issue M600 manually it goes to the same parking position, but problem does not show. And it  is reproducible, it does that every time. Also if I trigger from 220C (first layer) temperature drops to 190 something, if I trigger on second+ layer from 210 then it drops to 188 and does that all the time. I can upload video where temperature changes immediatly, but stays stable. If bad wiring if would not do that.

I've seen people complaining about this bug constantly, but looks like they did not notice temperature drop.

ellensp, 2022-01-08:
I cannot see how this can happen... but am presuming it does.  let see if we can tease out some more information.
Can you enable SHOW_TEMP_ADC_VALUES and rerun test, logging the drop in tempature?

rglory, 2022-01-08:
Actually you gave me an idea about wiring. You were right, that's hardware problem - filament switch affects extruder thermistor somehow. Not sure how it is linked but will realize soon, now I know what to look for. Sorry for false alarm and thank you for your help. You can close this issue.

closed 2022-01-08 by ellensp