MarlinFirmware/Marlin #23473: [BUG] (No bed leveling, but probe works and measures correctly)

TudorG22, 2022-01-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

So after installing the BL Touch probe to my Ender 3 Pro (4.2.2), and installing newest Marlin firmware, all seems to work great. The probe works, does its 9 points test, and does not slam into the bed or anything. However, no matter which combinations of command I give to the printer or the g-code (G29 after the G28), during the actual print the z-axis does not move and it's like I had installed nothing.
I used Teaching Tech's tutorial to do the installation, and his website for the firmware code modifications.
Any help would be appreciated!

### Version of Marlin Firmware

2.0.x

### Printer model

Ender 3 Pro

### Electronics

stock

### Add-ons

BL Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

ellensp, 2022-01-08:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Stephen-Bartel, 2022-01-08:
I also thought that the BLTouch was not doing anything, until I tested it and realised that it was working. 

When your printer is probing the bed before your print, put a ruler or something else underneath the probe at 2 or 3 locations. Then, when your printer is printing the first layer, observer whether or not the printer adjusts. 

The point here is to simulate very obvious and large deviations. I thought mine was also not working until I did this and realised it was doing something.

TudorG22, 2022-01-08:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

will this work with normal Marlin, non bugfix? i tried bugfix first and it had the exact same issue and outputs

TudorG22, 2022-01-08:
> I also thought that my BLTouch was not doing anything, until I tested it and realised that it was working.
> 
> When your printer is probing the bed before your print, put a ruler or something else underneath the probe at 2 or 3 locations. Then, when your printer is printing the first layer, observe whether or not the printer adjusts at those points where you put the ruler underneath the probe.
> 
> The point here is to simulate very obvious and large deviations. I thought mine was also not working until I did this and realised it was actually doing something.

I am sure it doesn't do anything, as when i slightly tilted the bed to test it, on side was squished and the other was too far

borland1, 2022-01-08:
Are there rounding errors that negate small corrections?  Exaugurated probed corrections seem to work just fine.

Stephen-Bartel, 2022-01-08:
@TudorG22 For me, my 3DTouch does not seem to really do anything on my regular bed. I still end up with an uneven squish, where some sides don't have enough squish or other ones have too much squish. Even though there are these slightly low and high points, it doesn't do anything. 

Test it with a very obvious difference. Like I mentioned, put a ruler underneath 2 or 3 probing points when it is doing G29 before the print. Then, when it is putting down the first layer, check to see if it makes adjustments at those very obvious deviations. It should. That's the only time I have seen my printer make adjustments. 

@borland1 I'm not too sure if this is an error or not. I will admit that my 3DTouch has been useless. I see no purpose of it, since it doesn't adjust for any of the imperfections. That's also why I have the G35 issue open, because I have to make my bed perfectly level because the BLTouch is not  really doing anything, so I have to make it level.

Stephen-Bartel, 2022-01-08:
And I would like to add that some differences are on the scale of fractions of a millimetre. On most beds, there are only very small deviations. There would only be a very very fine movement, not obvious at all, to adjust for a fraction of a millimetre. 

However, my 3DTouch still does not seem to make my first layer even at all. So, even though its adjustments would probably not really be visible, I still question whether it is doing anything at all, since it is not giving a perfect first layer. There is still unevenness in the way it prints the first layer, just like without a 3DTouch.

TudorG22, 2022-01-08:
i have tried using a ruler and it adjusted for it. it means it does not have enough precision

Stephen-Bartel, 2022-01-08:
> i have tried using a ruler and it adjusted for it. it means it does not have enough precision

Yes, exactly. That is what I was saying. 

As I mentioned, I have the same problem, where for me, the 3DTouch is useless and does not do anything. I would have similar first layer results without a 3DTouch.

TudorG22, 2022-01-08:
> 

what  I think is happening after some analysis is the mesh is inverted. how can I invert it?

Stephen-Bartel, 2022-01-08:
> > 
> 
> what I think is happening after some analysis is the mesh is inverted. how can I invert it?

What are you using to visualise the bed? Also, make sure that the depiction of the bed and its orientation is according to what you think you are seeing. E.g. make sure what you think is the front-left of the bed is actually that in the depicted image. 

This is the visualiser I use after running G29 in Pronterface : https://i.chillrain.com/index.php/3d-printer-auto-bed-leveling-mesh-visualizer/

TudorG22, 2022-01-08:
it's just a hypothesis, no proof of visualasing date to back it up

rowbotik, 2022-01-08:
I’m running into a very similar experience. Updated to Marlin 2.0.9.2 and had leveling issues, tried 2.0.9.3 and the bugfix-2.0.x. The bed topography is recorded appropriately by the probe but it does not appear that the bed leveling works while printing.  

[image]

TudorG22, 2022-01-08:
> I’m running into a very similar experience. Updated to Marlin 2.0.9.2 and had leveling issues, tried 2.0.9.3 and the bugfix-2.0.x. The bed topography is recorded appropriately by the probe but it does not appear that the bed leveling works while printing.
> 
> [image]

@rowbotik did you find a fix? I found out mine actually does leveling my seems inverted

rowbotik, 2022-01-08:
I haven’t found a fix yet but now that you mention it, I feel like mine is inverted as well. The right side has higher spots on the visualization and the nozzle seems to be higher on that side with lines not connecting but very low on the left side.

rowbotik, 2022-01-08:
@TudorG22 I haven’t found a fix yet but now that you mention it, I feel like mine is inverted as well. The right side has higher spots on the visualization and the nozzle seems to be higher on that side with lines not connecting but very low on the left side.

TudorG22, 2022-01-08:
> @TudorG22 I haven’t found a fix yet but now that you mention it, I feel like mine is inverted as well. The right side has higher spots on the visualization and the nozzle seems to be higher on that side with lines not connecting but very low on the left side.

@rowbotik now I will sleep but after I wake up tomorrow I will attempt to reverse the mesh in the firmware code and recompile, I will tell you how it goes

Stephen-Bartel, 2022-01-10:
Make sure that a fade height such as 10.0mm is defined. 

If you have a fade height that is 0.00mm then the mesh will not be used during the print.

TudorG22, 2022-01-10:
@MrBamse I will try another version of the firmware

borland1, 2022-01-10:
> Make sure that a fade height such as 10.0mm is defined.
> 
> If you have a fade height that is 0.00mm then the mesh will not be used during the print.

That's not what G-code M420 Marlin documentation says, fade will be disabled, not compensation...

`Set to 0 to disable fade, and leveling compensation will be fully applied to all layers of the print.`

Stephen-Bartel, 2022-01-10:
Thanks for the clarification. 

I stumbled on that info while searching for answers to my problem as to why the Z-Axis was not adjusting. Someone claimed changing that value to 10.0mm from a very low value (close to 0), fixed it for them. 

I guess the distinction here is that 0.00 and a very small value result in different behavior. That is something I should have said close to 0.00.

MrBamse, 2022-01-10:
> Thanks for the clarification.
>
> I stumbled on that info while searching for answers to my problem as to why the Z-Axis was not adjusting. Someone claimed changing that value to 10.0mm from a very low [...]

Just to be sure I tried it, same result though. :/
G28
G29
M420 S1 Z0

borland1, 2022-01-10:
There are other considerations not mentioned here, like` Z-Axis leadscrew backlash`.  Compensation needs to be greater than any unaccounted for backlash, otherwise there would be no compensation effects.

Most Z axis leadscrews have some measurable backlash which can be corrected for by hardware solution ( combination of spring loaded leadscrew nut and a solid or zero backlash bellows shaft coupler ), or Marlin firmware correction ( Backlash Compensation enabled in `Configuration_adv.h` ).

rowbotik, 2022-01-11:
Gave the Fade Height fix a try and same thing. I do have anti backlash nuts installed trying to fix this but no luck. I was originally running into this with UBL and figured I was doing it wrong and switch to Bilinear. Something to note, the G26 validation from UBL worked fine every time but prints did not, if that’s helpful to anyone.

tititopher68-dev, 2022-01-11:
@MrBamse 
Be careful with your startGcodes, most of the ones you do are wrong.
In ABL mode Bilinear  : 

soit : 
M280 P0 S160 ; Reset Alarm BLTouch
G4 P100 ; Delay 100ms for BLTouch
G28;
G29;

soit : 
M280 P0 S160 ; Reset Alarm BLTouch
G4 P100 ; Delay 100ms for BLTouch
G28;
M420 S1 ; Active the last Mesg in RAM and apply auto bed leveling on 10 first layers

or : 
M280 P0 S160 ; Reset Alarm BLTouch
G4 P100 ; Delay 100ms for BLTouch
G28;
M420 S1 Z'L'; Active the last Mesh in RAM and apply auto bed leveling on 'L' first layers

Having tested extensively until the last Bugfix marlin, I do not see any results, the correction is applied correctly whether in UBL or in ABL. Now in your situation, there are plenty of other possibilities so that the Mesh does not seem to be applied during printing (Physical geometry of the printer, perpendicularity X Axis vs Z Axis, Z screws, vSlot Wheels, etc. ..). Note that the movements for making a bed Leveling are not the same as those for printing. I have had users with this same kind of problem.

TudorG22, 2022-01-11:
@tititopher68-dev do you have any tips for me? stuff for me to check?

MrBamse, 2022-01-11:
> @MrBamse Be careful with your startGcodes, most of the ones you do are wrong. In ABL mode Bilinear :
> 
> either : M280 P0 S160 ; Reset Alarm BLTouch G4 P100 ; Delay 100ms for BLTouch G28; G29;
> 
> or : M280 P0 S160 ; Reset Alarm BLTouch G4 P100 ; Delay 100ms for BLTouch G28; M420 S1 ; Active the last Mesg in RAM and apply auto bed leveling on 10 first layers
> 
> or : M280 P0 S160 ; Reset Alarm BLTouch G4 P100 ; Delay 100ms for BLTouch G28; M420 S1 Z'L'; Active the last Mesh in RAM and apply auto bed leveling on 'L' first layers
> 
> Having tested extensively until the last Bugfix marlin, I do not see any results, the correction is applied correctly whether in UBL or in ABL. Now in your situation, there are plenty of other possibilities so that the Mesh does not seem to be applied during printing (Physical geometry of the printer, perpendicularity X Axis vs Z Axis, Z screws, vSlot Wheels, etc. ..). Note that the movements for making a bed Leveling are not the same as those for printing. I have had users with this same kind of problem.

Thank you, I'm well aware. Unfortunately these are some of the "solutions" circling around amongst the hundreds of people where ABL doesn't work anymore and I wanted to get that out of the way.

I have reflashed the FW reset everything, every axis is perfectly leveled including x axis. It is also obvious that the BLTouch works (as stated in so many threads I can't even count them all) since a mesh can be created.

But the problem is simple, using G29 or not doesn't make any difference. I've made numerous layer tests where 100% of them have identical results with G28 followed by G29 and  G28 alone. Ergo G29 doesn't make any difference like it used to, Z is absolutely not moving.

Sorry, but it simply doesn't work anymore. The best tip out there seem to be moving on to UBL which apparently works. I have not been able to verfy this though.

TudorG22, 2022-01-11:
@MrBamse I confirm, I tried almost everything imaginable. I tested UBL, it's worse for me: the axis moves, but litterally slams in the bed then prints in the void. probe itself works perfectly (0.0028
[image]
 accuracy)

TudorG22, 2022-01-11:
@MrBamse 
[image]

MrBamse, 2022-01-11:
> @MrBamse I confirm, I tried almost everything imaginable. I tested UBL, it's worse for me: the axis moves, but litterally slams in the bed then prints in the void. probe itself works perfectly (0.0028 accuracy)

That really sucks. :/

TudorG22, 2022-01-11:
@MrBamse I will try to change the plastic at the end of the bl touch 'endstop' wiring to plug it into the dedicated pins, and then download the firmware from creality website. if that doesn't work either, I will lose all hope.

thisiskeithb, 2022-01-14:
No one has followed the steps by ellensp in https://github.com/MarlinFirmware/Marlin/issues/23473#issuecomment-1007830118. Without more data, we're just guessing at what the issue could be.

midnightsmith89, 2022-01-14:
Tell me how to export the log you are requesting and i will get this over the weekend.  I'm ripping my hair out for two years and various bugfix firmwares trying to resolve this.  Do you mean to copy the entire terminal output to a text file?

MrBamse, 2022-01-14:
> No one has followed the steps by ellensp in [#23473 (comment)](https://github.com/MarlinFirmware/Marlin/issues/23473#issuecomment-1007830118). Without more data, we're just guessing at what the issue could be.

I could try as good as I can, haven't compiled custom firmware before but the rest of the steps seems super clear, well except for the export like @midnightsmith89 asks about.

Can I enable DEBUG_LEVELING_FEATURE and M114_DETAIL without compiling custom fw and/or reflashing?

borland1, 2022-01-14:
MrBamse,

No, you can't follow `ellensp` debugging instructions without recompiling your printer's firmware.  `thisiskeithb`  comments were directed towards user `TudorG22`

MrBamse, 2022-01-15:
> MrBamse,
> 
> No, you can't follow `ellensp` debugging instructions without recompiling your printer's firmware. `thisiskeithb` comments were directed towards user `TudorG22`

Cool.
Question @borland1 : If the bed is off by (lets say) 0.05mm it a spot where BLtouch is testing, is that enough for ABL to compensate for or what is the lower limit?

closed 2022-03-27 by github-actions[bot]

gls7, 2022-05-09:
Where did everybody go? I'm dying to hear the outcome so I can fix mine. Did everyone find a solution and go away quietly? Did the Marlin devs pay off everyone to be silent? Or is the landfill littered with rusting corpses of 3D printers?

midnightsmith89, 2022-05-10:
Marlin dev's just give up because they dont know whats wrong.  here's the latest thread with a ton of documentation, but no result https://github.com/MarlinFirmware/Marlin/issues/22322

gls7, 2022-05-10:
> 

That's all I find, tons of documented problems, but 0 solutions. Thanks for the current link.