MarlinFirmware/Marlin #23474: [BUG] Comipling Marlin 2.0.8 with STM32F103RET6_creality

Apollyus, 2022-01-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, when i tried to compile Marlin 2.0.8 by this tutorial: https://www.youtube.com/watch?v=kFRy_5lh2IQ&lc=z23uydyqxv2jsbjk5acdp43bphqqyvqj113by5s2wdpw03c010c it didn't worked. On another versions it compiled well, but here it gives me this: 

Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SoftwareSerialM

Unpacking  [------------------------------------]    0%
Unpacking  [##----------------------------------]    7%
Unpacking  [#####-------------------------------]   14%
Unpacking  [#######-----------------------------]   21%
Unpacking  [##########--------------------------]   28%
Unpacking  [############------------------------]   35%
Unpacking  [###############---------------------]   42%
Unpacking  [##################------------------]   50%
Unpacking  [####################----------------]   57%
Unpacking  [#######################-------------]   64%
Unpacking  [#########################-----------]   71%
Unpacking  [############################--------]   78%
Unpacking  [##############################------]   85%
Unpacking  [#################################---]   92%
Unpacking  [####################################]  100%
Library Manager: SoftwareSerialM @ 1.0.0 has been installed!
Library Manager: Installing U8glib-HAL @ ~0.4.4

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    3%
Unpacking  [#-----------------------------------]    4%
Unpacking  [#-----------------------------------]    5%
Unpacking  [##----------------------------------]    5%
Unpacking  [##----------------------------------]    6%
Unpacking  [##----------------------------------]    7%
Unpacking  [###---------------------------------]    8%
Unpacking  [###---------------------------------]    9%
Unpacking  [###---------------------------------]   10%
Unpacking  [####--------------------------------]   11%
Unpacking  [####--------------------------------]   12%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   14%
Unpacking  [#####-------------------------------]   15%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   17%
Unpacking  [######------------------------------]   18%
Unpacking  [######------------------------------]   19%
Unpacking  [#######-----------------------------]   19%
Unpacking  [#######-----------------------------]   20%
Unpacking  [#######-----------------------------]   21%
Unpacking  [########----------------------------]   22%
Unpacking  [########----------------------------]   23%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   25%
Unpacking  [#########---------------------------]   26%
[... 492 lines cut ...]
Compiling .pio\build\STM32F103RET6_creality\lib2bd\USBComposite\usb_setup.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib2bd\USBComposite\usb_x360_generic.c.o
Compiling .pio\build\STM32F103RET6_creality\lib2bd\USBComposite\usb_x360w.c.o
Compiling .pio\build\STM32F103RET6_creality\libf97\Servo\Servo.cpp.o
Archiving .pio\build\STM32F103RET6_creality\lib2bd\libUSBComposite.a
Archiving .pio\build\STM32F103RET6_creality\libf97\libServo.a
Linking .pio\build\STM32F103RET6_creality\firmware-20220107-233927.elf
c:/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:/Users/Vojtěch Faltýnek/Desktop/Marlin-2.0.8/buildroot/share/PlatformIO/ldscripts/creality.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RET6_creality\firmware-20220107-233927.elf] Error 1
========================================================================================= [FAILED] Took 41.46 seconds =========================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:41.461
==================================================================================== 1 failed, 0 succeeded in 00:00:41.461 ==================================================================================== 
The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

I saw on another forum guy with same problem and someone give him advice to delete .pio. But this didn't worked for me. Im sitting here for 4 hours and still don't know how to fix this. Can you guys help me please?

### Version of Marlin Firmware

2.0.8

Apollyus, 2022-01-07:
Forgot to say I have stock Creality Ender 3 with motherboard 4.2.2

ellensp, 2022-01-07:
Parts of vscode do no support non English characters in the path
Move marlin to another drive/path and avoid character such as ě as ý
This is outside of Marlins control.

closed 2022-01-07 by ellensp

Apollyus, 2022-01-08:
> Parts of vscode do no support non English characters in the path Move marlin to another drive/path and avoid character such as ě as ý This is outside of Marlins control.

Checked non-English characters and still have this problem :/

ellensp, 2022-01-08:
The error is clear
cannot open linker script file C:/Users/Vojtěch Faltýnek/Desktop/Marlin-2.0.8/buildroot/share/PlatformIO/ldscripts/creality.ld
Invalid argument
If you moved Marlin, it cannot be giving you the same error, the path will be different for a start

Apollyus, 2022-01-08:
Oh, i thinked you mean right in the code. Im not at home rn, I'll try it, but im not sure if it will work, beacuse with a another newer versions it worked

Apollyus, 2022-01-08:
> The error is clear cannot open linker script file C:/Users/Vojtěch Faltýnek/Desktop/Marlin-2.0.8/buildroot/share/PlatformIO/ldscripts/creality.ld Invalid argument If you moved Marlin, it cannot be giving you the same error, the path will be different for a start

Okey dude, i officially love you!! It worked thank you very much!!