MarlinFirmware/Marlin #23478: [BUG] Changes in pins.h/pins_RAMPS.h breaks pin mapping for MSK GEN L

drug123, 2022-01-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Default pin mapping for heaters and fan on MKS GEN L V1.0 in 2.0.9.2 and earlier was corresponding to silkscreen on the board as the following:
- E0 = 10
- E1 = 7
- BED = 8
- FAN = 9
Default mapping in the 2.0.9.3 (incl. bugfix) is the following:
- E0 = 10
- E1 = 9
- BED = 8
- FAN = 4
This leads to improper functioning of existing builds and the absence of a proper connector on D9 makes an impossible to connect a second extruder to the board w/o modifying the code.

### Bug Timeline

2.0.9.3

### Expected behavior

Mapping should be as in prev. version:
- E0 = 10
- E1 = 7
- BED = 8
- FAN = 9

### Actual behavior

Default mapping in the 2.0.9.3 (incl. bugfix) is the following:
- E0 = 10
- E1 = 9
- BED = 8
- FAN = 4

### Steps to Reproduce

N/A

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3

### Electronics

MKS GEN L

### Add-ons

Custom dual extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

ellensp, 2022-01-08:
This looks to work from my testesting

In Marlin/src/pins/ramps/pins_MKS_GEN_L.h

delete 
#define MOSFET_D_PIN 7

add
#define MOSFET_B_PIN 7
#define MOSFET_C_PIN 8
#define FAN_PIN 9

drug123, 2022-01-09:
As a temporary fix for myself, I had to add board check in pins.h to disable FET_ORDER_ defines:
`#if !defined BOARD_MKS_GEN_L
  #if NONE(FET_ORDER_EEF, FET_ORDER_EEB, FET_ORDER_EFF, FET_ORDER_EFB, FET_ORDER_SF)
...
`

borland1, 2022-01-09:
In `Configuration_adv.h` have enabled `E0_AUTO_FAN_PIN 9` , so there is no difference if I change in pins file pins_MKS_GEN_L.h, from

`line 38   #define MOSFET_D_PIN 7`
to
`line 38   #define MOSFET_D_PIN 9`

However if I change to

`line 38   #define FAN_PIN 9`

I get a complier error in PlatformIO.   `"You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN"`

My `Configuration.h` :

```
#define MOTHERBOARD BOARD_MKS_GEN_L
#define IS_RAMPS_EFB
```
Why is that?

drug123, 2022-01-09:
@borland1 This is totally unrelated to described bug.

PS: You have an issue because you confusing Extruder Fan and Cooling Fan. They cannot be connected to the same pin. If you wish to use auto fan feature you should have only one extruder (this is the case OPPOSITE to described setup) and connect extruder fan to E1 connector on MKS GEN_L and then use `#define E0_AUTO_FAN_PIN`

ellensp, 2022-01-09:
@borland1 FAN_PIN is a gcode controlled fan (ie M106/M107) can be turn on/off at any time.
E0_AUTO_FAN_PIN is an auto fan, turns when hotend ois over a certain temperature.
These are mutually exclusive functions.

descipher, 2022-01-16:
> IS_RAMPS_EFB

IS_RAMPS_EFB is now FET_ORDER_EFP

borland1, 2022-01-16:
>  IS_RAMPS_EFB is now FET_ORDER_EFB

Yes,  thanks.  Still on v2.07.2, but will change that when updating to later version.

Charmandrigo, 2022-03-01:
This issue also affected the MKS Gen 1.4, fans stopped working on this update

thinkyhead, 2022-04-10:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where this stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2023-04-20:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25717

closed 2023-04-20 by thisiskeithb