MarlinFirmware/Marlin #23486: [FR] (feature summary)

sbarajas, 2022-01-09:
### Is your feature request related to a problem? Please describe.

I need to use a Wifi module in the motherboard trigorilla pro but it isn't supported

### Are you looking for hardware support?

The motherboard trigorilla pro has a WiFi connector:
[image]
for an esp8266 WiFi module like that:
[image]
But in the motherboard definition in Marlin isn't defined de serial port to use it

### Describe the feature you want

I want to connect by wifi to the trigorilla mother board using the ESP3D graphical interface

ellensp, 2022-01-10:
Its a serial device.
PB11 is USART3_RX so one would home that PB10 USART3_TX is on that connector also.
All you do is enable the uart
#define SERIAL_PORT_2 3  and set the baud rate, if needed)
This does not require anything aditional.

closed 2022-01-10 by ellensp

sbarajas, 2022-01-10:
Is not enougth,
I have tried this:

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 3
#define BAUDRATE_2 115200   // Enable to override BAUDRATE

And this not enable de wifi module

[image]