MarlinFirmware/Marlin #23492: [BUG] Printer frequently stops for a moment during print

Smitty-Werbenjagermenjensen, 2022-01-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When printing, the printhead frequently pauses randomly. 

Video is here : https://streamable.com/chobwp

### Bug Timeline

I have been using 2.0.9.2 and this issue has only just started occurring. 

### Expected behavior

It shouldn't pause during the print. 

### Actual behavior

I randomly pauses during the print for a moment. 

### Steps to Reproduce

1. Start printing
2. Pay attention to the print and see when it stops/starts

### Version of Marlin Firmware

2.0.9.2

### Printer model

Creality Ender 3

### Electronics

Creality 4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7836656/Configuration.zip)

Smitty-Werbenjagermenjensen, 2022-01-10:
I changed the micro SD card to a newer and faster one, and I am no longer experiencing this problem. Thanks.

closed 2022-01-10 by Smitty-Werbenjagermenjensen

thinkyhead, 2022-04-10:
If you don't need Power-Loss Recovery, that can be turned off for better results — especially with vase printing.