MarlinFirmware/Marlin #23493: [BUG] STM32F103RC_btt_USB mass storage errors

Loafdude, 2022-01-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using bugfix-2.0.x as of Jan 9/2022
SKR Mini E4 v1.2 board (STM32F103RC) on Ender 3
I am upgrading from Marlin 2.0.5.3 -> latest

When building against STM32F103RC_btt_USB upon Marlin bootup;
Com port appears and mass storage drive appears.
Mass storage drive is inaccessible "Please insert disk into drive G:"
Com Port is also non-functional error given is `WindowsError(995, 'The I/O operation has been aborted because of either a thread exit or an application request.')`
If I open windows Disk Management it stalls indefinitely until I disconnect USB.
If I unplug and replug USB the drive is detected as fat32 and is mounted and accessible
Com port also becomes functional after replug.

When building against STM32F103RC_btt_USB_maple upon Marlin bootup;
Com port appears but no mass storage drive appears after replug

I have tried disabling ENDSTOP_INTERRUPTS_FEATURE and reducing SD spi speed to slowest settings but it has not helped

This was working in 2.0.5.3.

Please let me know how I can assist in debugging. 
I have lots of ROM space left for debug features

### Bug Timeline

2.0.5.3 -> Latest

### Expected behavior

USB to work on boot

### Actual behavior

USB is not working on boot

### Steps to Reproduce

1. Flash latest 2.0.x bugfix with USB enabled
2. USB mass storage device will either not appear or not work.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jan  9 2022 23:07:59) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff Cap:SERIAL_XON_XOFF:0 Cap:BINARY_FILE_TRANSFER:0 Cap:EEPROM:1 Cap:VOLUMETRIC:1 Cap:AUTOREPORT_POS:0 Cap:AUTOREPORT_TEMP:1 Cap:PROGRESS:0 Cap:PRINT_JOB:1 Cap:AUTOLEVEL:1 Cap:RUNOUT:0 Cap:Z_PROBE:1 Cap:LEVELING_DATA:1 Cap:BUILD_PERCENT:1 Cap:SOFTWARE_POWER:0 Cap:TOGGLE_LIGHTS:0 Cap:CASE_LIGHT_BRIGHTNESS:0 Cap:EMERGENCY_PARSER:1 Cap:HOST_ACTION_COMMANDS:0 Cap:PROMPT_SUPPORT:0 Cap:SDCARD:1 Cap:REPEAT:0 Cap:SD_WRITE:1 Cap:AUTOREPORT_SD_STATUS:0 Cap:LONG_FILENAME:0 Cap:EXTENDED_M20:0 Cap:THERMAL_PROTECTION:1 Cap:MOTION_MODES:0 Cap:ARCS:1 Cap:BABYSTEPPING:1 Cap:CHAMBER_TEMPERATURE:0 Cap:COOLER_TEMPERATURE:0 Cap:MEATPACK:0 Cap:CONFIG_EXPORT:0

### Printer model

Ender 3

### Electronics

SKR E3 v1.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

Loafdude, 2022-01-11:
Okay upon further testing this is more broken than I thought.

STM32F103RC_btt
- Serial Works and is stable
- SD Card menu EXTREMELY slow

STM32F103RC_btt_usb
- USB is unusable at boot
- USB is unstable after unplug/replug
- Serial Connection drops due to USB instability
- lsusb errors and returns to command prompt on first boot
- Upon replug `VendorID:0483 ProductID:5740  "USB Composite Device"   "STMicroelectronics"    "GENERICSTM32F103RE CDC in FS Mode"    "48DB83593032"    "GENERICSTM32F103RE CDC in FS Mode"    "GENERICSTM32F103RE CDC Interface"    "GENERICSTM32F103RE CDC Interface"    "GENERICSTM32F103RE CDC Interface"`

STM32F103RC_btt_maple
- Serial Works and is stable
- SD Card menu operates at decent speed
- `VendorID:1eaf ProductID:0004  "USB Serial Device"   "LeafLabs"    "Maple"`

STM32F103RC_btt_usb_maple
- Serial Works and is stable
- SD Card menu operates at decent speed
- Mass storage device is not presented to the host - is it even supported?
- `VendorID:1eaf ProductID:0029  "USB Composite Device"   "LeafLabs"    "Maple"`

These issues may be related to
https://github.com/MarlinFirmware/Marlin/issues/18157
github.com/MarlinFirmware/Marlin/pull/18496

thinkyhead, 2022-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Loafdude, 2022-05-09:
This is still non-working as of 2.0.9.3-330-g975b90c0b7 (pulled today)

thisiskeithb, 2022-05-09:
Please zip up & attach your configs as well as any other files you have modified.

Loafdude, 2022-05-09:
[Issue 23493 - STM32F103RC_btt_USB - USB non working.zip](https://github.com/MarlinFirmware/Marlin/files/8648753/Issue.23493.-.STM32F103RC_btt_USB.-.USB.non.working.zip)

Configs and boot loader images attached.
One stm32f1.ini also included. It is modified to allow use of full rom-size. (BTT SKR 1.2 chips are labeled STM32F103RC but actually have the rom size of STM32F103RE)
Nothing else is modified.

Let me know if you need anything else.

thisiskeithb, 2022-05-09:
> It is modified to allow use of full rom-size. (BTT SKR 1.2 chips are labeled STM32F103RC but actually have the rom size of STM32F103RE)

No, that change was reverted in this repo specifically due to instability. Please revert those changes and test with the correct size of 256k for RC chips.

Loafdude, 2022-05-09:
I have retested with 256k and it is the same (boot screens disabled).

FYI
I ran 2.0.5.3 as a RE 512k with zero stability issues for over a year.
I currently run as a RE 512k today using STM32F103RC_btt_maple with 100% stability
Perhaps there are unlucky people out there that did not get a RE in RC labeling but I did.

Loafdude, 2022-07-09:
@thinkyhead Please let me know if I can assist in troubleshooting this issue.

thisiskeithb, 2022-07-09:
Did you revert those changes as requested and test with the correct size of 256k for RC chips?

Overflashing these chips was removed and is not supported.

rhapsodyv, 2022-07-09:
I’m author of part of this work. The only issue I remember was the need to unplug the cable and plug again, if marlin start with the cable attached. This issue was in my task list in the past.
I don’t remember any of the issues you are reporting, but too many things changed.
I will try to revisite this usb and sd code on stm32.

Loafdude, 2022-09-14:
This issue is still active. 
I am still ready and willing to help if a dev is available to troubleshoot.

brew99, 2022-12-15:
I also have a similar issue with my SKR mini E3 V1.0 board.  bugfix 2.1.1 from 2 days ago, built with STM32F103RC_BTT_USB environment results in Win10 PC recognizing the serial, but after about a min, it comes up with "insert disk into drive:X", and displays no files.  Tried another SD card and same issue.  Worked with an older build of bugfix 2.0.x (maybe 1.5 yrs old) when building with Maple.  Since Maple is deprecated, I'm trying to build with the _USB only

closed 2023-02-28 by github-actions[bot]