MarlinFirmware/Marlin #23495: [FR] Linear Advance Pattern - Link printed elements to allow analysis off the print surface (and ease removal)

gpread, 2022-01-10:
### Is your feature request related to a problem? Please describe.

This is not technically about the firmware itself, is a request to enhance a tool in relation to configuration of the firmware. 

Its an enhancement request for the linear advance calibration pattern generator(https://marlinfw.org/tools/lin_advance/k-factor.html)

This is fundamentally already a good tool but it can be a bit difficult to use in practise to cover the necessary permutations/combinations of materials and settings.

The "problems":

- In my experience it can be very difficult to analyse/interpret the printed result while on the print surface, since there is often a lack of contrast depending on the specifics of the filament and bed surface materials being used (colour, reflectivity, translucency, etc)..
- The K-factor labels often won't print properly/clearly
- The K-factor label elements are tedious to remove/clean from the print surface (#22686)

### Are you looking for hardware support?

No, this issue/request is not specific to particular hardware

### Describe the feature you want

To be able to remove the entire printed output as a single piece and analyse it off the  print surface.

### Additional context

I have manually edited an output file from the current standard Marlin generator so as to experiment.  I achieved a workable design with no major changes to gcode, which allowed me to remove the entire print from the bed in one piece and then examine it off the print surface.

Fundamentally, I added an attachment line to the frame for each k-factor label, 1mm below them and tails to each number (and dot) to connect them to it and also connected the test area reference markers to the frame.

Below is an image of the printed result after removal from the bed and placed on top of a piece of white paper (there is some shadow caused by the print not lying completely flat on the paper) ->

[image]

I thought it would be helpful to share ,so as to at least get feedback and/or further support for such changes and their details. 

In the future if I can find the time (I can/do hack but IMHO I'm not proficient in javascript nor html) I will try to implement my changes in a fork of the code and contribute it back.

thisiskeithb, 2022-01-10:
Duplicate of #22686

gpread, 2022-01-10:
Below is a G-code file that produced the image above

[LIN_ADV_TEST_0.4Noz_0.2mm_10-50mms_K0_02-0_50_H230_B55.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/7840064/LIN_ADV_TEST_0.4Noz_0.2mm_10-50mms_K0_02-0_50_H230_B55.gcode.txt)