MarlinFirmware/Marlin #23497: [BUG] BLTouch correction seems inverted

TudorG22, 2022-01-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When it needs to raise z axis it seems to lower it, and when it needs to be lower it raises it.
In the firmware code i cannot invert the probe's logic beacause it won't compile. Any tips?

### Bug Timeline

always had this issue since i got the bl touch 2 weeks ago

### Expected behavior

I excpected the correction to work properly 

### Actual behavior

It was inverted

### Steps to Reproduce

1) compile marlin for ender 3 pro (board v4.2.2)
2) adjust the code for bl touch
3) flash the firmware
4) slice something with g29 after the g28 in the g code
5) watch the nozzle print in nothingness then slam into the bed

### Version of Marlin Firmware

latest bugfix version

### Printer model

Ender 3 Pro 

### Electronics

all stock

### Add-ons

genuine bl touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

MrBamse, 2022-01-10:
Yepp, this is very real.
See: https://github.com/MarlinFirmware/Marlin/issues/23473
Amongst other threads.

TudorG22, 2022-01-10:
> Yepp, this is very real. See: #23473 Amongst other threads.

@MrBamse that is my issue

TudorG22, 2022-01-10:
@MrBamse nvm I'm stupid

ellensp, 2022-01-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

tome9111991, 2022-01-11:
did you change anything about the Z-steps?

had a similar bug a long time ago where I changed Z-step values

TudorG22, 2022-01-11:
@tome9111991 nope

midnightsmith89, 2022-01-11:
I can second this, I have 3 machines with BLtouch on them.  Ender 3 machine is mounted on the left, Tronxy on the right, and homemade one on the front.  On EVERY machine, the opposite side of the BLtouch ends up being far too close to the bed, it is visibly compensating down towards the bed.  so the ender is right side, tronxy left, and homemade one rear.  

This has been occurring for over 2 years now.  BLtouch on ender is smart 3.0, tronxy smart 3.1, and homemade smart 3.0.  I have purchased new BLtouch devices to rule that out, latest bugfix and stable firmwares, different drivers from 2208 to 2209 drivers.  Ive used the bigtree tech 1.3 board, 1.4, and 1.4 turbo boards, all the same effect.  New Z rod couplers to eliminate backlash as the issue.

The software is clearly moving the Z axis, but its always lower on the opposing mounting point side of the BLtouch.  In fact, the rear side of the tronxy and ender 3 is slightly too close, and that reinforces the issue because they are also about 5-10 mm in front of the nozzle as far as mounting.  

Search github and youll see hundreds of issues, open and "closed" with no solutions on this same subject.  I has a very old marlin build, 1.8.something and that one worked perfectly with the ABL, but ever since upgrading, its never worked again.

This is much bigger than just a bunch of users not knowing what they are doing, its a marlin issue.  I'm about to put Klipper on the same machine just to prove it.

Please Devs, look at this objectively, I'm willing to do what needs to be done to put this to bed once and for all, but I'm not a coder.    I just want my printer to WORK again.  Heres my last configs. 
[Marlin-11-6-21EnderEZ.zip](https://github.com/MarlinFirmware/Marlin/files/7849107/Marlin-11-6-21EnderEZ.zip)
[Marlin-Tronxy-Dual-Fan-December.zip](https://github.com/MarlinFirmware/Marlin/files/7849109/Marlin-Tronxy-Dual-Fan-December.zip)
[Marlin-ReDuplicator.zip](https://github.com/MarlinFirmware/Marlin/files/7849111/Marlin-ReDuplicator.zip)

TudorG22, 2022-01-11:
@mriscoc my problem, I found out, is it only does correction when there is a big difference in bed level (like a ruler applied under a probe point). but when the bed is level and there is nothing on it, it goes in a straight line and the first layer is uneven

rowbotik, 2022-01-14:
@TudorG22 How many points are you using for leveling? I’m finding to have somewhat better results with a 3x3 over the 5x5 or 7x7.

thisiskeithb, 2022-01-14:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23473

closed 2022-01-14 by thisiskeithb

TudorG22, 2022-01-15:
@rowbotik i tried 3x3, 4x4, and 5x5. results were identical. i have now reutrned the bl touch and will try out the cr touch