MarlinFirmware/Marlin #23499: [BUG] WiFi does not work in trigorilla pro mother board

sbarajas, 2022-01-11:
Its a serial device.
PB11 is USART3_RX, PB10 USART3_TX is on that connector
In Marlin all you do is enable the uart
#define SERIAL_PORT_2 3  and set the baud rate, #define BAUDRATE_2 115200
This does not require anything additional.

_Originally posted by @ellensp in https://github.com/MarlinFirmware/Marlin/issues/23486#issuecomment-1008454173_

sbarajas, 2022-01-11:
I have tried these options in 2.0.9.3 version, and didn't work

ellensp, 2022-01-11:
Please add way more details.

Does the web interface work?
Did you flash the Esp01 with ESP3D? What version? What config files?
Did the ESP01 work on a previous Marlin?
Did you change the baud rate in the web interface?
What build environment did you use trigorilla_pro_maple or trigorilla_pro? What Configs?

sbarajas, 2022-01-11:
If I plug this module in another printer with marlin 2.0.7 and a BTT SKR 1.4 it works fine.
I can control the other printer, and the web interface works okey

sbarajas, 2022-01-11:
I have tried with trigorilla_pro, not trigorilla_pro_maple

sbarajas, 2022-01-12:
More test:

I have tried with trigorilla_pro and trigorilla_pro_mapple environments of the 2.0.9.3 versión and not work.

I have also tried with a fork of version 2.0.0.5 of this repo: https://github.com/SXHXC/Marlin-Anycubic-Predator-Trigorilla-PRO of @SXHXC that, is supposed that WiFi works, and I can't get the module working

closed 2022-03-25 by github-actions[bot]