MarlinFirmware/Marlin #23500: kHz

jwrw, 2022-01-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It's a bit of trivia but I saw a load of changes to make kHz -> KHz.  'kHz' is the CORRECT SI spelling for kilohertz (even if it looks weird). Just to confuse, MHz and GHz are correct for megahertz and gigahertz.

### Version of Marlin Firmware

N/A

tome9111991, 2022-01-11:
This ?
https://github.com/MarlinFirmware/Configurations/commit/c57f74b8369b9bce073135f07ce2a6e353b3403e

jwrw, 2022-01-11:
Sorry - yes that's the one.  I was juggling too many repos!

thisiskeithb, 2022-01-12:
See https://github.com/MarlinFirmware/Marlin/pull/23512

thisiskeithb, 2022-01-12:
Aaaand merged! https://github.com/MarlinFirmware/Marlin/pull/23512

closed 2022-01-12 by thisiskeithb