MarlinFirmware/Marlin #23501: [BUG] RAMPS 1.4 Plus pins is not correct with REPRAP_DISCOUNT_SMART_CONTROLLER

fduraibi, 2022-01-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have the RAMPS 1.4 Plus board with REPRAP_DISCOUNT_SMART_CONTROLLER (LCD2004), and the current pins file [`pins_RAMPS_PLUS.h`](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/ramps/pins_RAMPS_PLUS.h) doesn't work with it since the pins settings are a bit different.

[image]

In ordered to make it work I changed the file to match the pins in the image, and disabled the code for the other type of LCD (LCD12864)

Here is the content of my file `pins_RAMPS_PLUS.h` that works with my printer setup:

```c
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Arduino Mega with RAMPS v1.4Plus, also known as 3DYMY version, pin assignments
 * The differences to the RAMPS v1.4 are:
 *  - Swap heater E0 with E1
 *  - Swap pins 8 and 10. Bed/Fan/Hotend as labeled on the board are on pins 8/9/10.
 *  - Change pins 16->42, 17->44 and 29->53 used for display.
 *
 * Applies to the following boards:
 *
 *  RAMPS_PLUS_EFB (Extruder, Fan, Bed)
 *  RAMPS_PLUS_EEB (Extruder, Extruder, Bed)
 *  RAMPS_PLUS_EFF (Extruder, Fan, Fan)
 *  RAMPS_PLUS_EEF (Extruder, Extruder, Fan)
 *  RAMPS_PLUS_SF  (Spindle, Controller Fan)
 */

#include "env_validate.h"

#define BOARD_INFO_NAME "RAMPS 1.4 Plus"

#define MOSFET_A_PIN                           8
#define MOSFET_C_PIN                          10

//
// Steppers
//
#define X_CS_PIN                              -1
#define Y_CS_PIN                              -1
#define Z_CS_PIN                              -1

// Swap E0 / E1 on 3DYMY
#define E0_STEP_PIN                           36
#define E0_DIR_PIN                            34
#define E0_ENABLE_PIN                         30
#define E0_CS_PIN                             -1

#define E1_STEP_PIN                           26
#define E1_DIR_PIN                            28
#define E1_ENABLE_PIN                         24
#define E1_CS_PIN                             -1

#define EXP1_03_PIN                           44
#define EXP1_04_PIN                           42
#define EXP1_05_PIN                           23
#define EXP1_06_PIN                           33
#define EXP1_07_PIN                           41
#define EXP1_08_PIN                           31
#define EXP1_09_PIN                           35
#define EXP1_10_PIN                           37

#define EXP2_03_PIN                           27
#define EXP2_04_PIN                           49
#define EXP2_05_PIN                           51
#define EXP2_06_PIN                           25
#define EXP2_07_PIN                           53
#define EXP2_08_PIN                           29
#define EXP2_09_PIN                           52
#define EXP2_10_PIN                           50

#include "pins_RAMPS.h"

// ** THIS MUST BE DISABLED SINCE IT OVERRIDES MY PREVIOUS SETTINGS **
// #if IS_ULTRA_LCD && NONE(REPRAPWORLD_GRAPHICAL_LCD, CR10_STOCKDISPLAY) && !BOTH(IS_NEWPANEL, PANEL_ONE)
//   #if DISABLED(MKS_12864OLED) || ENABLED(MKS_12864OLED_SSD1306)
//     #undef LCD_PINS_RS
//     #define LCD_PINS_RS                       42  // 3DYMY boards pin 16 -> 42
//     #undef LCD_PINS_ENABLE
//     #define LCD_PINS_ENABLE                   44  // 3DYMY boards pin 17 -> 44
//   #endif
//   #undef LCD_PINS_D7
//   #define LCD_PINS_D7                         53  // 3DYMY boards pin 29 -> 53
// #endif

```

The fix i think is simple which is to check if `REPRAP_DISCOUNT_SMART_CONTROLLER` is defined use the suitable pins.

### Version of Marlin Firmware

2.0.9.3

descipher, 2022-01-11:
Try just the basic ramps 1.4, works for me.
#define MOTHERBOARD BOARD_RAMPS_14_EFB
The plus pins file looks odd.

fduraibi, 2022-01-12:
@descipher The board I have is different from the original ramps 1.4, it is called [ramps 1.4 plus](https://reprap.org/wiki/Ramps1.4plus) which integrates the MEGA and the ramps in a single board.

It is nice and small, however, the designer for some reason didn't use the same pin out as the original board, maybe for easier and more efficient pcb trace routing.

ellensp, 2022-01-12:
Is this it? https://reprap.org/wiki/Ramps1.4plus

ellensp, 2022-01-12:
gah, you already had a link.. ignore moe

ellensp, 2022-01-12:
The above pins_RAMPS_PLUS.h looks fine. It shouldn't need any lcd settings since they are defined in ramps.h and  use EXP PIN definitions. 
Just needs a PR created to formally change it.

ellensp, 2022-01-12:
@fduraibi Do you need help creating the PR, or are you going to give this a try yourself?

fduraibi, 2022-01-12:
@ellensp I will give it a try myself and by the weekend should have a PR ready

thinkyhead, 2022-01-17:
Poking around the interwebs led me to the old PR #6033 … and the pins as suggested end up matching very closely. Only the changed pins are commented here, and all of them correspond with that PR.
```cpp
#define EXP1_03_PIN                           44  // (29) D7
#define EXP1_04_PIN                           42  // (27) D6
#define EXP1_05_PIN                           23  // (25) D5
#define EXP1_06_PIN                           33  // (23) D4
#define EXP1_07_PIN                           41  // (16) RS
#define EXP1_08_PIN                           31  // (17) EN
#define EXP1_09_PIN                           35
#define EXP1_10_PIN                           37

#define EXP2_03_PIN                           27  // (41) ENC
#define EXP2_04_PIN                           49
#define EXP2_05_PIN                           51
#define EXP2_06_PIN                           25  // (33) EN1
#define EXP2_07_PIN                           53
#define EXP2_08_PIN                           29  // (31) EN2
#define EXP2_09_PIN                           52
#define EXP2_10_PIN                           50
```

thinkyhead, 2022-01-17:
The description of the linked PR also notes "their 2004LCD has a different pinout than the REPRAP_DISCOUNT_SMARTCONTROLLER it clones."

So we actually need a new LCD definition that distinguishes the clone LCD controller from the standard one, not just the differing pinouts for EXP1/2.

fduraibi, 2022-01-17:
@thinkyhead I have the clone REPRAP_DISCOUNT_SMART_CONTROLLER (LCD2004) and it works fine by just fixing the EXP1 & 2, no other modifications are needed.

closed 2022-02-23 by fduraibi