MarlinFirmware/Marlin #23504: M105 seems to reply with atypical format

SidSkiba, 2022-01-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin reply to M105 on 2 extruder printer returns the following format in pronterface:

ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.08 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.06 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.08 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.70 /0.00 T0:23.00 /0.00 T1:23.06 /0.00 @:0 B@:0 @0:0 @1:0

T0 is repeated in the string.

The result is Pronterface cannot display the actual hot end temps nor the bed temps.

Sid

### Bug Timeline

I noticed it in 2.0.9.2 release

### Expected behavior

The format seems to be incorrect for pronterface to process.

### Actual behavior

ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.08 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.06 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.62 /0.00 T0:23.00 /0.00 T1:23.08 /0.00 @:0 B@:0 @0:0 @1:0
ok T0:23.00 /0.00 B:22.70 /0.00 T0:23.00 /0.00 T1:23.06 /0.00 @:0 B@:0 @0:0 @1:0

### Steps to Reproduce

Build the latest Marlin for a 2 extruder machine. Connect with pronterface. Watch the M105 responses and note there is no temps displayed.

### Version of Marlin Firmware

Latest bugfix-2.0.x

### Printer model

Self build using Zonestar ZRIB V5.3 (Mega 2560) board

### Electronics

Zonestar ZRIB V5.3, TMC2209, Bed Probe, Dual Z optical endstops, Zonestar LCD

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7851562/Marlin.zip)

ellensp, 2022-01-12:
To replicate is enough enough
Default config
#define EXTRUDERS 2
#define TEMP_SENSOR_1 1
#define HEATER_1_PIN 7 (as ramps does have this pin) 

bugfix 2.0.x. 2.0.9.3 and 2.0.9.2  results in T0 being displayed twice 
T0:17.54 /0.00 T0:17.54 /0.00 T1:20.70 /0.00 @:0 @0:0 @1:0

2.0.9.1 results in
T:17.50 /0.00 T0:17.50 /0.00 T1:20.86 /0.00 @:0 @0:0 @1:0

This reverts it to behave as it used to
```DIFF
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index c3cd4f10d6..0305e5e98e 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -3630,7 +3630,7 @@ void Temperature::isr() {
     OPTARG(HAS_TEMP_REDUNDANT, const bool include_r/*=false*/)
   ) {
     #if HAS_TEMP_HOTEND
-      print_heater_state(H_E0, degHotend(target_extruder), degTargetHotend(target_extruder) OPTARG(SHOW_TEMP_ADC_VALUES, rawHotendTemp(target_extruder)));
+      print_heater_state(H_NONE, degHotend(target_extruder), degTargetHotend(target_extruder) OPTARG(SHOW_TEMP_ADC_VALUES, rawHotendTemp(target_extruder)));
     #endif
     #if HAS_HEATED_BED
       print_heater_state(H_BED, degBed(), degTargetBed() OPTARG(SHOW_TEMP_ADC_VALUES, rawBedTemp()));
diff --git a/Marlin/src/module/temperature.h b/Marlin/src/module/temperature.h
index 089694ab4a..8621aa0a87 100644
--- a/Marlin/src/module/temperature.h
+++ b/Marlin/src/module/temperature.h
@@ -56,7 +56,8 @@ typedef enum : int8_t {
   H_BOARD = HID_BOARD,
   H_CHAMBER = HID_CHAMBER,
   H_BED = HID_BED,
-  H_E0 = HID_E0, H_E1, H_E2, H_E3, H_E4, H_E5, H_E6, H_E7
+  H_E0 = HID_E0, H_E1, H_E2, H_E3, H_E4, H_E5, H_E6, H_E7,
+  H_NONE = -6
 } heater_id_t;
 
 // PID storage

```

ellensp, 2022-01-12:
reading up on M105 https://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature

It would seem marlin has departed from the standard defined. here

thisiskeithb, 2022-01-12:
https://github.com/MarlinFirmware/Marlin/pull/23505 has been merged.

commit 3d79b55c (2022-01-12): 🚑️ Fix M105 regression (#23505)

Fixes #23504
commit 75b4ff9c (2022-01-12): 🚑️ Fix M105 regression (#23505)

Fixes #23504
commit 7135c3b1 (2022-01-12): 🚑️ Fix M105 regression (#23505)

Fixes #23504
closed 2022-01-12 by thisiskeithb