MarlinFirmware/Marlin #23507: [FR] ESP32-S2 support?

alan-student, 2022-01-12:
### Are you looking for hardware support?

Does marlin support ESP-S2?

### Describe the feature you want

Hello everyone. I'm developing a new board based on the ESP32. However ,original ESP32 has too few IOs. So I wonder if it possible to run on the ESP-S2, cause it has more IOs. How can I modified the firmware to make it run on the S2. Or if it can not be solved by simple modified, will the marlin support the S2?

ellensp, 2022-01-12:
esp's traditionally use I2S IO EXPANDER's

ellensp, 2022-01-12:
You would need to write a new HAL, or if if is close enough modify the existing one. Not a easy or simple task.

alan-student, 2022-01-12:
Hello Ellenson.
I know many boards use the i2s expanders. Anyway, the speed is limited to the expanded IOs. and the soft usart can not be used on the ESP32 now, it's tough to use the stepper driver based on the usart, like TMC2208 etc. I wonder if there is good way to solve the problem.

ellensp, 2022-01-12:
You need a developer, someone who has done micro-controller development for at least a decade and who knows the esp chips.

simon-jouet, 2022-01-27:
Hey @alan-student I have actually modified the ESP32 HAL I wrote to also support the ESP32-S2 some time ago. You should be able to see the code here https://github.com/simon-jouet/Marlin/commits/esp32s2 

I haven't worked on it in a while but it does work fine (both my printers use it) with my ESP32Controller R3 (esp32-s2) board https://github.com/simon-jouet/ESP32Controller

But as @ellensp is saying the best approach with an esp32 (s2 and not s2) is to use I2S as a GPIO expander

alan-student, 2022-01-27:
Thank you so much for you excellent work!@simon-jouet
I will check if it can run on ESP32-S2 in the coming future. And add the condition to the issue.

MZZQ-BEN, 2022-03-18:
！[上传E01DC11D-87E0-43C4-B265-B8F104135A1A.jpeg...]()
A boss has already done it.

ahakkoca07, 2024-11-04:
> ！上传E01DC11D-87E0-43C4-B265-B8F104135A1A.jpeg... A boss has already done it.

hi, the image you posted seems to be broken, if you still have the image, can you share it again please?