MarlinFirmware/Marlin #23508: [BUG] Compiler error for declaration NOZZLE_TO_PROBE_OFFSET. Possible re-occuring of issue #16767

EbbiDos, 2022-01-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling Marlin 2.0.93 with Arduino IDE 1.8.19 for an Arduino Due with RADDS shield I get the following compiler error:

`                 
                 Arduino: 1.8.19 (Windows 10), Board: "Arduino Due (Programming Port)"

                 In file included from E:\Plotter\Marlin-2.0.9.3\Marlin\src\inc\MarlinConfigPre.h:39:0,
                                  from E:\Plotter\Marlin-2.0.9.3\Marlin\src\inc\MarlinConfig.h:28,
                                  from E:\Plotter\Marlin-2.0.9.3\Marlin\src\MarlinCore.h:24,
                                  from E:\Plotter\Marlin-2.0.9.3\Marlin\src\MarlinCore.cpp:31:

                 Configuration.h:1267:46: error: could not convert '0' from 'int' to 'XYZval<float>::<anonymous union>'
                  #define NOZZLE_TO_PROBE_OFFSET { 0, 26, -1.5 }
                                                               ^
                 E:\Plotter\Marlin-2.0.9.3\Marlin\src\module\probe.h:217:11: note: in expansion of macro 'NOZZLE_TO_PROBE_OFFSET'
                            NOZZLE_TO_PROBE_OFFSET
                            ^
                 Configuration.h:1267:46: error: too many initializers for 'const xyz_pos_t {aka const XYZval<float>}'
                  #define NOZZLE_TO_PROBE_OFFSET { 0, 26, -1.5 }
                                                               ^
                 E:\Plotter\Marlin-2.0.9.3\Marlin\src\module\probe.h:217:11: note: in expansion of macro 'NOZZLE_TO_PROBE_OFFSET'
                            NOZZLE_TO_PROBE_OFFSET
                            ^
                 In file included from E:\Plotter\Marlin-2.0.9.3\Marlin\src\MarlinCore.cpp:194:0:
                 E:\Plotter\Marlin-2.0.9.3\Marlin\src\module\probe.h:222:117: error: could not convert                 'Probe::build_time::default_probe_xyz_offset.XYZval<float>::<anonymous>.XYZval<float>::<anonymous union>::<anonymous>.XYZval<float>::<anonymous union>::<anonymous struct>::x' from 'const float' to 'XYval<float>::<anonymous union>'

                        static constexpr xy_pos_t default_probe_xy_offset = { default_probe_xyz_offset.x,  default_probe_xyz_offset.y };
                                                                                                                                      ^
                 E:\Plotter\Marlin-2.0.9.3\Marlin\src\module\probe.h:222:117: error: too many initializers for 'const xy_pos_t {aka const XYval<float>}'
                 exit status 1
                 Exception in thread "AWT-EventQueue-0" java.lang.Error: Error: could not match input
                 	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.zzScanError(CPlusPlusTokenMaker.java:2095)
                 	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.yylex(CPlusPlusTokenMaker.java:2343)
                 	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.getTokenList(CPlusPlusTokenMaker.java:1919)
                 	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:431)
                 	at org.fife.ui.rsyntaxtextarea.SyntaxView.getLineWidth(SyntaxView.java:306)
                 	at org.fife.ui.rsyntaxtextarea.SyntaxView.calculateLongestLine(SyntaxView.java:110)
                 	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.refreshFontMetrics(RSyntaxTextArea.java:2106)
                 	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2545)
                 	at processing.app.EditorTab.applyPreferences(EditorTab.java:333)
                 	at processing.app.Editor.lambda$selectTab$52(Editor.java:1457)
                 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
                 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
                 	at java.awt.EventQueue.access$500(EventQueue.java:97)
                 	at java.awt.EventQueue$3.run(EventQueue.java:709)
                 	at java.awt.EventQueue$3.run(EventQueue.java:703)
                 	at java.security.AccessController.doPrivileged(Native Method)
                 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
                 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
                 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
                 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
                 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
                 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
                 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
                 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

                 could not convert '0' from 'int' to 'XYZval<float>::<anonymous union>'
`

I attached my config files.
[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/7854251/Marlin_Config.zip)

Before reporting this issue I've checked previous bug reports and found #16767. Mine seems very similar.

I've tested it on two different computers with the same result. On my laptop I've even tested with a clean installation of everything (including Windows10).

Works for the standard board BOARD_RAMPS_14_EFB when applying neccessary changes in both configs (reducing MIXING_STEPPERS in MIXING_EXTRUDER to 2 and deactivating USE_CONTROLLER_FAN). But it doesn't work for BOARD_RAMPS_DUO_EFB.

### Bug Timeline

On first try with Marlin 2.0.93

### Expected behavior

No compilation errors

### Actual behavior

Compiler Error
maybe type cast errors in compiler definitions

### Steps to Reproduce

Thats what I did on my laptop:
1. Install Aurduino IDE
2. Install Due board configs
3. Unpack latest Marlin firmware
4. Copy config files in Marlin folder
5. Select correct board in Arduino IDE
6. Compile

### Version of Marlin Firmware

2.0.93

### Printer model

Delta

### Electronics

Arduino Due with RADDS shield

### Add-ons

BLtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

ellensp, 2022-01-12:
vscode + platformio  takes 10 seconds to compile
ardduino ide 2m52 on same hardware!
Do yourself a favor and use vscode with platformio

ellensp, 2022-01-12:
In saying that... here is the fix.  3 issues in total.

```DIFF
diff --git a/Marlin/src/lcd/e3v2/enhanced/dwin_popup.cpp b/Marlin/src/lcd/e3v2/enhanced/dwin_popup.cpp
index 89909e5c27..2f9734a90b 100644
--- a/Marlin/src/lcd/e3v2/enhanced/dwin_popup.cpp
+++ b/Marlin/src/lcd/e3v2/enhanced/dwin_popup.cpp
@@ -27,6 +27,10 @@
  * Date: 2021/11/06
  */
 
+
+#include "../../../inc/MarlinConfigPre.h"
+#if ENABLED(DWIN_CREALITY_LCD_ENHANCED)
+
 #include "dwin.h"
 #include "dwin_popup.h"
 
@@ -53,3 +57,4 @@ void DWIN_Popup_ConfirmCancel(const uint8_t icon, FSTR_P const fmsg2) {
   Draw_Select_Highlight(true);
   DWIN_UpdateLCD();
 }
+#endif
diff --git a/Marlin/src/module/probe.h b/Marlin/src/module/probe.h
index 3c97afcb89..d9c50b0125 100644
--- a/Marlin/src/module/probe.h
+++ b/Marlin/src/module/probe.h
@@ -212,14 +212,14 @@ public:
 
     // constexpr helpers used in build-time static_asserts, relying on default probe offsets.
     class build_time {
-      static constexpr xyz_pos_t default_probe_xyz_offset =
+      static constexpr xyz_pos_t default_probe_xyz_offset = xyz_pos_t(
         #if HAS_BED_PROBE
           NOZZLE_TO_PROBE_OFFSET
         #else
           { 0 }
         #endif
-      ;
-      static constexpr xy_pos_t default_probe_xy_offset = { default_probe_xyz_offset.x,  default_probe_xyz_offset.y };
+      );
+      static constexpr xy_pos_t default_probe_xy_offset = xy_pos_t({ default_probe_xyz_offset.x,  default_probe_xyz_offset.y });
 
     public:
       static constexpr bool can_reach(float x, float y) {

```

thisiskeithb, 2022-01-12:
https://github.com/MarlinFirmware/Marlin/pull/23510 has been merged.

closed 2022-01-12 by thisiskeithb