MarlinFirmware/Marlin #23511: Marlin 2.0.9.3 Lerdge K

yzheka, 2022-01-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Tried latest Marlin 2.0.9.3 release on Lerdge K board. after firmware update screen gets stuck at "firmware update ok! ^v^" message. After power cycle the board I get black screen and soft-bricked board. Reverted back to Marlin 2.0.8.1 fixed the board, boots and works ok.

### Bug Timeline

new

### Expected behavior

Marlin 2.0.9.3 should work on Lerdge K board

### Actual behavior

Marlin 2.0.9.3 is not booting on Lerdge K board

### Steps to Reproduce

1. take latest Marlin 2.0.9.3 release
2. set env to LERDGEK_usb_flash_drive in platformio.ini
3. set 
#define MOTHERBOARD BOARD_LERDGE_K
#define TFT_GENERIC
#define TFT_INTERFACE_FSMC
#define TFT_RES_480x320
#define TFT_COLOR_UI
in Configuration.h

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anycubic Kossel Linear Plus

### Electronics

Lerdge K board

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7855814/Configuration.zip)

thinkyhead, 2022-01-17:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

yzheka, 2022-01-17:
Ok. I will try to find but it may take some time. I will get back with some info

Witawat, 2022-01-18:
how to upload file to Lerdge K ?

yzheka, 2022-01-19:
@Witawat copy Lerdge_K_firmware_force.bin to SDcard\Lerdge_K_system\Firmware\ insert sd into board and restart it

yzheka, 2022-01-24:
@thinkyhead sorry, I will not able to find what commit caused the problem. I have sold the printer with that board.

closed 2022-01-24 by thisiskeithb