MarlinFirmware/Marlin #23513: [BUG] DGUS_LCD_UI_MKS with DEBUG_DGUSLCD won't compile

DejitaruJin, 2022-01-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am currently trying to port the MKS DWIN screen firmware to a slightly different screen, and I hoped debugging would help me figure out why it won't do the thing. However, if I enable `DEBUG_DGUSLCD` in Marlin/src/lcd/extui/dgus/DGUSDisplay.h, it encounters errors:
`/mnt/DNI/Downloads/Marlin/Marlin-2.0.9.3-test/Marlin/src/lcd/extui/dgus/mks/DGUSScreenHandler.cpp:759:39: error: 'class GCodeQueue' has no member named 'length'`
`/mnt/DNI/Downloads/Marlin/Marlin-2.0.9.3-test/Marlin/src/core/serial.h:297:75: error: conversion from 'const int16_t {aka const int}' to 'EnsureDouble' is ambiguous`
And a cascade of others that are seemingly just side effects. I could probably fix this myself but I'm already three or four levels deep in "I need to fix A so I can fix B" so handing this one off would be pretty neat. None of the other DWIN options have such an issue.

Incidentally, debugging is _entirely_ broken in the bugfix branch as of right now, so that's of little help. The offending lines still exist in the relevant file, though.

### Bug Timeline

<= 2.0.9.3

### Expected behavior

Compile with DGUS/DWIN debug output enabled

### Actual behavior

Compile error; references nonexistent class member.

### Steps to Reproduce

1. Download Marlin 2.0.9.3 (current)
2. Edit Configuration.h to uncomment `#define DGUS_LCD_UI_MKS` (no other edits needed to test)
3. Edit Marlin/src/lcd/extui/dgus/DGUSDisplay.h to uncomment `#define DEBUG_DGUSLCD`
4. Attempt to compile in the latest environment of your choice

### Version of Marlin Firmware

2.0.9.3

### Printer model

Longer LK4 Pro

### Electronics

Stock

thinkyhead, 2022-01-17:
You have not attached configurations that can be used to build with Marlin bugfix-2.0.x, so this cannot be explored further.

DejitaruJin, 2022-01-17:
I had assumed it'd be easier to simply uncomment line 2618 rather than mess around downloading and copying files, especially since other files need editing anyways, but if you "cannot" investigate the issue without it, I've now provided the Configuration.h I just modified from today's bugfix branch. I've also verified that this same config file (now 0.4K larger?) still produces the same output on the 2.0.9.3 master, irrespective of whatever else has changed in the branch. These errors are identical whether using a config specifically for my machine, or freshly downloaded with only the one option uncommented.

Please bear in mind that even with this assistance, A) You still need to uncomment line 39 of Marlin/src/lcd/extui/dgus/DGUSDisplay.h, and B) The bugfix branch has a more serious underlying issue that seems to affect all DWIN debug output. It would likely be easier to resolve the new issue with, say, `DGUS_LCD_UI_ORIGIN` selected instead, then tackle the bug that had been plaguing MKS specifically. Testing further, options like `PID_DEBUG` and `DEBUG_LEVELING_FEATURE` do _not_ cause a compile error.

DejitaruJin, 2022-01-22:
The _current_ DWIN debug bug, affecting all modules, has been found in commit [755c10d3030ed18762e05b1f86396237bca8dfbf](https://github.com/MarlinFirmware/Marlin/commit/755c10d3030ed18762e05b1f86396237bca8dfbf) . I shall investigate what's broken there, and then move on to the MKS-specific bug that was found initially.

DejitaruJin, 2022-01-22:
In Marlin/src/core/serial.h, I am almost positive that `#define _SELP_1(s) serial_print(F(s "\n"));` is supposed to be `#define _SELP_1(s) serial_println(F(s));`. There's still plenty of errors to go, but the compiler's not mistaken when it says that's invalid syntax.

closed 2022-04-03 by github-actions[bot]

thisiskeithb, 2022-04-03:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/23601

PR #23601 (2022-01-25): Fix debugging in DWIN screens
### Description

Partially reverts commit 755c10d3030ed18762e05b1f86396237bca8dfbf, where `DEBUG_ECHOLNPAIR_F` was swapped to `DEBUG_ECHOLNPGM` in cases where it can not be.

Updates MKS screen handler debugging to be compatible with commit ec42be346d5c0d072feb8a1b63ef2fdbd6dc1e98, which moved queue length into a struct.

I did not see the same issues repeated in the dgus_reloaded code, but I also couldn't find a combination of config options that would pass sanity checks and let it compile either; I didn't edit that code so I'm considering it out-of-scope.

### R [... cut]