MarlinFirmware/Marlin #23516: [BUG] G27 not working

Charmandrigo, 2022-01-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This might not be a bug on the firmware but is still something I don't understand, recently did a dual extruder upgrade to my anet a8 plus and also upgraded the motherboard to a mks gen 1.4, all went good and I can print! but the nozzle_park_feature doesn't seem to work and causes issues when pausing the print.
I have set #define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 10 } which are the defaults of Anet A8 Plus.
I was on marlin 2 before when my printer had the stock Anet motherboard, and ADVANCED_PAUSE_FEATURE worked well too
I am unsure if having dual extruders affect this feature, or can be other things like something on the printer wiring which I doubt 

### Bug Timeline

when upgraded to MKS Gen 1.4 motherboard

### Expected behavior

Park the nozzle to the set NOZZLE_PARK_POINT when pausing

### Actual behavior

if I pause from LCD:
nothing happens, the printer keeps printing BUT the LCD message changes to "PARKING" nothing happens and keeps printing, sometimes the extruder is turned off but keeps doing travel movements forever.

If I pause from a command line:
The print pauses but the nozzle stays in place, it does not park and causes the print to melt on the spot.

### Steps to Reproduce

Executing G27 in any way

### Version of Marlin Firmware

2.0.9.2

### Printer model

Anet A8 Plus

### Electronics

MKS Gen 1.4

### Add-ons

dual e3dv6 clone extruders

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7860961/Configuration.files.zip)

thinkyhead, 2022-01-17:
I'm unable to compile Marlin bugfix-2.0.x using the supplied configuration files. Could you provide configuration files that can be used with the latest code ?

Charmandrigo, 2022-01-18:
> I'm unable to compile Marlin bugfix-2.0.x using the supplied configuration files. Could you provide configuration files that can be used with the latest code ?

Sorry, It might be because I forgot to include my modified pins_ramps.h and pins_mks1_3.h
here's the files again:
[configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/7887602/configuration.files.zip)

thinkyhead, 2022-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2022-05-02 by github-actions[bot]