MarlinFirmware/Marlin #23517: [BUG] M81 not working while printing from SD card

mistic100, 2022-01-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I try to have the printer shuts down automatically at the end of the print (from SD card). So I inserted "M81 S" at the very end of my file.

```
M107
G91 ;Relative positioning
G1 E-30 Z10 F2700 ;Retract and raise Z
G90 ;Absolute positionning
G1 X0 Y235 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
M81 S
;End of Gcode
```

I investigated a bit into M80_M81.cpp and power.cpp but couldn't understand this behavior. Is the firmware preventing to shut downs while the print job is still running ?

### Expected behavior

The printer should wait to reach 50° on the nozzle then trigger the power supply pin.

### Actual behavior

The printer does not shut downs. The menus show "Switch Power ON", indicating the command was somehow acknowledged but it it is still on.

Powerring off from the menu or by sending the command on serial works.

### Version of Marlin Firmware

bugfix-2.0.x 2021-01-08

### Printer model

Creality Ender 3

### Electronics

BTT SKR Mini E3 3.0

### Add-ons

BTT Relay 1.2

### Your Slicer

Cura

### Additional information & file uploads

I am running exactly this version of Marlin https://github.com/mistic100/Marlin/tree/custom

this is the bugfix branch from 2021-01-08 + BTT changes for SKR Mini E3 3.0 + my rencent PR #23476 + custom settings

relevant config :

```cpp
#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  //#define MKS_PWC                 // Using the MKS PWC add-on
  #define PS_OFF_CONFIRM          // Confirm dialog when power off
  //#define PS_OFF_SOUND            // Beep 1s when power off
  #define PSU_ACTIVE_STATE HIGH      // Set 'LOW' for ATX, 'HIGH' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 250   // (ms) Delay for the PSU to warm up to full power

  //#define POWER_OFF_TIMER               // Enable M81 D<seconds> to power off after a delay
  #define POWER_OFF_WAIT_FOR_COOLDOWN   // Enable M81 S to power off only after cooldown

  //#define PSU_POWERUP_GCODE  "M355 S1"  // G-code to run after power-on (e.g., case light on)
  //#define PSU_POWEROFF_GCODE "M355 S0"  // G-code to run before power-off (e.g., case light off)

  //#define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
  #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    #define AUTO_POWER_E_FANS
    #define AUTO_POWER_CONTROLLERFAN
    #define AUTO_POWER_CHAMBER_FAN
    #define AUTO_POWER_COOLER_FAN
    #define POWER_TIMEOUT              30 // (s) Turn off power if the machine is idle for this duration
    //#define POWER_OFF_DELAY          60 // (s) Delay of poweroff after M81 command. Useful to let fans run for extra time.
  #endif
  #if EITHER(AUTO_POWER_CONTROL, POWER_OFF_WAIT_FOR_COOLDOWN)
    #define AUTO_POWER_E_TEMP        50 // (°C) PSU on if any extruder is over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) PSU on if the chamber is over this temperature
    //#define AUTO_POWER_COOLER_TEMP   26 // (°C) PSU on if the cooler is over this temperature
  #endif
#endif
```

ellensp, 2022-01-14:
PS_OFF_CONFIRM when printing from sdcard?

mistic100, 2022-01-14:
This shouldn't be the case PS_OFF_CONFIRM is only used in MarlinUI for screen buttons, not on the gcode execution itself.

I'll check again.

mistic100, 2022-01-14:
I tested another end code

```
M107
G91 ;Relative positioning
G1 E-30 Z10 F2700 ;Retract and raise Z
G90 ;Absolute positionning
G1 X0 Y235 ;Present print
M106 S0 ;Turn-off fan
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
M109 S50 ;Turn-off hotend
M81
;End of Gcode
```

the important bit is waiting for hotend temp with `M109 S50` and using `M81` instead of `M81 S`

so this has something to do with POWER_OFF_WAIT_FOR_COOLDOWN while the job is running

zeleps, 2022-01-16:
@mistic100 are you using a suicide pin to power off?

mistic100, 2022-01-16:
> @mistic100 are you using a suicide pin to power off?

No

zeleps, 2022-01-16:
How's your PSU_ON_PIN defined?

zeleps, 2022-01-16:
There's an omission in the power.cpp code, suicide() is not called when cooldown or delayed power off is selected. But since this isn't the case for you, I'm trying to figure out what might be the issue.

When you print with M81 S, does the menu show "Switch power ON" immediately after the end of print, or after cooldown temp has been reached?

mistic100, 2022-01-16:
> When you print with M81 S, does the menu show "Switch power ON" immediately after the end of print, or after cooldown temp has been reached?

I don't know, each time I was there (sleeping :-) ). I'll make a quick test gcode and report.

zeleps, 2022-01-16:
no need, best way is to heat your hotend partially (up to 60°C) and send an M81 S

mistic100, 2022-01-16:
I am not perfectly sure but here is what I think I saw :

Power off from menu :
- the info screen shows "Ender-3 OFF"
- when 50° is reached, the menu shows "Switch power ON"
- 1-2 seconds later the relay switches

Power off from "M81 S" :
- the info screen shows "Ender-3 OFF"
- when 50° is reached, the menu still shows "Switch power OFF"
- 1-2 seconds later the menu show "Switch power ON" and the relay switches just after that

zeleps, 2022-01-16:
So, M81 S works? A small delay (1-2sec) is normal. Also, the info screen message is issued immediately with the command, so that's also normal.

mistic100, 2022-01-16:
Yes that's what I said in my first message :) the command itself works, just not during a print.

zeleps, 2022-01-16:
Do you have a serial (USB?) connection with the printer? I can give you a build that sends debug output to the serial to see what's going on.

mistic100, 2022-01-16:
I have a ESP-01S running ESP3D connected on the serial port, I can also connect Pronterface.

If you want I could compile the debug build myself, keeping all my config.

zeleps, 2022-01-16:
gimme 5'

mistic100, 2022-01-16:
Hmmm quick update

I runned this test code, basically my start gcode + end gcode 

```
M82 ;absolute extrusion mode
M190 S50 ; Wait for Heat Bed temperature
M104 S200 ; Set Extruder temperature
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 X115 Y115 Z2.0 F3000 ; Go to start position
M109 S200 ; Wait for Extruder temperature
; End of custom start GCode

M107
G91 ;Relative positioning
G1 E-30 Z10 F2700 ;Retract and raise Z
G90 ;Absolute positionning
G1 X0 Y235 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
M81 S
```

And it powered off as expected...  
But it still failed twice on real prints.

Perhaps don't spend more time on the subject, I will try to have a real reproductible scenario.

zeleps, 2022-01-16:
Still, try [this branch](https://github.com/zeleps/Marlin/tree/bugfix-2.0.x-FixPowerOffBySuicide), I've added some serial logging to see what happens with power after M81 is issued.

zeleps, 2022-01-16:
Only power.cpp and M80-M81.cpp are changed.

mistic100, 2022-01-16:
In what conditions to you want I test ? Just sending "M81 S" over serial ?

mistic100, 2022-01-16:
sending

```
M107 S60
M104 S0
M81 S
```

here is the result

```
[heating lines omitted]
ok T:60.17 /60.00 B:22.91 /0.00 @:8 B@:0
M104 S0 T0
ok T:60.18 /0.00 B:22.89 /0.00 @:0 B@:0
M81 S
ok T:59.90 /0.00 B:22.83 /0.00 @:0 B@:0
M106 P0 S0
//action:notification Ender-3 OFF.
M81 command issued.
Delaying power off...
ok T:59.05 /0.00 B:22.68 /0.00 @:0 B@:0
ok T:58.54 /0.00 B:22.54 /0.00 @:0 B@:0
ok T:57.62 /0.00 B:22.97 /0.00 @:0 B@:0
ok T:56.93 /0.00 B:22.79 /0.00 @:0 B@:0
ok T:56.00 /0.00 B:22.83 /0.00 @:0 B@:0
ok T:55.38 /0.00 B:23.05 /0.00 @:0 B@:0
ok T:54.06 /0.00 B:23.12 /0.00 @:0 B@:0
ok T:53.02 /0.00 B:23.18 /0.00 @:0 B@:0
ok T:52.11 /0.00 B:22.89 /0.00 @:0 B@:0
ok T:51.57 /0.00 B:22.77 /0.00 @:0 B@:0
ok T:50.77 /0.00 B:23.05 /0.00 @:0 B@:0
ok T:50.15 /0.00 B:22.64 /0.00 @:0 B@:0
Power off cooling wait complete.
Powering off...
Power off pin triggered
ok T:48.96 /0.00 B:22.87 /0.00 @:0 B@:0
[actual power off]
```

mistic100, 2022-01-16:
All of this is very weird. Currently starting another long print. I'll keep in touch.

Perhaps I am just dumb !

zeleps, 2022-01-16:
This is the correct test and expected behavior. Now let's see how it works when issued after a print.

mistic100, 2022-01-17:
Today printing from SD card (and switched to Pronterface because yesterday the ESP-01S crashed mid-print...)

```
echo:busy: processing
M106 P0 S0
//action:notification Bed Cooling...
//action:notification xyzCalibration_cube.gcode
echo:busy: processing
M106 P0 S0
//action:notification Ender-3 OFF.
//action:notification 31m 27s
Done printing file
//action:prompt_end
//action:prompt_begin Print Done
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
[and so on...]
[nothing more even after reaching 50°C]
```

And **the screen is not showing any prompt**, it is on the info screen with the last notification : "31m 27s"

Anyway if I push the control button once :
```
Power off cooling wait complete.
Powering off...
Power off pin triggered
[actual power off]
```

So this has nothing to do with the power off feature but with `HOST_PROMPT_SUPPORT` I guess, although the config file does not explain what it does.

mistic100, 2022-01-17:
Shouldn't there be a timeout on this prompt like there seems to be the line bellow for `HAS_RESUME_CONTINUE` ?

Anyway I will compile without `HOST_ACTION_COMMANDS` as I have no use for it.

zeleps, 2022-01-17:
Ok, so M81 is ok - there is an issue with suicide() not invoked, I'll open a new pull request for fixing that - so we can close this issue.

mistic100, 2022-01-17:
@zeleps may suggest some optimizations :
- add a compile time warning if `PSU_CONTROL` and `HOST_ACTION_COMMANDS` are enabled together : something along the line "Power-off from M81 command won't work if an host is connected to the printer" (I can do a PR for that)
- add a timeout to this particular prompt in M1001.cpp (I cannot do a PR for that...)
- or add options to choose on which serial ports action commands are possible, this would allow to enable action commands by USB only (I probably could do a PR for that but I have no personal need)

zeleps, 2022-01-17:
@mistic100 I haven't used `HOST_ACTION_COMMANDS`, so I don't really have an opinion on the matter. I intervened in this issue since the recent changes to M81 were partly implemented by me, and I wanted to make sure they worked. I suggest you start a new issue for that conflict, and we'll see who takes it up (I could take a look at it, perhaps this weekend, but it's not something I'm familiar with atm).

zeleps, 2022-01-17:
(idk, I'm just a contributor, I'm not pulling any strings around here)

mistic100, 2022-01-17:
To be fair I read the compilation warnings I got, but I didn't read the documentation from top to bottom (well M80 is short, I could have seen it :-))

Thank you for your time nonetheless

closed 2022-01-17 by mistic100