MarlinFirmware/Marlin #23524: [BUG] Unknown environment names sanguino_atmega1284p

oliskcz, 2022-01-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

**Marlin does not compile in VS Code / Platformio with the following configuration in platformio.ini:**

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = sanguino_atmega1284p
include_dir  = Marlin

**error message:** 

**PS C:\Users\olisk\OneDrive\Počítač\Marlin-2.0.x> platformio run -e BIGTREE_E3_RRF
Error: Unknown environment names 'ATmega1284'. Valid names are 'mks_robin_pro, teensy31, malyan_M300, mks_robin_e3p_maple, FYSETC_CHEETAH_V20, STM32G0B1RE_btt, REMRAM_V1, mks_robin_lite3_maple, mks_robin_e3_maple, mks_monster8_usb_flash_drive_msc, mks_robin_nano_v3_usb_flash_drive, DUE_USB, STM32F103VE, mks_robin_e3, mks_robin_nano_v1_3_f4, mks_robin_lite_maple, chitu_f103_maple, DUE_archim, STM32F103RE_btt, BIGTREE_E3_RRF, STM32F070CB_malyan, STM32F407VE_black, mks_robin_lite3, simulator_linux_debug, LERDGEX_usb_flash_drive, STM32F103RC_meeb, STM32F103RE_btt_USB_maple, sanguino1284p, Anet_ET4_OpenBLT, FYSETC_S6_8000, simulator_macos_release, STM32F103RC_ZM3E2_USB, ASTM32F103VE_longer_maple, mks_eagle_usb_flash_drive_msc, mingda_mpx_arm_mini, FLYF407ZG, mks_robin_e3p, LPC1768, simulator_macos_debug, mks_robin_nano35_maple, mks_robin_nano35, STM32F103RC_btt_USB_maple, MightyBoard1280, FLY_MINI, BIGTREE_OCTOPUS_V1, BTT_SKR_SE_BX, BIGTREE_SKR_2_F429, ERYONE_ERY32_MINI_maple, jgaurora_a5s_a1, mks_robin_nano_v3_usb_flash_drive_msc, rumba32, STM32F103RC_btt_maple, simulator_windows, STM32F103RET6_creality, BIGTREE_SKR_PRO, mega2560ext, LERDGES_usb_flash_drive, FYSETC_F6, STM32F103ZE, BIGTREE_BTT002, teensy41, DUE, mega1280, LERDGEK_usb_flash_drive, sanguino644p, Artillery_Ruby, STM32F103CB_malyan, STM32F103VC_ZM3E4_USB, STM32F103RC_btt, STM32F103RE_btt_maple, at90usb1286_dfu, mks_monster8_usb_flash_drive, sanguino1284p_optimized, BIGTREE_OCTOPUS_PRO_V1_F429, include_tree, BIGTREE_SKR_2_F429_USB, mks_robin_pro2, LERDGEK, mks_eagle_usb_flash_drive, esp32, STM32F103RET6_creality_maple, trigorilla_pro, mks_robin_mini, STM32F070RB_malyan, BIGTREE_SKR_PRO_usb_flash_drive, melzi_optiboot, BIGTREE_BTT002_VET6, teensy20, BIGTREE_GTR_V1_0, SAMD51_grandcentral_m4, teensy35, LERDGES, chitu_v5_gpio_init_maple, mks_robin_pro_maple, rambo, BIGTREE_OCTOPUS_V1_USB, BIGTREE_GTR_V1_0_usb_flash_drive, STM32F103VE_longer, trigorilla_pro_maple, mks_eagle, STM32F103RE_maple, STM32F103CB_malyan_maple, STM32F103RC_fysetc, melzi, STM32F103VE_GTM32, at90usb1286_cdc, LERDGEX, LPC1769, chitu_f103, BIGTREE_SKR_2_USB, STM32F103RC_ZM3E2_USB_maple, FLY_MINI_maple, FYSETC_S6, STM32F103RE_btt_USB, STM32F103VE_ZM3E4V2_USB, BIGTREE_SKR_2_F429_USB_debug, mks_robin, mks_robin_mini_maple, linux_native, NUCLEO_F767ZI, chitu_v5_gpio_init, mks_robin_nano_v3, teensy36, melzi_optimized'**

### Bug Timeline

Today.

### Expected behavior

I expected it to work.

### Actual behavior

It didn't work and gave me error.

### Steps to Reproduce

1. Have **Visual Studio Code**
2. idk...

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3.

### Electronics

Stock electronics.

### Add-ons

BL touch.

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

thisiskeithb, 2022-01-14:
> default_envs = sanguino_atmega1284p
> PS C:\Users\olisk\OneDrive\Počítač\Marlin-2.0.x> platformio run -e BIGTREE_E3_RRF
Error: Unknown environment names 'ATmega1284'.

Not only do these not match, you need to match the correct PIO environment to your motherboard. I suggest installing [Auto Build Marlin](https://marlinfw.org/docs/basics/auto_build_marlin.html) which will present the correct environments for the enabled motherboard.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-01-14 by thisiskeithb

oliskcz, 2022-01-14:
I already have installed Auto Build Marlin...