MarlinFirmware/Marlin #23525: [BUG] Bug in Thermal Runaway. No message on LCD

roel8032, 2022-01-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Bug in Thermal Runaway. When it happens, the usual beeps. But no message on LCD. So you can't see what caused it. The nozzle retreats. Looks new to me. You can test this if you place a switch in the cable to bed or nozzle heater.

### Expected behavior

Trigger alarm and report on LCD as usual.

### Actual behavior

No message on LCD.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Infitary M506

### Electronics

Stock board

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

thinkyhead, 2022-01-17:
You have provided no helpful information such as your configuration or your printer or LCD model, so this is not actionable.

sbarajas, 2022-01-18:
I have this problem too.

Printer Model: Anycubic Predator
Electronics: Trigorilla Pro
Version: 2.0.9.3
Step to Reproduce: When the target nozzle temperature is higher than 200 Celsius Degrees

In the screen appears a message that says something that the printer is halted because a thermal runaway problem

roel8032, 2022-01-19:
If the alarm is triggered, the printer will be reset.

roel8032, 2022-01-19:
[image]

In Marlin 2.0.9.2 this screen appears.

mzhu1113, 2022-01-24:
I have this problem as well testing 2.0.9.3 on a test printer for validation before deploying to fleet.

Forced printer to trigger a thermal runaway condition via both E0 and Bed, printer will beep and briefly display the appropriate message while retreating the toolhead away from current position to XY max. The printer will reset once toolhead parks.

SOFT_RESET_VIA_SERIAL and SOFT_RESET_ON_KILL are not enabled in configuration_adv

CR10StockDisplay
3 linear axis
Typical Ender 3 setup with SKR V2.0 board.

roel8032, 2022-03-26:
Problem is fixed yet?

closed 2022-06-05 by github-actions[bot]