MarlinFirmware/Marlin #23531: Retract Problem in 2.0.9.3

Steve021171, 2022-01-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

it looks like there is a problem with the retract in Marlin 2.0.9.3. The printer produces a lot of little holes in the wall.
[image]
On the left side is a hollow cube, printed with 2.0.9.3, on the right side is the same cube (same gcode over Repetier-server) printed with 2.0.9.2.
I turned of LA (with M900 K0), then you get fewer holes, but it is still there.

If you print in vase mode this problem does not occur:
[image]

The printer is an Artillery Sidewinder X1, with MKS SGEN L mainboard (32Bit), TMC2209 in UART, BMG Wind extruder and ABL

I made a Marlin 2.0.9.3 for a friend, he has an Artillery Genius with stock hardware (MKS GEN L - 8Bit board, TMC2100, Titan extruder), and it seems he has not that particular problem.  

Also, I recognized there is a problem with the Z-Offset when #define RESTORE_LEVELING_AFTER_G28 is activated. It will mess up your offset and the nozzle is too close to the bed or far to high. 
Deactivating this option fixes the problem. So that doesn't bother me, but I thought it is worth to be mentioned...

### Bug Timeline

01/13/2022

### Expected behavior

clean nice walls

### Actual behavior

lots of small holes in wall

### Steps to Reproduce

printed serveral test cubes

### Version of Marlin Firmware

2.0.9.3

### Printer model

Artillery Sidewinder X1

### Electronics

MKS SGEN L, TMC2209

### Add-ons

BMG Wind extruder with BiMetal heartbreak and MicroSwiss nozzle 0.6mm

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Additional information & file uploads

[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/7875461/Archiv.zip)
config files

thinkyhead, 2022-01-17:
See if it helps to disable `POWER_LOSS_RECOVERY`.

Steve021171, 2022-01-17:
> See if it helps to disable `POWER_LOSS_RECOVERY`.

That makes no sense. Can you explain why I should try that ?

Dooces, 2022-01-18:
just a shot in the dark, but since he knows what he's talking about, lets brainstorm to potentially not waste his time with having to answer...
POWER_LOSS_RECOVERY -
perhaps requires a lot of CPU time, causing hiccups?
perhaps there are other known issues that involve extrusion, and he wants to know if he should spend time investigating a new issue, or if its likely within the same scope of what he's doing now?
perhaps there is an interrupt conflict with the sidewinder, or other mainboard chip, where when power loss recover is enabled, there are random conflicts?

Can i explain why i should try that?  Yes.  It was suggested by thinkyhead.  he makes things work.  im guessing he would not have wasted his time if he did not think it had a very good chance of solving the problem.

Steve021171, 2022-01-18:
I will try that today when I'm back from work.

bisky00, 2022-01-18:
My freiend has a Geeetech A20T, I compiled a Firmware for him. He says that te retract is about the double the value set in the slicer.

descipher, 2022-01-19:
Your settings seem excessive try scaling back.

#define DEFAULT_XJERK                 8.0
#define DEFAULT_YJERK                 8.0
#define DEFAULT_ZJERK                 0.3
#define DEFAULT_EJERK                 5.0

Steve021171, 2022-01-19:
> Your settings seem excessive try scaling back.
> 
> #define DEFAULT_XJERK 8.0 #define DEFAULT_YJERK 8.0 #define DEFAULT_ZJERK 0.3 #define DEFAULT_EJERK 5.0

Tried that already, no change...

closed 2022-03-31 by github-actions[bot]