MarlinFirmware/Marlin #23532: [BUG] when fan starts 3d printing stops or slow down

costasdk, 2022-01-15:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Incident : After printed first level when object cooling fan started, printing freezes .
(2nd attempt) I power off printer anet a8, i re-upload software, i reopen repetier host V.2.2.3 and re-sliced with slic3r (1.3.1-dev build commit 8c69278 ), same thing, after first level that object cooling fan starts, printing freezes .
(3rd attempt) i sliced object directly from slic3r and open from repetier-host to print it .
It started printing, when first object cooling fan started i had no issue but after fan stopped and started again after a while printing got very very slow motion that lasted 5-10 seconds and then it started printing normally .
[gcode_slic3r_that_printed_after_slow_motion.zip](https://github.com/MarlinFirmware/Marlin/files/7875639/gcode_slic3r_that_printed_after_slow_motion.zip)
[M115.zip](https://github.com/MarlinFirmware/Marlin/files/7875640/M115.zip)
[gcode_repetier.zip](https://github.com/MarlinFirmware/Marlin/files/7875641/gcode_repetier.zip)
[gcode_extruded_4rth_attempt.zip](https://github.com/MarlinFirmware/Marlin/files/7875642/gcode_extruded_4rth_attempt.zip)
[anet_a8_filament_holder_v2.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/7875643/anet_a8_filament_holder_v2.gcode.zip)
[ConfigurationFiles.zip](https://github.com/MarlinFirmware/Marlin/files/7875644/ConfigurationFiles.zip)

### Bug Timeline

NEW

### Expected behavior

Not to stop printing when object cooling fan starts .
This has never happened again .

### Actual behavior

Stopped printing or slowed down when object cooling fan started .

### Steps to Reproduce

I compiled marlin.ino from marlin version 2.0.9.3 with arduino 1.8.19-linux64 and configuration files for anet a8 from marlin also 2.0.9.3 in debian Buster x86_64 .
i have 0.8mm extruder and nothing else changed .
i had success after one week of trying to set up and configure standard autolevel probe, 
i tried to print anet_a8_filament_holder_v2.stl . 

I dont know if those warnings when compiling marlin has to do with the bug :
/home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/lcd/marlinui.cpp:1194:68: warning: integer overflow in expression [-Woverflow]
                                   adc_other_button = HAL_ADC_RANGE * 1000 / 1024;
/home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/lcd/marlinui.cpp:1194:77: warning: overflow in constant expression [-fpermissive]
                                   adc_other_button = HAL_ADC_RANGE * 1000 / 1024;
                                                                             ^~~~
/home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/lcd/marlinui.cpp:1194:77: warning: overflow in constant expression [-fpermissive]
/home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/lcd/marlinui.cpp:1194:77: warning: overflow in constant expression [-fpermissive]

Sketch uses 112648 bytes (88%) of program storage space. Maximum is 126976 bytes.
Global variables use 3467 bytes of dynamic memory.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Anet A8

### Electronics

Stock Electronics, standard probe sensor

### Add-ons

no

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Repetier Host

### Additional information & file uploads

thanks in advance

descipher, 2022-01-16:
PR #23520 does fix a fan issue, more space usage is required for some of the current 2.0.9.3 code. The Mega1284 will be tight. No issues with those warnings.

You can test it if you're interested.
https://github.com/descipher/Marlin/archive/refs/heads/1284.fan.fix.zip
or git clone -b 1284.fan.fix https://github.com/descipher/Marlin.git bugtest

costasdk, 2022-01-16:
> PR #23520 does fix a fan issue, more space usage is required for some of the current 2.0.9.3 code. The Mega1284 will be tight. No issues with those warnings.
>
> You can test it if you're interested. [...]

Thanks a lot, i will try it .

costasdk, 2022-01-16:
> https://github.com/descipher/Marlin/archive/refs/heads/

i overwrite old libraries and now i cannot compile due to multiple errors repeted :
n file included from /home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/board.h:54:0,
                 from /home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/lwipopts.h:44,
                 from /home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/lwip/opt.h:47,
                 from /home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/netdb.h:32,
                 from /home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/bs_type.h:27,
                 from /home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/bs.h:10,
                 from /home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/mczip.h:12,
                 from /home/user/arduino-1.8.19/Marlin-2.0.x/Marlin/src/gcode/eeprom/M500-M504.cpp:30:
/home/user/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr32/io.h:161:8: error: #error No known AVR32 part defined
 #      error No known AVR32 part defined
        ^~~~~
i am afraid that i will not find again, all -probably 100- headers that i found in the net to compile marlin.ino like bs.h, bs_type.h mczip.h and so on .....

ellensp, 2022-01-16:
do yourself a favor an drop arduunio ide.
Use platformio under vscode https://marlinfw.org/docs/basics/install_platformio_vscode.html

thinkyhead, 2022-01-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-01-17 by thinkyhead