MarlinFirmware/Marlin #23533: Lets move Pause / resume print away from stop print.  better yet lets make the menu layout editable in marlin

kpdelucia, 2022-01-15:
### Is your feature request related to a problem? Please describe.

when using the standard LCD smart full graphic smart controller,  sometimes when you push the knob after selecting an item it selects the choice below it because you ever so slightly rotated the knob when you pushed down.  so there have been many times I started printing the wrong file and a few times I aborted the print by mistake because stop is right near resume and pause.  

### Are you looking for hardware support?

smart full graphic controller

### Describe the feature you want

can you make a feature that allows us to select the order or menu item layout.  for instance If printers worked perfectly The only selection i need is print from SD card.  So why isn't that the first item on the main menu?.  The most popular menu commands should be fastest to get to with the least number of clicks.

ellensp, 2022-01-15:
In Configuration_adv.h  #define MEDIA_MENU_AT_TOP

kpdelucia, 2022-01-16:
ok I will try that and see what it does.  thank you

DejitaruJin, 2022-01-16:
This is a very neat idea, but to do that in anything more than an extremely basic way, you're looking at some additional config file (XML maybe?) and then an additional script to parse that into something that can be compiled with the rest.

I thought I'd seen a config option for adjusting such rotary encoders, but it looks like those relate to motors, not the control knob. Might be useful to have a "minimum n ticks in x millisecond" required to register a menu change.

ellensp, 2022-01-16:
re-arranging menus would be a mammoth task. there are 10 ish displays types all running unique code (since they are all vastly different displays ) Also this would invalidate all current documentation, How can you document where any option is in the menu when it can move?

kpdelucia, 2022-01-16:
Ok so like an ASUS motherboard menu you can create favorites. Like if you put preheat pla on the top menu. It is just a macro that enters top menu, prepare, preheat, preheat PLA all at once.  And you arrive at your command.  There is a chain of selections that leads to each selection.  Or the customization could just be the order of each sub menu.  I was really just looking to not be so paranoid when I hit resume.  A simple "are you sure you want to stop the print?" Would be a nice safety check.

DejitaruJin, 2022-01-16:
Confirmation dialogs, too, are a pretty good idea.

Adding to the complexity of rearranging menus, I believe progressively more printers are starting to use DWIN modules? Which are something like The Worst Thing Ever™ when it comes to open-source mods.

thinkyhead, 2022-01-17:
Making some system to rearrange menu items is not really needed, because the code that describes menus for MarlinUI (`src/lcd/menu/...`) is staggeringly straightforward and easy to edit. You can rearrange things at will by modifying those files.