MarlinFirmware/Marlin #23535: [FR] Save "Tune" parameters in the "POWER_LOSS_RECOVERY" SD file

zkrx, 2022-01-16:
### Describe the feature you want

I use POWER_LOSS_RECOVERY (along with NOZZLE_PARK_FEATURE) to save the current job state on the SD card. That way, I can turn off my printer overnight and resume the next day, for obvious safety concerns.

I noticed this morning that "Tune" parameters (flow, feed rate, ...) were not saved and needed to be manually set upon restart. Omitting that step may potentially ruin a print (unfortunately, I am still tuning my printer and my default flow rate is currently insufficient). I think that these "Tune" settings should be saved in the recovery file too.

You may find my config on my personal branch, based on 2.0.9.2: https://github.com/zkrx/Marlin/tree/zkrx_migbot/Marlin

I contemplated filing a bug report but went with a feature request instead.

thinkyhead, 2022-01-17:
The PLR file ought to be as small as possible, but sure, you could store all sorts of adjustable parameters in there. I'll tag this as a good project for a beginner, and someone may take it on soon.

vovodroid, 2024-02-20:
I would extend it to all current configuration (at least in the case it was updated since last save). During the print many such parameters could be changed, not only tuning values. E.g. accelerations, max.speed, jerk, non-linear extrusion coefficients, etc.

> The PLR file ought to be as small as possible

Why? It's purpose to continue print  with the same parameters as before outage. I don't think a few tens of bytes will change something.

buppgard, 2024-06-02:
It's been a while since there was any activity on this feature request. I've started working on it and added a few lines in powerloss.h and powerloss.cpp to store and restore whatever the current feed rate percentage is. I want to at least implement storing and restoring anything that could be changed in the Tune menu. I should be able to do that within the next few weeks.

vovodroid, 2024-06-03:
Hi @buppgard,

I've add some info to PLR, check it https://github.com/vovodroid/Marlin/commit/13989b24626cdd7e6c9dbe70461134fdd28091a9

And also I have saving and loading all M503 config https://github.com/vovodroid/Marlin/commit/1cfc65784a7e307ee867e798079c50cf2f516f4e

buppgard, 2024-06-03:
That's cool. I like the implementation. I was working on this for a project that's required in a software engineering course that I'm taking at UF - they're asking us to submit a contribution to an open source project. If you already have this one taken care of, I may have to find another feature or bug to work on. Let me check with the TA's.

buppgard, 2024-06-03:
@vovodroid the professor for the course got back to me. As far as the assignment goes, whatever I modify, doesn't even need to be submitted as a pull request. They do want me to write the code without outside help. I'll keep going with what I was working on. Once I turn it in, I'll share what I end up with.

closed 2024-06-26 by thinkyhead