MarlinFirmware/Marlin #23537: CoreXY (Sensorless) Homing using single stepper

IamSLyToo, 2022-01-16:
### Describe the feature you want

Hello,
I would like Marlin to allow me homing the X and Y axis on my CoreXY printer this way: 
 - De-energise one of the XY stepper
 - Rotate the second XY stepper -> The print head carriage will end up in one of the corner
 - Either:
     - The Sensorless capable driver triggers as the stepper cannot be rotated any further
     Or
     - The X and Y end-of-course microswitches are both triggered
 - Re-energise the first XY stepper
 - X and Y are now homed 
Cheers! :)

thinkyhead, 2022-01-17:
What is the difference between de-energizing one of the stepper drivers and simply not moving one of the steppers while it is still energized? How does this differ from just using `QUICK_HOME` with core kinematics?

IamSLyToo, 2022-01-18:
Hello,
The difference is that if you keep the first sepper energized, the hot-end carriage will stop as soon as it gets to the end-stop of any X or Y rail. If you have it de-energized, the carriage will do the same, but then, continue until it reaches the second end-stop, placing itself in one of the corner. X an Y homing is therefore done as a single action. 
Furthermore, it would be more reliable in case of a CoreXY sensorless homing because if you home X, then Y afterwards, there is a possibility that the second homing would skip a step on one of the stepper only and this way mess with the first homing ...

Sylvain ✈️