MarlinFirmware/Marlin #23546: Controllable Fan not working as intented

TheRealStock, 2022-01-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The part-cooling fan(radial fan) on a Creality CR10S is not working correct. In fact it is not controllable even tho it is in the right PIN on the CR V2.2.1 board. I am using the pre-made config for a CR 10 V2 

### Expected behavior

I expect to controll the speed of the fan.

### Actual behavior

Always blasting on 100%

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality CR10

### Electronics

upgraded board, CR V2.2.1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7883184/configs.zip)

descipher, 2022-01-17:
What do you think is the right pin? e.g. How would anyone else know you have the right pin?
What is the last version that worked for you?

TheRealStock, 2022-01-17:
[image]
Excuse me for beeing to unprecise! Completely new here, and also figured out by now that I am in the wrong sub-thread.
However as explained, I am using the CR V2.2.1 board wich has this pin-layout. I am using the Pins for "Controllable Fan". I tried both pins, but there is no change in behaviour. I also tried using the VIN/Fan pins above but as I expected, no change.
In terms of version: that is the problematic part: I used to run the printer on the stock mainboard and since I "upgraded" I have the problem.

descipher, 2022-01-17:
Comes down to you need to know what actual ATMega2560 pins the controller fans connect to in order to configure Marlin.
Since there are no public "Creality Schematics" for that board, you could ask Creality that question.

TheRealStock, 2022-01-17:
Well, thank you! Than I am going to write an Email towards creality. Sorry for the wrong thread here!

ellensp, 2022-01-18:
I wonder if this is related to the recent mosfet order changes
Marlin/src/pins/ramps/pins_RAMPS_CREALITY.h has this

```CPP
// Power outputs EFBF or EFBE
#define MOSFET_D_PIN      7
```
but MOSFET_D_PIN is then never used

I wonder if your fan is actually on this mosfet.

As a test add `#define FAN1_PIN 7`  to you Configuration.h  (add it under    #define MOTHERBOARD BOARD_RAMPS_CREALITY)

now try to manually control it with M107 P1 and M106 P1 S255   and other S values, does this change your fan speeds?

TheRealStock, 2022-01-18:
@ellensp Thank you for your reply and idea!
I have choosen a sort of diffrent approach, since I found someone who has the same board and used his configuration file(it was an older version, 2.0.7.x if I remember right). While his fan was nicely controllable, mine was not.

Thats why I do not believe that it is a problem caused by the software or any sort of a bug, more like that the fan is broken or anything else related to the hardware side. Ordered some new fans and hope that this will change anything.
If this won't help, I will try your idea and if this won't make any change, well than screw it and I will buy a new board.

TheRealStock, 2022-01-21:
@ellensp My hardware theory turns out to be wrong, gonna test. The new 24V fan just kept blowing and was not controllable.
Gonna try out your idea next!

thisiskeithb, 2022-01-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23418

closed 2022-01-24 by thisiskeithb