MarlinFirmware/Marlin #23554: Reverse order of orders of magnitude in menus

Dooces, 2022-01-18:
### Is your feature request related to a problem? Please describe.

I dont know if its the LCD or marlin, i apologize if im barking up the wrong tree.

when i want to move something, i have to start at the smallest increment, then increase from there.  
like if i was backing a car into driveway, and i started going incredibly slow, then speeding up as i got closer... its backwards.  why would you not always start with the largest increment first?  and why have to cycle through all the options to go one resolution smaller?  

its by far the most annoying thing, otherwise this is amazing software, you guys are amazing.  stay out of UI designing though.  you are terrible at it.  

everything else though.... you guys deserve more money than i donate!

### Describe the feature you want

reverse increment order on everything.  start big, then go smaller, instead of the inverse.

### Additional context

why would i want to start incrementing my temperature, from 0, by 1°??  does someone think this is a safety feature?  really there must be something i am missing.

ellensp, 2022-01-19:
Imagine moving Z down, you start off big, oops you just berried the extruder into the bed...   so at least in some places it is needed.

Dooces, 2022-01-19:
yup, i can imagine scenerios where things can go bad, not worth designing a UI over imagined scenerio.  put in an option to reverse it perhaps?

imagine someone flies a plane into a building, imagine someone drives a car into a parade.  these things happen, designing around the worst case scenario is not reasonable.

IMHO anyway.

ellensp, 2022-01-19:
It comes down to assumptions

your assuming that your are stating the move far from the destination or limits
we have to presume the move is close to the destination or limits to avoid hardware damage.  

I think your position of "designing around the worst case scenario is not reasonable" is very very wrong.

Dooces, 2022-01-19:
We can agree to disagree though, but the cost of everyone spending extra time everytime they change temperature or movement comes down to cost.  if the amount of damage caused by having it reverse would be catastrophic, then i agree, its worth having it in reverse order.  You believe that would happen often, I find that incredibly unlikely, if this is where we differ, so be it, i wont be able to see things from your view and we can leave it here.

thank you for your time.