MarlinFirmware/Marlin #23555: [FR] fysetc v2.1 on mks gen L not working

coolum86, 2022-01-18:
### Is your feature request related to a problem? Please describe.

i have fysetc 12864 mini v2.1 and im trying to get it to work with my mks gen L 1.4 for my anet printer. the panel ligths up and rgb test works but no text is displayed. i got to a point wher it was displayed very very dim but i dont know how i got there. please help.

### Describe the feature you want

fysetc on mks gen L

ellensp, 2022-01-18:
It should just plug in.

I have a ramps (plus lcd  board with keys on the right just like the mks gen L) and the fysetc mini v2.1 
I enabled #define FYSETC_MINI_12864_2_1   

It works straight away

This lcd does have contrast settings
Try sending g-code M250 C220

coolum86, 2022-01-18:
> It definitely is supported. so this is not a feature request
> 
> It should just plug in.
> 
> I have a ramps (plus lcd board with keys on the right just like the mks gen L) and the fysetc mini v2.1 Your board is mostly ramps compatible, exp1 and exp2 use identical pins
> 
> I enabled #define FYSETC_MINI_12864_2_1
> 
> It works straight away
> 
> This lcd does have contrast settings Try sending g-code M250 C220 And then save it with M500

i did m250 but what is c220? and where do i put them? conditionals lcd H?

ellensp, 2022-01-19:
It g-code. The codes the printer runs on
You send it to the printer from a control program
M250 is the gcode, C220 is the option to the gcode ie set contrast to 220 
You send both together as one line.

https://marlinfw.org/docs/gcode/M250.html

ellensp, 2022-01-19:
It g-code. The codes the printer runs on
You send it to the printer from a control program, such a pronterface or console on octoprint.
M250 is the gcode, C220 is the option to the gcode ie set contrast to 220 
You send both together as one line.

M250 documattion https://marlinfw.org/docs/gcode/M250.html
overview of gcodes https://www.youtube.com/watch?v=2TByiMNduss 
overview of pronterface https://www.youtube.com/watch?v=lwENk8bxtU4

coolum86, 2022-01-19:
> It g-code. The codes the printer runs on You send it to the printer from a control program, such a pronterface or console on octoprint. M250 is the gcode, C220 is the option to the gcode ie set [...]

ok so i got 2 saves one with my printer settings and the lcd worksd now but cant control it. the second save is only the basic functions but the lcd works and controls work its just not made for my printer settings and doesnt have all of the files. i tried copying and pasting my setting from the one file over but ended up in a bunch of errors. any suggestions?

coolum86, 2022-01-19:
what .h file has the lcd controls?

closed 2022-01-19 by ellensp