MarlinFirmware/Marlin #23556: [BUG] Turning off sound in menu makes speaker beep continuously

SparkyDan555, 2022-01-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Disabling "Sound" under the configuration menu makes the beeper sound continuously. 

### Bug Timeline

17/1/22

### Expected behavior

Turning off sound in the menu prevents beeper sounding for UI feedback and M300.

### Actual behavior

The beeper sounds continuously until "Sound" is turned back on. When turned on it functions as expected, only sounding for UI feedback and M300.

### Steps to Reproduce

1. Enable #define SOUND_MENU_ITEM
2. Upload to printer
3. Navigate to Configuration > Sound
4. Toggle to "Off"

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3

### Electronics

SKR Mini E3 V1.2 - Stock LCD

### Add-ons

N/A

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/7888103/Configuration.Files.zip)

SparkyDan555, 2022-01-18:
I have also just discovered this happens after starting a PID autotune from the LCD menu.

SparkyDan555, 2022-01-18:
This behaviour does not occur when the buzzer is defined as a speaker.

thinkyhead, 2022-04-21:
Perhaps your board has an "active low" buzzer. With `DIRECT_PIN_CONTROL` enabled you can experiment to see how the speaker's pin behaves if you set it to different values, where it might even activate PWM.

SparkyDan555, 2022-04-21:
It turns out my buzzer is indeed active low, but since it is a passive version it needs the PWM signal to drive it. Is there a way to invert the output for the beeper pin through configuration?

SparkyDan555, 2022-06-27:
Commenting to keep issue open. This is still an issue with no way to invert the beeper pin.

thisiskeithb, 2023-09-17:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/26278

closed 2023-09-17 by thisiskeithb