MarlinFirmware/Marlin #23559: [BUG] M600 doesn't resume at the correct height

dmarnel, 2022-01-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When a filament change (M600) is complete and the print is resumed, the z-position is too high by ~0.2mm.

I tried debugging this the last couple of days by editing the gcode and adding absolute positioning commands so the print head should have been right where it left off after the M600 call, but no matter what I tried, the z-position was still too high after resuming the print. The printer's display even reported the correct layer height after resuming. This makes me think that there is a G92 (Set Position) or similar command being issued at some point during the M600 filament change sequence which causes all subsequent height values to be off by ~0.2mm.

### Bug Timeline

This started happening sometime after March 2021, the last time I successfully used this feature.

### Expected behavior

The print should resume at the correct layer height following the M600 command.

### Actual behavior

The print resumes about 0.2mm too high, leaving a small gap in the print where the filament change occurred.

### Steps to Reproduce

Print using the M600 command to change filament colors
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7892111/Archive.zip)
.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 Pro

### Electronics

BIGTREETECH_TFT35_V3.0

### Your Slicer

Cura

### Host Software

OctoPrint

Guilouz, 2022-01-19:
Just used today M600 feature with BigTreeTech screen (latest firmware) and Marlin latest bugfix without issue.

dmarnel, 2022-01-19:
@Guilouz, If you mean the level_z_pos, I already had that set to 0.0. 

The attached photo shows the gap between filament changes. The object in the photo was removed from the print bed for the photo to better show the gap. It's dimensions (in millimeters) are 10 x 10 x 2. The layer height is 0.2mm and there are 10 layers. The filament change occurs at layer 6 (height 1.2mm).

[image]

Guilouz, 2022-01-24:
Make sure you used latest BTT firmware from master. I don't have issue with two variants of BTT screens (TFT43 and TFT70), M600 working as expected.

dmarnel, 2022-01-24:
@Guilouz, thanks for the help. Just to be clear, I am using the latest Marlin firmware: bugfix-2.0.x, and the latest TTF35 firmware: BIGTREE_TFT35_V3.0.27.x.bin, both updated this morning, and am still seeing the same issue.

Bob-the-Kuhn, 2022-02-05:
Reminds me of pictures of **abs to bed** and **layer adhesion** issues I've seen.  

Does all look good when printing but delaminates after cooling down?  The top looks like it was smooth at one point but is now wavy.

Long shot - increase hot end temp by 10 degrees?  Hopefully will have better layer-layer adhesion.

dmarnel, 2022-02-05:
Hi Bob. It prints fine up until layer 6. After I do the filament change (using the same color due to testing) I can see the print nozzle move back into position but it ends up about 0.2mm higher than it should. It then proceeds to print with the air gap as you can see from my photo. The wavy top is due to the air gap while printing the rest of the layers.

kisslorand, 2022-02-12:
Do you have "Emulated M600" enabled in the TFT's firmware? (config.ini -> "emulated_m600:1")

dmarnel, 2022-02-14:
@kisslorand I have tried with and without M600 emulation, neither works. I've even tried adding gcode for absolute positioning just before printing layer 6. The odd thing is that the LCD display shows that the height of layer 6 is just what it should be: 1.2mm. I'll make another build of both Marlin and BTT and test it again.

closed 2022-06-05 by github-actions[bot]