MarlinFirmware/Marlin #23561: [BUG] Compile error MKS TinyBee with MKS Mini 12864 V3

wirebeard, 2022-01-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Cant compile Marlin firmware for MKS TinyBee motherboard and graphical LCD MKS Mini 12864 v3 with Neopixel RGB backlight

```
.pio/build/mks_tinybee/firmware.elf
.pio/build/mks_tinybee/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_eps_hw_spi_fn(_u8g_t*, unsigned char, unsigned char, void*)'
```

### Bug Timeline

After added in release 2.0.9.3 MKS TinyBee

### Expected behavior

Compile firmware

### Actual behavior

Compile error

### Steps to Reproduce

1. Download Marlin 2.0.9.3 (current)
2. Edit Configuration.h to uncomment #define MKS_MINI_12864_V3;  #define NEOPIXEL_LED
3.  Attempt to compile in platformio

### Version of Marlin Firmware

2.0.9.3

### Electronics

Motherboard MKS TinyBee

### Add-ons

LCD MKS Mini 12864 V3

### Additional information & file uploads

[image]
[MKS_TinyBee.zip](https://github.com/MarlinFirmware/Marlin/files/7895669/MKS_TinyBee.zip)

ellensp, 2022-01-19:
Orriginal MKS firmware on https://github.com/makerbase-mks/MKS-TinyBee also fails to compile if you enable MKS_MINI_12864_V3

ellensp, 2022-01-19:
You need to edit 1 file and add annother missing file with edits.

Here is the diff for the edit on src/lcd/dogm/HAL_LCD_com_defines.h

```DIFF
diff --git a/Marlin/src/lcd/dogm/HAL_LCD_com_defines.h b/Marlin/src/lcd/dogm/HAL_LCD_com_defines.h
index a30dd4ca17..27cc31da29 100644
--- a/Marlin/src/lcd/dogm/HAL_LCD_com_defines.h
+++ b/Marlin/src/lcd/dogm/HAL_LCD_com_defines.h
@@ -57,7 +57,12 @@
     #define U8G_COM_HAL_SW_SPI_FN     u8g_com_std_sw_spi_fn
     #define U8G_COM_HAL_HW_SPI_FN     u8g_com_stm32duino_hw_spi_fn
 
-  #elif defined(__AVR__)
+  #elif defined(ESP32)
+
+  uint8_t u8g_eps_hw_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
+  #define U8G_COM_HAL_HW_SPI_FN     u8g_eps_hw_spi_fn
+
+  #else defined(__AVR__)
 
     uint8_t u8g_com_HAL_AVR_sw_sp_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
     #define U8G_COM_HAL_SW_SPI_FN     u8g_com_HAL_AVR_sw_sp_fn

```

Unzip this file add into  Mrlin/src/HAL/ESP32/u8g_esp32_spi.cpp
[u8g_esp32_spi.zip](https://github.com/MarlinFirmware/Marlin/files/7896517/u8g_esp32_spi.zip)

ellensp, 2022-01-19:
Is now a PR awaiting review and merge

thisiskeithb, 2022-02-22:
#23562 has been merged.

closed 2022-02-22 by thisiskeithb