MarlinFirmware/Marlin #23569: [BUG] Delta printer skips steps when leveling the bed

Salvogi, 2022-01-19:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Delta printer skips steps when leveling the bed.
https://youtu.be/WpirlzJCKW4
While no problems when printing.
I verified that there are no mechanical jamming, that the nozzle is able to reach all the points to be probed and I tried to increase the stepper current.

### Bug Timeline

New bug

### Expected behavior

The printer is successfully probing the bed

### Actual behavior

The printer loses steps and the bed leveling fails

### Steps to Reproduce

1. Homing all axis
2. Send G1 X0 Y0 Z5
3. Select bed leveling from the printer display
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7899896/Configuration.zip)

### Version of Marlin Firmware

2.0.9.3

### Printer model

Custom delta printer with smart effector that use the nozzle as a probe

### Electronics

Makerbase MKS SGEN_L V1.0 32bit

### Add-ons

Simple electrical contact normally closed for bed probing

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

rq3, 2022-01-19:
I have a delta printer, and use a piezo sensor for nozzle contact sensing. I find that the bed and nozzle have to be absolutely clean during bed leveling. The smallest amount of filament ooze, or a cat hair on the bed, will make it skip steps.

Just a thought.

Salvogi, 2022-01-19:
I was very careful to clean the nozzle. While on the plate I left some traces of hair spray.
However, between the various attempts I got step skips even before the first point was probed.

Salvogi, 2022-01-21:
Today, while I was recording a video, the printer skips steps immediately after homing, before touching the first point. Here is the video https://youtu.be/fWo61x27y6M

Salvogi, 2022-01-23:
I tried to change the MKS SGEN_L V1.0 32bit board whit TMC2208 to a RAMPS 1.4 one with A4988 but the problem persists.
The printer prints fine but I can't do autmatic leveling https://youtu.be/R38wIFLa53g
[image]
Maybe I have a compiler problem, I tried to uninstall Vscode and delete the Vscode and Platformio folders from my Windows user folder and reinstall everything but the problem persists.
As soon as I have another PC available, I try to compile again.

Salvogi, 2022-01-23:
I installed Vscode + Platformio Linux version on a Chromebook, I compiled the firmware but the result was the same.
On the Chromebook I got two warnings:
- Unable to resolve configuration with compilerPath "C:/Users/Salvo/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc.exe". Using "/usr/bin/cpp" instead.
- CMakeLists.txt was not found in the root of the folder 'Marlin-2.0.9.3_Delta_MKS_TMC2208'. How would you like to proceed?

Bob-the-Kuhn, 2022-02-03:
I've seen this behavior when the probe is not being triggered when doing calibration. 

I think what is happening is that the probe hits the minimum allowed Z before the probe triggers.  

Try changing **Z_PROBE_LOW_POINT** from -1 to something like -10 and see what happens.

Another possibility is your **DELTA_HEIGHT** is set too low.

I like to use the M48 command to see how the probing is behaving.  Move the head near the failed calibration probe point and see what M48 does.  You'll get some weird Z axis movement if the probe isn't triggering.

---

I'm using a Duet Smart effector on my delta.  It's still alive despite all I've done to it.  

I noticed a few other items when going through your **configuration.h** file.

A piezo probe/head will only trigger AFTER it touches the bed.  Therefore your **NOZZLE_TO_PROBE_OFFSET** should **ALWAYS** be negative so your setting of `{ 0, 0, 0.2 }` looks wrong.  My setting is `{ 0, 0, -7.7 }`.  This assumes that the nozzle just touches the bed when Z is zero and bed leveling is turned off.

How did you come up with the delta values in your configuration.h file?  I use the results from the G33 command.  

I see you are using **AUTO_BED_LEVELING_BILINEAR**.  I prefer **AUTO_BED_LEVELING_UBL** because it allows you to fine tune the mesh so that the first layer can be dialed in.  Takes a lot longer to setup but the result is better.  It can also be frustrating.

closed 2022-04-15 by github-actions[bot]