MarlinFirmware/Marlin #23570: Diagonal X and Y movements

nicken83, 2022-01-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

X and Y movements only diagonal when driven by repetier host or pronterface, Motors do work separately, but not like a corexy should. Steppers and motors are ok. Corexy is defined in configuration.h.

### Bug Timeline

New configuration

### Expected behavior

I expected straight linear movement from X and Y axis.

### Actual behavior

When X movement asked, only 1 motor moves instead of 2 counterclockwise
When Y movement asked, only 1 motor moves instead of 2 in same direction.

### Steps to Reproduce

Configure marlin 2.0.9.3 for corexy with BTT.SKR.2_revb motherboard and BTT TMC2209 stepper drivers in Uart mode.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Tronxy X5S

### Electronics

BTT SKR 2 revb

### Add-ons

BTT TMC2209 stepper drivers

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7899990/Configuration.zip)

ellensp, 2022-01-25:
Which of the following  build environment are you using?
env:BIGTREE_SKR_2 
env:BIGTREE_SKR_2_USB 
env:BIGTREE_SKR_2_USB_debug 
env:BIGTREE_SKR_2_F429 
env:BIGTREE_SKR_2_F429_USB 
env:BIGTREE_SKR_2_F429_USB_debug

nicken83, 2022-01-25:
Hello,
The BIGTREE_SKR_2.

ellensp, 2022-01-25:
So you have the older non F429 Micro processor. 
I don't have that board but I do have the BIGTREE_SKR_2_F429

Using your configs (updated for current bugfix) and changing the build environment. I cannot replicate this issue
[Bugfix Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7931664/Bugfix.Configuration.zip)

I'm watching the actual signals.

enable works as expected

step acts on both A and B steppers when moving X or Y any angles other than diagonally
Diagonal moves just move A or B as expected

Directions are inverted from A to B, so when A is high, B is Low and they change to the opposite as expected  when changing direction.

Did you reset eeprom back to firmware defaults, ie M502 (load defaults) then M500 (save)

I also notice you have STOP_ON_ERROR enabled. So if a stepper driver should get into a error state it will stop.
This would be reported on the serial port.

thinkyhead, 2022-01-26:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

nicken83, 2022-01-26:
issue is solved. The problem was the SD card. Firmware was never flashed.
Active firmware was preinstalled version from factory. That's why the
movements weren't correct. Thx for the advice!

closed 2022-01-26 by nicken83